espressif/esp-mqtt #2: race conditions in MQTT_Publish when queue is full

st0ff3r, 2016-11-16:
there seems to some kind of race conditions in MQTT_Publish(), happens sometimes when the queue is full.

I have added some debug printing to QUEUE_Puts and QUEUE_Gets:

MQTT_Publish id: 125
QUEUE_Puts: failed to add 192 to queue
QUEUE_Puts: queue size(20480/20480)
QUEUE_Gets: removed 192 from queue
QUEUE_Puts: queue size(20286/20480)
QUEUE_Puts: failed to add 192 to queue
QUEUE_Puts: queue size(20480/20480)
QUEUE_Gets: removed 192 from queue
QUEUE_Puts: queue size(20286/20480)
QUEUE_Puts: failed to add 192 to queue
QUEUE_Puts: queue size(20480/20480)
QUEUE_Gets: removed 192 from queue
QUEUE_Puts: queue size(20286/20480)
QUEUE_Puts: failed to add 192 to queue
QUEUE_Puts: queue size(20480/20480)
QUEUE_Gets: removed 192 from queue
QUEUE_Puts: queue size(20286/20480)
QUEUE_Puts: failed to add 192 to queue
QUEUE_Puts: queue size(20480/20480)
QUEUE_Gets: removed 192 from queue
QUEUE_Puts: queue size(20286/20480)
QUEUE_Puts: failed to add 192 to queue
QUEUE_Puts: queue size(20480/20480)
QUEUE_Gets: removed 192 from queue
QUEUE_Puts: queue size(20286/20480)
QUEUE_Puts: failed to add 192 to queue
QUEUE_Puts: queue size(20480/20480)
QUEUE_Gets: removed 192 from queue
QUEUE_Puts: queue size(20286/20480)
QUEUE_Puts: failed to add 192 to queue
QUEUE_Puts: queue size(20480/20480)
QUEUE_Gets: removed 192 from queue
QUEUE_Puts: queue size(20286/20480)
QUEUE_Puts: failed to add 192 to queue
QUEUE_Puts: queue size(20480/20480)
QUEUE_Gets: removed 192 from queue
QUEUE_Puts: queue size(20286/20480)
QUEUE_Puts: failed to add 192 to queue
QUEUE_Puts: queue size(20480/20480)
QUEUE_Gets: removed 192 from queue
QUEUE_Puts: queue size(20286/20480)
QUEUE_Puts: failed to add 192 to queue
QUEUE_Puts: queue size(20480/20480)
QUEUE_Gets: removed 192 from queue
QUEUE_Puts: queue size(20286/20480)
QUEUE_Puts: failed to add 192 to queue
QUEUE_Puts: queue size(20480/20480)
QUEUE_Gets: removed 192 from queue
QUEUE_Puts: queue size(20286/20480)
QUEUE_Puts: failed to add 192 to queue
QUEUE_Puts: queue size(20480/20480)
QUEUE_Gets: removed 192 from queue
QUEUE_Puts: queue size(20286/20480)
QUEUE_Puts: failed to add 192 to queue
QUEUE_Puts: queue size(20480/20480)
QUEUE_Gets: removed 192 from queue
QUEUE_Puts: queue size(20286/20480)
QUEUE_Puts: failed to add 192 to queue
QUEUE_Puts: queue size(20480/20480)
QUEUE_Gets: removed 192 from queue
QUEUE_Puts: queue size(20286/20480)
QUEUE_Puts: failed to add 192 to queue
QUEUE_Puts: queue size(20480/20480)
QUEUE_Gets: removed 192 from queue
QUEUE_Puts: queue size(20286/20480)
QUEUE_Puts: failed to add 192 to queue
[... 260 lines cut ...]
QUEUE_Puts: queue size(20480/20480)
QUEUE_Gets: removed 192 from queue
QUEUE_Puts: queue size(20286/20480)
QUEUE_Puts: failed to add 192 to queue
QUEUE_Puts: queue size(20480/20480)
QUEUE_Gets: removed 192 from queue
QUEUE_Puts: queue size(20286/20480)
QUEUE_Puts: failed to add 192 to queue
QUEUE_Puts: queue size(20480/20480)
QUEUE_Gets: removed 192 from queue
QUEUE_Puts: queue size(20286/20480)
QUEUE_Puts: failed to add 192 to queue
QUEUE_Puts: queue size(20480/20480)
QUEUE_Gets: removed 192 from queue
QUEUE_Puts: queue size(20286/20480)
QUEUE_Puts: failed to add 192 to queue
QUEUE_Puts: queue size(20480/20480)
QUEUE_Gets: removed 192 from queue
QUEUE_Puts: queue size(20286/20480)
QUEUE_Puts: failed to add 192 to queue
QUEUE_Puts: queue size(20480/20480)
QUEUE_Gets: removed 192 from queue
QUEUE_Puts: queue size(20176/20480)
QUEUE_Puts: added 192 to queue
QUEUE_Puts: queue size(20371/20480)

st0ff3r, 2016-11-18:
supposed fix:

diff --git a/mqtt/queue.c b/mqtt/queue.c
index 7fa50aa..4a7e146 100644
--- a/mqtt/queue.c
+++ b/mqtt/queue.c
@@ -34,6 +34,11 @@
 #include "os_type.h"
 #include "mem.h"
 #include "proto.h"
+
+uint8_t *last_rb_p_r;
+uint8_t *last_rb_p_w;
+uint32_t last_fill_cnt;
+
 void ICACHE_FLASH_ATTR QUEUE_Init(QUEUE *queue, int bufferSize)
 {
   queue->buf = (uint8_t*)os_zalloc(bufferSize);
@@ -41,7 +46,20 @@ void ICACHE_FLASH_ATTR QUEUE_Init(QUEUE *queue, int bufferSize)
 }
 int32_t ICACHE_FLASH_ATTR QUEUE_Puts(QUEUE *queue, uint8_t* buffer, uint16_t len)
 {
-  return PROTO_AddRb(&queue->rb, buffer, len);
+  uint32_t ret;
+  
+  last_rb_p_r = queue->rb.p_r;
+  last_rb_p_w = queue->rb.p_w;
+  last_fill_cnt = queue->rb.fill_cnt;
+
+  ret = PROTO_AddRb(&queue->rb, buffer, len);
+  if (ret == -1) {
+    // rolling ring buffer back
+    queue->rb.p_r = last_rb_p_r;
+    queue->rb.p_w = last_rb_p_w;
+    queue->rb.fill_cnt = last_fill_cnt;
+  }
+  return ret;
 }
 int32_t ICACHE_FLASH_ATTR QUEUE_Gets(QUEUE *queue, uint8_t* buffer, uint16_t* len, uint16_t maxLen)
 {

tuanpmt, 2016-11-18:
@st0ff3r can you send a pull-request for this

Thanks

st0ff3r, 2016-11-18:
> On 18 Nov 2016, at 04.46, Tuan PM <notifications@github.com> wrote:
> 
> @st0ff3r can you send a pull-request for this

done, and thanx for the nice work :)

closed 2017-02-26 by tuanpmt