espressif/esp-mqtt #5: mqtt_publish from another task ?

cherva, 2017-04-06:
Can anyone tell me how to send data to the broker from another task ?
What client should I use as a pointer to mqtt_publish ?
I tried the one that is returned from mqtt_start(&settings), but as soon as it sends the data the esp crashes and reboots.....

tuanpmt, 2017-04-08:
```
//global client ref
mqtt_client_t *client = mqtt_start(&settings);

//in another task:
mqtt_publish(client, "/test", "howdy!", 6, 0, 0);
```

forthlightning, 2017-04-20:
I have had the same experience of crashes when referencing a global client object. 

@cherva the suggestion from @tuanpmt is exactly what you tried correct?

cherva, 2017-04-22:
Sorry I didn't have the time to update the issue.
It turns out I have to stop coding at 2AM...
I did extactly what @tuanpmt posted, but the location I was creating the mqtt client object was stupid.
The scope was being destroyed and that is why I was able to send one message and on the second I got the crash.

Sorry about that. Closing issue as it is clearly my bad.

closed 2017-04-22 by cherva