espressif/esp-mqtt #18: Only first message processed in multiple PUBLISH messages in single TCP read

DavidAntliff, 2017-12-13:
I'm still investigating this however it would be good to have a discussion with someone else familiar with the code. I'm not sure what the purpose of the loop in `deliver_publish()` is. I thought it might be to handle fragmented TCP reads, but the data callback is invoked before any subsequent reads occur. Or perhaps it is to handle multiple Publish Message instances in a single TCP read?

I have a case that fails in the circumstance where multiple Publish Messages are present in a single TCP read. `deliver_publish()` is called with a total length covering all included messages, but only the first message is processed.

Packet data (two _intentionally_ sent publish messages, that happen to have the same value coincidentally):

```
0000   24 0a c4 0e 4e f4 b8 27 eb 00 44 6c 08 00 45 00  $...N..'..Dl..E.
0010   00 5c fd 22 40 00 3f 06 58 c5 c0 a8 b2 2c c0 a8  .\."@.?.X....,..
0020   b2 36 07 5b 57 2a eb 45 10 42 00 00 1a 03 50 18  .6.[W*.E.B....P.
0030   72 10 db 3d 00 00 30 18 00 08 78 6d 61 73 2f 72  r..=..0...xmas/r
0040   65 64 30 2e 35 33 33 37 35 35 33 30 32 34 32 39  ed0.533755302429
0050   30 18 00 08 78 6d 61 73 2f 72 65 64 30 2e 35 33  0...xmas/red0.53
0060   33 37 35 35 33 30 32 34 32 39                    3755302429
```

I've added additional debug to show the values of each variable just after they are set:

```
[MQTT INFO] Read len 52
[MQTT INFO] msg_type 3, msg_id 0, pending_type 0, pending_id 0
[MQTT INFO] deliver_publish
[MQTT INFO] deliver_publish: length 52
[MQTT INFO] client->mqtt_state.message_length 26
[MQTT INFO] client->mqtt_state.message_length_read 52
[MQTT INFO] event_data.topic_length 8
[MQTT INFO] event_data.data_length 14
[MQTT INFO] total_mqtt_len -12
[MQTT INFO] mqtt_len 14
[MQTT INFO] event_data.data_total_length 65524
[MQTT INFO] event_data.data_offset 0
[MQTT INFO] event_data.data_length 14
[MQTT INFO] Data received: 14/-12 bytes 
```

Note the negative value of `total_mqtt_len` and subsequent nonsense value of `event_data.data_total_length`. The loop only executes _once_.

If I could better understand the intent of the loop I could look at improving it, however as it stands I don't think it properly handles multiple messages per read, and I'm not sure about fragmented reads either.

[image]

tuanpmt, 2018-02-27:
Could you please try the latest commit of `espmqtt`?

DavidAntliff, 2018-02-27:
@tuanpmt my project doesn't use this library any more, unfortunately, because I found that it drops messages (in both directions) due to treating the TCP socket as a "packet" interface rather than a "stream" - for example, if multiple messages from the broker arrive at once, they will be combined in a single socket read, but the code would drop all but the first message, resulting in missed messages.

If you've resolved this issue I would be willing to try out your library again - please advise.

DavidAntliff, 2018-02-27:
@tuanpmt sorry, I deleted my previous comment as I thought this was a different issue.

Yes, I will try out your library again when I get a chance. It won't be soon though, as I will need to reintegrate it into my project. Thank you for looking into this and other issues.

DavidAntliff, 2018-02-27:
BTW, have you addressed the problem where incoming messages (received from the broker) are dropped if there is more than one in the TCP read? The code, as written a few months ago, would discard all but the first message, when it really should loop on the read, and consume each message one by one. Otherwise this leads to dropped messages. Has this been addressed at the same time as this issue?

tuanpmt, 2018-02-27:
I changed the read method using select and sequentially processed each packet, instead of starting a read only task like before

cgawron, 2018-05-01:
I think the issue is that deliver_publish eagerly reads **all available bytes** from the network:
```
len_read = transport_read(client->transport,
                                      (char *)client->mqtt_state.in_buffer,
                                      client->mqtt_state.in_buffer_length,
client->config->network_timeout_ms);
```
Actually it should read no more than the remaining bytes belonging to the current message. See the patch I posted on issue #34 today - that should also fix this.

steadfastnz, 2018-06-24:
I've come across this issue in my work.

I've got the latest master version and checked that the code contains cgrawron's patch, but that patch doesn't appear to fix this issue.

I have a device that subscribes to several topics.  I'm controlling my devices via Home Assistant (HA) and using a local mosquitto instance as the broker.  When I send individual messages from HA my device gets each one, but when I use a Scene within HA multiple messages are sent rapidly, and some end up within the same TCP packet. mqtt_client is only passing on the first message in the packet so my device code is not seeing them all.  

I'm getting debug messages like "deliver_publish, message_length_read=201, message_length=49"

Has there been any progress on a fix for this?  Any advice on where I should start?

LaurentLouf, 2018-09-30:
Hello there. I'm having a look at this lib for some project I'm working on. And I've noticed a couple of problems and things that could be improved. Don't hesitate to have a look at my modifications and to leave comments there https://github.com/LaurentLouf/esp-mqtt/commits/fix-message-sizes . I'll probably do a PR later on, but if you see a problem, let me know.

LaurentLouf, 2018-10-01:
So a little update on my progress. The patch of @cgawron did improve the situation a bit for multiple messages in a single TCP read, but isn't enough. Problem is here : `client->mqtt_state.in_buffer_length = buffer_size;` (in `esp_mqtt_client_init`) which has a default value of 1024, way larger than probably most of MQTT messages. Then in `mqtt_process_receive` we do a read from the transport using this buffer length and only after that are we using `deliver_publish` . Problem is, at this point, we already have read 1024, so even if you don't read anything more from the transport in that function, it's already too late since we don't iterate in `mqtt_process_receive`. So the fix would be to iterate on what have been read, until we reach the end of the buffer.

LaurentLouf, 2018-10-02:
I've pushed a couple of commits in my master that seem to fix this problem (tested on my end with QoS0 messages sent rapidly one after the other, and I receive them all, even when packed five into one TCP packet). If any of the maintainers of the repo is interested, please let me know how to proceed with PRs (I'm working alone on my fork, so I could force push to have different branches for different PRs).

LaurentLouf, 2018-10-03:
Will do, but there's a lot of things going wrong with the lib, because part of the transports and SSL have been integrated into the last version of ESP-IDF, causing conflicts with functions having the same name. But of course, in the implementation in ESP-IDF they aren't using fixed width integers (I really don't know why, in embedded systems development, someone would chose not to have these kind of stronger types), and I also need them. So I'm gonna stick with my fork for a bit, but would gladly make PRs in this repo if instructed how to.

cgawron, 2018-10-03:
@LaurentLouf Actually I posted my patch because I ran into problems with messages larger than one packet (I'm using MQTT for OTA updates). I do not receive enough small messages to have the issue you describe.

Submitting a PR is straight forward:
- create a fork of the repo on github,
- create a feature branch in your fork which contains your fix,
- click on the "New pull request" button next to the branch icon
[image]

david-cermak, 2019-03-21:
Hi all, 
I'm closing this, as these issues have already been resolved in these commits:
https://github.com/espressif/esp-mqtt/commit/db64b791205c8dbcd0f3826a9544f0aa0301baaf
https://github.com/espressif/esp-mqtt/commit/cf5b8eda89057f5eb49654d2eaf4d1a209af2d11
https://github.com/espressif/esp-mqtt/commit/d4b66556186141e09db89c57781c6709e0e71d8a

Note that this should solve multiple mqtt messages in one buffer, larger mqtt messages than the buffer or overlapping between buffers. Also a  solution to ws transport's been merged to IDF.

Please feel free to reopen or open a new issue in case you experience any dropped messages.

closed 2019-03-21 by david-cermak