espressif/esp-mqtt #21: Guru Meditation Error

AmrutaCh, 2018-02-21:
ESP-idf version- v3.1-dev-239-g1c3dd23f
MQTT server- local mosquitto server

Hello,
I am using mqtt-tcp example and modified it as below:
`esp_mqtt_client_config_t mqtt_cfg = {
    .event_handle = mqtt_event_handler,
    .host = "192.168.1.9",
    .port = 1883, 
    .client_id = "mqtt_client_id",
    .username = "user",
    .password = "password",
    .keepalive = 10,
    .lwt_topic = "/lwt",
    .lwt_msg = "offline",
    .lwt_qos = 2,
    .lwt_retain = 0,
};`

Its crashing while connecting to server
`ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5656
load:0x40078000,len:0
ho 12 tail 0 room 4
load:0x40078000,len:13716
entry 0x40078fb4
[0;32mI (30) boot: ESP-IDF v3.1-dev-239-g1c3dd23-dirty 2nd stage bootloader[0m
[0;32mI (30) boot: compile time 16:02:35[0m
[0;32mI (32) boot: Enabling RNG early entropy source...[0m
[0;32mI (37) boot: SPI Speed      : 40MHz[0m
[0;32mI (41) boot: SPI Mode       : DIO[0m
[0;32mI (45) boot: SPI Flash Size : 4MB[0m
[0;32mI (49) boot: Partition Table:[0m
[0;32mI (53) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (60) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (68) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (75) boot:  2 factory          factory app      00 00 00010000 00100000[0m
[0;32mI (83) boot: End of partition table[0m
[0;32mI (87) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x1713c ( 94524) map[0m
[0;32mI (128) esp_image: segment 1: paddr=0x00027164 vaddr=0x3ffb0000 size=0x0332c ( 13100) load[0m
[0;32mI (134) esp_image: segment 2: paddr=0x0002a498 vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (136) esp_image: segment 3: paddr=0x0002a8a0 vaddr=0x40080400 size=0x05770 ( 22384) load[0m
[0;32mI (153) esp_image: segment 4: paddr=0x00030018 vaddr=0x400d0018 size=0x6e5d0 (452048) map[0m
[0;32mI (310) esp_image: segment 5: paddr=0x0009e5f0 vaddr=0x40085b70 size=0x0ba24 ( 47652) load[0m
[0;32mI (330) esp_image: segment 6: paddr=0x000aa01c vaddr=0x400c0000 size=0x00000 (     0) load[0m
[0;32mI (341) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (341) boot: Disabling RNG early entropy source...[0m
[0;32mI (342) cpu_start: Pro cpu up.[0m
[0;32mI (346) cpu_start: Starting app cpu, entry point is 0x40080f30[0m
[0;32mI (337) cpu_start: App cpu up.[0m
[0;32mI (357) heap_init: Initializing. RAM available for dynamic allocation:[0m
D (363) heap_init: New heap initialised at 0x3ffae6e0[0m
[0;32mI (368) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[0m
D (374) heap_init: New heap initialised at 0x3ffb9248[0m
[0;32mI (379) heap_init: At 3FFB9248 len 00026DB8 (155 KiB): DRAM[0m
[0;32mI (386) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM[0m
[0;32mI (392) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
D (399) heap_init: New heap initialised at 0x40091594[0m
[0;32mI (404) heap_init: At 40091594 len 0000EA6C (58 KiB): IRAM[0m
[0;32mI (410) cpu_start: Pro cpu start user code[0m
D (422) clk: RTC_SLOW_CLK calibration value: 3536256[0m
D (96) intr_alloc: Connected src 46 to int 2 (cpu 0)[0m
D (97) intr_alloc: Connected src 57 to int 3 (cpu 0)[0m
D (97) intr_alloc: Connected src 24 to int 9 (cpu 0)[0m
[0;32mI (102) cpu_start: Starting scheduler on PRO CPU.[0m
D (0) intr_alloc: Connected src 25 to int 2 (cpu 1)[0m
[0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
D (117) heap_init: New heap initialised at 0x3ffe0440[0m
D (127) heap_init: New heap initialised at 0x3ffe4350[0m
D (137) intr_alloc: Connected src 16 to int 12 (cpu 0)[0m
[0;32mI (137) MQTT_SAMPLE: [APP] Startup..[0m
[0;32mI (137) MQTT_SAMPLE: [APP] Free memory: 274420 bytes[0m
[0;32mI (147) MQTT_SAMPLE: [APP] IDF version: v3.1-dev-239-g1c3dd23-dirty[0m
I (187) wifi: wifi firmware version: 403db1d
I (187) wifi: config NVS flash: enabled
I (187) wifi: config nano formating: disabled
[0;32mI (187) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[0;32mI (197) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
I (227) wifi: Init dynamic tx buffer num: 32
I (227) wifi: Init data frame dynamic rx buffer num: 32
I (227) wifi: Init management frame dynamic rx buffer num: 32
I (237) wifi: wifi driver task: 3ffc08ac, prio:23, stack:4096
I (237) wifi: Init static rx buffer num: 10
I (237) wifi: Init dynamic rx buffer num: 32
I (247) wifi: wifi power manager task: 0x3ffc54ec prio: 21 stack: 2560
[0;32mI (257) MQTT_SAMPLE: start the WIFI SSID:[ExploreEmbedded] password:[******][0m
[0;32mI (287) phy: phy_version: 366.0, ba9923d, Dec 29 2017, 14:25:06, 0, 0[0m
I (287) wifi: mode : sta (24:0a:c4:00:20:9e)
[0;32mI (287) MQTT_SAMPLE: Waiting for wifi[0m
I (657) wifi: n:3 1, o:1 0, ap:255 255, sta:3 1, prof:1
I (1637) wifi: state: init -> auth (b0)
I (1637) wifi: state: auth -> assoc (0)
I (1647) wifi: state: assoc -> run (10)
I (1717) wifi: connected with ExploreEmbedded, channel 3
I (4647) wifi: pm start, type:0

[0;32mI (5087) event: sta ip: 192.168.1.10, mask: 255.255.255.0, gw: 192.168.1.1[0m
D (5087) MQTT_CLIENT: MQTT client_id=mqtt_client_id[0m
D (5087) TRANSPORT_TCP: [sock=4096],connecting to server IP:192.168.1.9,Port:1883...[0m
Guru Meditation Error: Core  1 panic'ed (LoadProhibited)
. Exception was unhandled.
Register dump:
PC      : 0x400014fd  PS      : 0x00060d30  A0      : 0x8012d2b0  A1      : 0x3ffc8830  
A2      : 0x00000000  A3      : 0xfffffffc  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3ffc87e0  
A10     : 0x00000000  A11     : 0x3f40b597  A12     : 0x3ffc8a74  A13     : 0x3ffae8d8  
A14     : 0x00000000  A15     : 0x00000001  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace: 0x400014fd:0x3ffc8830 0x4012d2ad:0x3ffc8840 0x4012eaa9:0x3ffc8b50 0x40083151:0x3ffc8b80 0x4010cb47:0x3ffc8bd0`

Any helps will be greatly appreciated

AmrutaCh, 2018-02-21:
Found the issue. It crashes at L649 in mqtt_client.c. It may be because of `client->config->scheme` remains uninitialized when uri is not used in mqtt client but setting it to `null` didn't work. Removing `client->config->scheme` from the log at L649 solved the issue.

tuanpmt, 2018-02-22:
Thank you @AmrutaCh 

I've updated the library to add default scheme, instead remove it

commit dd696909 (2018-02-22): Add default scheme, resolve #21, remove sdkconfig in examples
closed 2018-02-22 by tuanpmt