espressif/esp-mqtt #25: Auto-reconnect leaks memory

tijnkooijmans, 2018-02-27:
When connection to broker is lost, the library automatically re-connects. Unfortunately we lose about 30 kB of heap here when using SSL. Hope you have a fix for this.

tuanpmt, 2018-02-27:
@tijnkooijmans This commit should resolve this problem: https://github.com/tuanpmt/espmqtt/commit/a5ef145163ff202fd0519d7142da8d5e2656b7d6

Thank you for the information

tuanpmt, 2018-02-27:
You can test memory leak by:

```
while (1) {
    esp_mqtt_client_start(client);
    ESP_LOGI(TAG, "[APP] Starter free memory: %d bytes", esp_get_free_heap_size());
    vTaskDelay(5000/portTICK_RATE_MS);
    esp_mqtt_client_stop(client);
    ESP_LOGI(TAG, "[APP] Stopped Free memory: %d bytes", esp_get_free_heap_size());
    vTaskDelay(1000/portTICK_RATE_MS);
}
```

tijnkooijmans, 2018-02-27:
Yes this works, excellent! Thanks a lot.

closed 2018-02-27 by tijnkooijmans

borch84, 2018-05-24:
Hi @tijnkooijmans I did some tests using the memory leak test code provided by @tuanpmt. I am adding the results here. I noticed that after 10 minutes I lost 3000 bytes. I did not applied the commit because lib/transport.c already has the changes. I git cloned the component into ESP-IDF folder just few weeks ago. 
[memory-leak-evidence.txt](https://github.com/tuanpmt/espmqtt/files/2036556/memory-leak-evidence.txt)