espressif/esp-mqtt #32: keepalive timeout

sportybj, 2018-03-31:
I tried this code and set the CPU FREQ 240MHz, keepalive of esp_mqtt_client_config_t to 120
my problems are: 
1. pingreq sent interval is 60sec instead of 120sec
2. if WIFI is connected but internet connect failed, it takes more than 30 minutes to find mqtt disconnection (seems found by socket layer instead of MQTT layer)

tuanpmt, 2018-04-02:
Hi @sportybj 

The pingreq interval = 1/2 KEEP_ALIVE_TIME, to make sure we do not send pingreq later than KEEP_ALIVE_TIME

panitaxx, 2018-05-13:
The issue with part 2  I think that when a keepalive timeout  occurs keepalive_tick  does not get updated. if a timeout occurs the break exits the switch before keepalive_tick gets updated. 

https://github.com/tuanpmt/espmqtt/blob/75d3c65cb65a5ce493d06d41bf79e85e8c110112/mqtt_client.c#L700-L706

my suggestion to fix it would be to put a keepalive_tick update also after this:

https://github.com/tuanpmt/espmqtt/blob/75d3c65cb65a5ce493d06d41bf79e85e8c110112/mqtt_client.c#L682

panitaxx, 2018-05-16:
If you are using SSL this line blocks 

https://github.com/tuanpmt/espmqtt/blob/f3f223e03195e95f970c1c310e409c07a2adc3af/lib/transport_ssl.c#L197

vtunr, 2018-06-01:
https://github.com/tuanpmt/espmqtt/pull/54 might fix the MQTT layer disconnection detection.

tmedicci, 2018-08-01:
I am facing the same issue related with 2.

david-cermak, 2019-03-29:
Hi all, can we close this issue? 
AFAIK there's no problem anymore with the library to actively disconnect on no reply to ping message.

david-cermak, 2020-01-09:
Closing this old issue, as I believe there's no such problem with the current version and no update here. Feel free to reopen or open another issue.

closed 2020-01-09 by david-cermak