espressif/esp-mqtt #33: [help request]How to send and receive stuff?

SouXx, 2018-03-31:
# Send and Receive

Hello,

first of all thank you for your work!
I am quite new to how networks and MQTT works. So I hope my question isnt too dump.
I know at least the Basics of the freeRTOS and I am aware of the ESP-IDF API. And I think I also got a raw idea how your code Works but I am far away from knowing whats going on there!

My goal is to receive and process a MQTT paket. To do so I tried to understand how the code works.
I try to Start from the TCP Example.

## 1. Init

I call the Example in a Task, so renamed the ```app_main``` and called it ```mqtt_client```
```
xTaskCreatePinnedToCore(mqtt_client, "MQTT Client", 4000, NULL, 2, NULL, 0);

        while(1){
                printf("main: Idle\n");
                vTaskDelay(1000/portTICK_PERIOD_MS);
        }
        vTaskDelete(NULL);
}
```
I pretty much left the Example as it is.
Just added a Delay and vTaskDelete() at the end of ```mqtt_client``` but it should not reached there.

```
    while(1){
                vTaskDelay(1000/portTICK_PERIOD_MS);
        }
        vTaskDelete(NULL);
    }
```
So far so good, I flashed it
the output seems to be okay:

```
I (0) cpu_start: App cpu up.
I (355) heap_init: Initializing. RAM available for dynamic allocation:
I (362) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (368) heap_init: At 3FFB9148 len 00026EB8 (155 KiB): DRAM
I (374) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (381) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (387) heap_init: At 4008E8EC len 00011714 (69 KiB): IRAM
I (393) cpu_start: Pro cpu start user code
I (76) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
This is ESP32 chip with 2 CPU cores, WiFi/BT/BLE, silicon revision 0, 4MB external flash
I (83) MQTT_ClIENT: [APP] Startup..
I (93) MQTT_ClIENT: [APP] Free memory: 270276 bytes
I (93) MQTT_ClIENT: [APP] IDF version: v3.1-dev-601-gda278163-dirty
I (133) wifi: wifi firmware version: c202b34
I (133) wifi: config NVS flash: enabled
I (133) wifi: config nano formating: disabled
I (133) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (143) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (173) wifi: Init dynamic tx buffer num: 32
I (173) wifi: Init data frame dynamic rx buffer num: 32
I (173) wifi: Init management frame dynamic rx buffer num: 32
I (173) wifi: wifi driver task: 3ffc18a0, prio:23, stack:4096
I (183) wifi: Init static rx buffer num: 10
I (183) wifi: Init dynamic rx buffer num: 32
I (183) wifi: wifi power manager task: 0x3ffc64e0 prio: 21 stack: 2560
I (193) MQTT_ClIENT: start the WIFI SSID:[Boot] password:[******]
I (273) phy: phy_version: 383.0, 79a622c, Jan 30 2018, 15:38:06, 0, 0
I (273) wifi: mode : sta (24:0a:c4:07:5d:c0)
I (273) MQTT_ClIENT: Waiting for wifi
main: Idle
main: Idle
I (1603) wifi: n:11 0, o:1 0, ap:255 255, sta:11 0, prof:1
I (2583) wifi: state: init -> auth (b0)
I (2583) wifi: state: auth -> assoc (0)
main: Idle
I (2593) wifi: state: assoc -> run (10)
I (2623) wifi: connected with Boot, channel 11
main: Idle
I (3593) event: sta ip: 192.168.2.116, mask: 255.255.255.0, gw: 192.168.2.1
I (3593) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (3743) MQTT_CLIENT: Sending MQTT CONNECT message, type: 1, id: 0000
I (3883) MQTT_ClIENT: MQTT_EVENT_CONNECTED
I (3883) MQTT_ClIENT: sent subscribe successful, msg_id=51413
I (3883) MQTT_ClIENT: sent subscribe successful, msg_id=59147
I (3893) MQTT_ClIENT: sent unsubscribe successful, msg_id=3030
I (4043) MQTT_ClIENT: MQTT_EVENT_SUBSCRIBED, msg_id=51413
I (4043) MQTT_ClIENT: sent publish successful, msg_id=0
I (4193) MQTT_ClIENT: MQTT_EVENT_SUBSCRIBED, msg_id=59147
I (4193) MQTT_ClIENT: sent publish successful, msg_id=0
I (4313) MQTT_ClIENT: MQTT_EVENT_UNSUBSCRIBED, msg_id=3030
I (4443) MQTT_CLIENT: deliver_publish, message_length_read=19, message_length=19
I (4443) MQTT_ClIENT: MQTT_EVENT_DATA
TOPIC=/topic/qos0
DATA=data
main: Idle
I (4663) MQTT_CLIENT: deliver_publish, message_length_read=19, message_length=19
I (4663) MQTT_ClIENT: MQTT_EVENT_DATA
TOPIC=/topic/qos0
DATA=data
main: Idle
I (5593) wifi: pm start, type:0

```
## 2. Stuggle

So but my first struggle is "How do is send stuff?" and even more interesting for me atm.
"How do I receive stuff?"

I went through your Code an found the 

```
static esp_err_t mqtt_event_handler(esp_mqtt_event_handle_t event)
```
It has as well as the  ```mqtt_app_start```
a line which say ```// .user_context = (void *)your_context``` (not exactly but it seems to belong to each other)

But what does that mean exactly? 
I need a Struct which exists in ```esp_mqtt_event_handle_t```
and it may carrys my payload? Event If I am right, how do i use it in a good Style?

I also found the function ```esp_mqtt_client_publish()``` the Name seems to be the right point to start, but
this function needs a "client" passed, its type is ```esp_mqtt_client_handle_t``` this type is also used in the 
```esp_mqtt_client_init``` 

It says:
```esp_mqtt_client_handle_t client = esp_mqtt_client_init(&mqtt_cfg);```

But I am a bit confused here, because it seems that I have to Initialize my MQTT Client each time I want to send something?

What ever I think I dont got it right so I would appreciate If some could help me out. What Functions do I have to call or to what do I have to listen If I want to send/receive stuff to an Topic.

thank you, 
Greetings

wdouglas1999, 2018-05-04:
@Souxx,
Good day,
Have you managed to figure out how to subscribe and publish massages yet? I would also like a basic example of how to do this.

AmrutaCh, 2018-05-04:
@SouXx and @wdouglas1999
Declare `mqttClient` as global variable and use it

> //  global declaration
> esp_mqtt_client_handle_t mqttClient;
> 
> // initialize
>     esp_mqtt_client_config_t mqtt_cfg = {
> ...
> ...
> ...};
> 
> mqttClient = esp_mqtt_client_init(&mqtt_cfg);
>  esp_mqtt_client_start(mqttClient);
> 
> // subscribe
> esp_mqtt_client_subscribe(mqttClient, topic, qos);
> 
> // publish
> esp_mqtt_client_publish(mqttClient, topic, data,  dataLength, qos,  retain);

Hope this is helpful

DarioB92, 2018-06-06:
Hello, thanks for your work, however I still cannot understand how to to receive a message from a subscribed topic. Do I have to set a callback function? A commented exemple of publishing and receiving a message could be real helpful. Thanks!

wdouglas1999, 2018-06-07:
@ DarioB92,

I ended up using the example following example code on github.

https://github.com/jessebraham/esp32-mqtt-client/blob/master/main/esp32-mqtt-client.c

I hope it will help you.

AmrutaCh, 2018-06-07:
@DarioB92  `mqtt_event_handler` is the required call back you are talking about and it is already set up. Whenever any message is available for subscribed topic `MQTT_EVENT_DATA` event will occur. 

>    case MQTT_EVENT_DATA:
>        ESP_LOGI(TAG, "MQTT_EVENT_DATA");
>        printf("TOPIC=%.*s\r\n", event->topic_len, event->topic);
>        printf("DATA=%.*s\r\n", event->data_len, event->data);
>        break; 

This will give you all required data regarding the received message. If you have basic knowledge of mqtt protocol, I think, the examples given in the repo are self-explanatory. You just need to try them.

david-cermak, 2019-03-28:
Closing this older issue, as it seems it has been answered. Please feel free to reopen or raise a new one if it's not the case.

closed 2019-03-28 by david-cermak