espressif/esp-mqtt #34: Receiving large payloads

lukecyca, 2018-04-01:
In order to support OTA firmware updates over MQTT, I need to be able to accept very large payloads. Right now any incoming payload larger than `MQTT_BUFFER_SIZE_BYTE` results in a crash immediately after receiving the first chunk:

```
I (172815) HOMIE: MQTT_EVENT_DATA
TOPIC=homie/30aea41b1168/$implementation/ota/firmware/13727f4f93b6e6862bfe16f4894d68a4
DATA=6QECQJzyEEAA8BBAaAUAABAQAABQ9RBAHEsAQMwkAED///8///8PQP9/EED///9f8P//PwAQAAAc4gBAAEoAQExKAEAABwBgAAAAgOgrAEDwMABAoC8AQLcdwQQAEgBgABAA63wSAGASwdAJsfmh/QEpTzhPIeb/KiNLPwxEAeb/wAAAjFIMEoYHAAAAIeH/KQ8oDygCKS8oDygSKT84H4svAd7/wAAADAIdDwix+KESwTAN8AAAABLBwAnx+eH9ASmPKI8i0hApDzLPECgPQqAIAdH/wAAAjEIMEkYuAAAoD4siKQ8MAikfBiUADAIpbwwCKX8yzxgoDwyEAcf/wAAAjDIMIsYjACgPiyIpDyhvKT8MAiJPCCg/McH/JzMEDBIiTwgoPzG//yezDCg/Mb7/JzMEDBIiTwgoPzG8/yezBSKgASJPCDIPCAwSICMwICB0jJIofzgPKiMpD8YGADg/SH8oDwGs/8AAAIxCDDJGCQAAKH84DyojKQ8oHxsiKR8iDxE4HyezAsbX/xGo/yhfPQKgAwAMAh0PCPH44RLBQA3wEsGwAmETwmES0mER4mEQ+fH9ASmPOZ9Jry0B7QIojyAgBMxiKJ8gIAQWQgAMEoYvACGX/yk/KD8LIilPKD+NAgwJgC0l0NkR0NIg0MgRKD9tAgwHYC0l0LcRsLIg0KYRKD/7IiAkQcAiESARwC0BLQIpXzivKD8qIwsyKD8gIGAgIxApDyiPKR8onykvBhYAKC84PwGA/8AAAAF//8AAAIxCDCIGEgAAOF8oH0g/AXH/wAAAjEIMMgYNAAA4XygvSD8Bdv/AAACMQgxCBggAADgfKD8qIykfOC8oPyojKS84Dyg/ICPAKQ8oD1YS+gwCHQ4dDwIhE8IhEtIhEeIhEPjxEsFQDfAAkqCgkBHAAmEn8mEmEPEgIqAJKQ
[...]
I (172915) HOMIE: MQTT_EVENT_DATA
Guru Meditation Error: Core  0 panic'ed (LoadProhibited)
. Exception was unhandled.
Core 0 register dump:
PC      : 0x4000c250  PS      : 0x00060230  A0      : 0x800daab1  A1      : 0x3ffc7d50
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000400  A5      : 0x3f400830
A6      : 0x3ffc7d80  A7      : 0xff000000  A8      : 0x80082d98  A9      : 0x3ffc7d20
A10     : 0x00000000  A11     : 0xffffffff  A12     : 0x00000004  A13     : 0x00000001
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff

Backtrace: 0x4000c250:0x3ffc7d50 0x400daaae:0x3ffc7d70 0x400d2c59:0x3ffc8080 0x4010ec0a:0x3ffc80d0 0x4010eeab:0x3ffc8130 0x4010f0d1:0x3ffc8150 0x4010f361:0x3ffc8170 0x4010f51e:0x3ffc81b0 0x4010f6f4:0x3ffc81e0
0x400daaae: _vfprintf_r at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vfprintf.c:1522

0x400d2c59: printf at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/printf.c:58

0x4010ec0a: homie_handle_mqtt_event at /Users/luke/Code/esp32/esp32-homie-example/components/esp32-homie/./homie.c:25

0x4010eeab: mqtt_event_handler at /Users/luke/Code/esp32/esp32-homie-example/components/esp32-homie/./homie.c:76 (discriminator 1)

0x4010f0d1: esp_mqtt_dispatch_event at /Users/luke/Code/esp32/esp32-homie-example/components/esp32mqtt/./mqtt_client.c:808

0x4010f361: deliver_publish at /Users/luke/Code/esp32/esp32-homie-example/components/esp32mqtt/./mqtt_client.c:808

0x4010f51e: mqtt_process_receive at /Users/luke/Code/esp32/esp32-homie-example/components/esp32mqtt/./mqtt_client.c:808

0x4010f6f4: esp_mqtt_task at /Users/luke/Code/esp32/esp32-homie-example/components/esp32mqtt/./mqtt_client.c:808

Rebooting...
```

Would it be possible to receive them in multiple chunks, each sized to `MQTT_BUFFER_SIZE_BYTE`? For example this is how [marvinroger/async-mqtt-client  does it](https://github.com/marvinroger/async-mqtt-client/blob/master/docs/3.-Memory-management.md)

cgawron, 2018-05-01:
I ran into the same issue.  The major issue seems to be that the code assumes that the message length is 16 bit long at maximum, e.g. here
```
const char* mqtt_get_publish_topic(uint8_t* buffer, uint16_t* length);
const char* mqtt_get_publish_data(uint8_t* buffer, uint16_t* length);
```
and in deliver_publish as well (mqtt_data_len, mqtt_offset are declared as uint16_t).

Another point is that the the topic is only set in the first message.

cgawron, 2018-05-01:
I have created a patch which fixed the bug for me:
[large_message_patch.txt](https://github.com/tuanpmt/espmqtt/files/1964793/large_message_patch.txt)

Thank you for providing espmqtt!

DavidAntliff, 2018-05-01:
@cgawron your patch would make a useful pull request, since it appears (I haven't tested it) to fix a legitimate problem with this library where the _TCP stream_ isn't treated correctly. Will it handle partial reads? I.e. a read from a socket that returns only the first part of a packet, needing a subsequent read to get the rest?

cgawron, 2018-05-02:
@DavidAntliff I hope that - apart from handling messages larger than 64k - this patch also fixes receiving multiple messages within one packet and messages spread over several packets.
I'm tested ota over mqtt using this patch and it works for me.
Be aware that you will receive several events for large messages and only the first one contains the topic. Events having non-zero `current_data_offset` will contain no topic.

fermuch, 2018-05-02:
@cgawron great work! Do you have an example of using espmqtt with OTAs?

closed 2018-05-03 by tuanpmt

fermuch, 2018-05-03:
Thank you @cgawron !

PR #40 (2018-05-03): fix handling of large messages
This patch fixes #34 and probably #18.