espressif/esp-mqtt #37: How is QOS 1 intended to work?

etcetc, 2018-04-17:
I can see in the code that when qos 1 is specified, the published message is saved in the mqtt_outbox and/or state pending_message field, and when the ack is received, it is cleared.  The problem I'm investigating is how to handle mqtt disconnects, ostensibly because of networking problems.  That is, a few messages are undelivered because the transport failed, we are notified of the mqtt_disconnect, and so we tear down and rebuild the connection.  At this point, I would like to retry sending un-ack'ed messages.

The way I imagined it would work is some form of retry mechanism on pending messages (for QOS1) (e.g. on reconnection), but don't see that in the code anywhere.  Given that the topic is not saved with the outbox item, or anywhere else, I don't see how a retry could be implemented, even though there is a retry_count field (that seems to not be used).   Any guidance is appreciated.

david-cermak, 2019-03-28:
Hi @etcetc 

Support for resending qos1 and qos2 messages on reconnection has been recently added. Please check this commit https://github.com/espressif/esp-mqtt/commit/6a0d1e7bff1e3085669d41ae3412c649db20108e

david-cermak, 2020-01-09:
Closing as no recent feedback

closed 2020-01-09 by david-cermak