espressif/esp-mqtt #39: How to get issue

donny681, 2018-05-01:
I try your method and get a certificate which is different to your certificate in the mqtt_ssl demo.
`openssl s_client -showcerts -connect iot.eclipse.org:8883 </dev/null 2>/dev/null|openssl x509 -outform PEM >iot_eclipse_org.pem`

`-----BEGIN CERTIFICATE-----
MIIGCTCCBPGgAwIBAgISBCgMMsgO+rB7pl+8oxmRJzKnMA0GCSqGSIb3DQEBCwUA
MEoxCzAJBgNVBAYTAlVTMRYwFAYDVQQKEw1MZXQncyBFbmNyeXB0MSMwIQYDVQQD
ExpMZXQncyBFbmNyeXB0IEF1dGhvcml0eSBYMzAeFw0xODA0MjMxMzM2MTNaFw0x
ODA3MjIxMzM2MTNaMBoxGDAWBgNVBAMTD2lvdC5lY2xpcHNlLm9yZzCCASIwDQYJ
KoZIhvcNAQEBBQADggEPADCCAQoCggEBANwqiLf1e6OY1alW19RLpey5IldDjVIW
FLC2vF8Nv+WkHvShN5phMbazrb1Tm3LC7mNvX73wHIMZcjJ0qnWLPOPA/fKCEZSn
RY0D/x4KN9tsL5NLJSPXcwuY3/SP8Gx5kPM2csDSDpAR9bZ/CgEe+gZphT3Ink0L
honSW7QCkW5APu/MYQddlXr6NrZpEK/PnDbtmVr7xo/BUqCSqrWJlaNftF6eKXJW
Ay4Zu3fhuQOuLRhAI+Sinv7I2f8LqfpmNzVYUc/ahi4KEdJUnMVmh5tziPED/fG0
qwVgwboNwJzXgKXhRUR7ihXL4iJJpVgTeSJfAPJjMpChXmZljYn4Xj0CAwEAAaOC
AxcwggMTMA4GA1UdDwEB/wQEAwIFoDAdBgNVHSUEFjAUBggrBgEFBQcDAQYIKwYB
BQUHAwIwDAYDVR0TAQH/BAIwADAdBgNVHQ4EFgQU+k6Am15WnUPA1ZSQXaK7JZ0R
DaswHwYDVR0jBBgwFoAUqEpqYwR93brm0Tm3pkVl7/Oo7KEwbwYIKwYBBQUHAQEE
YzBhMC4GCCsGAQUFBzABhiJodHRwOi8vb2NzcC5pbnQteDMubGV0c2VuY3J5cHQu
b3JnMC8GCCsGAQUFBzAChiNodHRwOi8vY2VydC5pbnQteDMubGV0c2VuY3J5cHQu
b3JnLzAaBgNVHREEEzARgg9pb3QuZWNsaXBzZS5vcmcwgf4GA1UdIASB9jCB8zAI
BgZngQwBAgEwgeYGCysGAQQBgt8TAQEBMIHWMCYGCCsGAQUFBwIBFhpodHRwOi8v
Y3BzLmxldHNlbmNyeXB0Lm9yZzCBqwYIKwYBBQUHAgIwgZ4MgZtUaGlzIENlcnRp
ZmljYXRlIG1heSBvbmx5IGJlIHJlbGllZCB1cG9uIGJ5IFJlbHlpbmcgUGFydGll
cyBhbmQgb25seSBpbiBhY2NvcmRhbmNlIHdpdGggdGhlIENlcnRpZmljYXRlIFBv
bGljeSBmb3VuZCBhdCBodHRwczovL2xldHNlbmNyeXB0Lm9yZy9yZXBvc2l0b3J5
LzCCAQQGCisGAQQB1nkCBAIEgfUEgfIA8AB1ACk8UZZUyDlluqpQ/FgH1Ldvv1h6
KXLcpMMM9OVFR/R4AAABYvLvA64AAAQDAEYwRAIgCDZaOTZnCU82Uo6lCMcqg79Q
wJXk2A5zqtBZAMc+eDsCID9p45HDrbOlHftbnmnjA75CRD7GIB52izggv2XafOTV
AHcA23Sv7ssp7LH+yj5xbSzluaq7NveEcYPHXZ1PN7Yfv2QAAAFi8u8D0AAABAMA
SDBGAiEA6a/IgfiCbbNYygRemL6MdbM9QQbB7DxGrwvirVxQneYCIQDy74FAsA8W
Lj8Ajy6WtBChRJrg8X2avdDR0Qk6cRDFNDANBgkqhkiG9w0BAQsFAAOCAQEASdA1
MZkWskB0423mxJ/WsR+MUjSZV3CXhQbPY+HQtiBKvQurUG1egZTy1D6AZg0LtMNG
T60hoCOA2kw+nAQt1tJ1eBcOzHY2w1qEeUl4qmycsHf7oxfQc0WfCZuZu/rwl6Td
SoTKsOsLYZ5tCqrFQISTgAhitqnJcMFsJz4lnA3BjbmQ6/M3LlN+j/wS1ESvj0I2
q5ejIhe6ge72A27IgMysWISdmReluu2fWmFmcOKswGuqfQCsYV9U+JEQeQ+rstHi
BgvQvG/06a6RsYMvKUJawjNM2Ar58k3JRrn/++xb+26FQVS7HIMeNobR1fVdI0Lc
hXx2rq59Hgy8T/vBrg==
-----END CERTIFICATE-----`

How to get the right certificate ?

david-cermak, 2018-10-10:
Hi @donny681 ,

The certificate is correct but the library cannot verify it without knowing the 'higher-instance' certificate. You can fix it by adding those certificates to tls configuration. From the  perspective of this lib, a simple fix is just to put the other certificate to the same txt file, for example

my_cert.pem:
```
-----BEGIN CERTIFICATE----- 
<certificate from iot_eclipse_org>
-----END CERTIFICATE-----
-----BEGIN CERTIFICATE-----
<certificate from let's encrypt>
-----END CERTIFICATE-----
```

Thanks,
David

david-cermak, 2019-03-28:
Hi @donny681 ,

Any luck using the CA certificate? Please let us know, so we can close the issue.
Thanks!

david-cermak, 2020-01-09:
Closing as the original issue seems to be a wrong certificate and there's no recent feedback

closed 2020-01-09 by david-cermak