espressif/esp-mqtt #41: Not connecting to Local Mosca SSL Broker

Kasuhikdas, 2018-05-03:
Mosca SSL Broker is working fine with mqtt client.But ESP32 not connecting to it.
 const esp_mqtt_client_config_t mqtt_cfg = {
        .uri = "mqtts://192.168.43.187",
        .port = "8883",
        .event_handle = mqtt_event_handler,
        .cert_pem = (const char *)ca_crt_start,
        .transport = MQTT_TRANSPORT_OVER_SSL,
        .lwt_topic = "Death",
        .lwt_msg = "Device is Disconnected",
    };
this is the config.
and output is 
I (3107) event: sta ip: 192.168.43.114, mask: 255.255.255.0, gw: 192.168.43.1
I (3107) MQTTS_SAMPLE: [APP] Free memory: 239004 bytes
I (3107) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
E (3127) TRANSPORT_SSL: mbedtls_net_connect returned -52
E (3127) MQTT_CLIENT: Error transport connect
I (3127) MQTT_CLIENT: Reconnect after 10000 ms
I (3137) MQTTS_SAMPLE: MQTT_EVENT_DISCONNECTED

What can be the possible reason?

tuanpmt, 2018-05-04:
Could you please help me test the lastest commit from master?

Kasuhikdas, 2018-05-05:
sure will test and let you know.

david-cermak, 2019-03-28:
Hi @Kasuhikdas 

Any update?

Kasuhikdas, 2019-03-29:
That was a certificate issue. Working absolutely fine.

closed 2019-03-29 by Kasuhikdas