espressif/esp-mqtt #43: Connection refused, Unknow reason

ogensyts, 2018-05-06:
Hi,

I am trying to use espmqtt via TCP (no SSL) on my ESP32 with a broker hosted on a Android phone. 
When I try `espmqtt `with `HBMQTT `on a Windows machine (python MQTT broker) it works fine but when I try to use it with the broker on a Android application I made it says `Connection refused, Unkow reason`. 

The MQTT broker on my Android application is made with `Xamarin Android` and `MQTTNet` as a MQTT broker. It works great with many MQTT clients, just not with espmqtt :

I tried with another MQTT client for esp32, which is Nkolban [PubSubClient](https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/PubSubClient.cpp) which works with my Android phone broker and ESP32. But I want to use espmqtt, which seems way better from my point of view.

I am using the espmqtt tcp sample from the repo. I only changed WIFI configuration and the URI.
Here is the ESP32 serial output :

```
I (5050) event: sta ip: 192.168.43.160, mask: 255.255.255.0, gw: 192.168.43.1
I (5050) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (5060) MQTT_CLIENT: Sending MQTT CONNECT message, type: 1, id: 0000
W (5080) MQTT_CLIENT: Connection refused, Unknow reason
I (5080) MQTT_CLIENT: Error MQTT Connected
I (5080) MQTT_CLIENT: Reconnect after 10000 ms
I (5090) MQTT_SAMPLE: MQTT_EVENT_DISCONNECTED

```

I tried :

- In make menuconfig ->Component config -> ESPMQTT Configurations , to disable `Enable MQTT protocol 3.1.1`

- Instead of using `URI` from `esp_mqtt_config_t`, to use `host `and other parameters to set up MQTT configuration
- Tried to use different topic names (just in case) 

It's possible it's an incompatibility from the Android Broker I am using and espmqtt, but all clients i have tested work on my Android MQTT broker very fine and I don't know why just espmqtt wouldn't work on my Android Application.

Do you have any ideas on how I could proceed or debug?

tuanpmt, 2018-05-08:
Hi @ogensyts 

Maybe MQTTNet needs clientId length longer than 10, please try option `.clientId = "esp_0123456789"`

And please provide server log

ogensyts, 2018-05-13:
Hi. 

I tried with `. client_id = "esp_0123456789"` and got the same error. 

It is bizzare, I finally used [this MQTT library for ESP32](https://github.com/256dpi/esp-mqtt) and everything worked very well with my Android Broker and other brokers like Mosquitto.  

Unfortunately `MQTTNet `(the broker library i am using for Android) does not give such server log like Mosquitto broker when i debug my application on Xamarin so I am sorry i cannot help you at this point :(

closed 2018-09-23 by ogensyts