espressif/esp-mqtt #45: Sap Leonardo (ssl:// with authentication)

inukisoft, 2018-05-12:
Hi,
Im trying connect with sap leonardo iot platform. It require 3 certificates (CA, cert, key cert). I create single file with 3 certificates (ca+cert+key), but is imposible connect.  Config client is: 

`    const esp_mqtt_client_config_t mqtt_cfg = {
        .uri = "mqtts://iotae-ekt.eu10.cp.iot.sap:8883",
        .event_handle = mqtt_event_handler,
        .cert_pem = (const char *)client_pem_start,
		.client_id = (const char *)client_id
    };`

error message: 

`I (101040) event: sta ip: 192.168.43.47, mask: 255.255.255.0, gw: 192.168.43.1
I (101040) MQTTS_SAMPLE: [APP] Free memory: 237468 bytes
I (103320) wifi: pm start, type:0

I (104260) MQTT_CLIENT: Sending MQTT CONNECT message, type: 1, id: 0000
W (104570) MQTT_CLIENT: Connection refused, server unavailable
I (104570) MQTT_CLIENT: Error MQTT Connected
I (104580) MQTT_CLIENT: Reconnect after 10000 ms
I (104580) MQTTS_SAMPLE: MQTT_EVENT_DISCONNECTED
I (127450) MQTT_CLIENT: Sending MQTT CONNECT message, type: 1, id: 0000
W (127770) MQTT_CLIENT: Connection refused, server unavailable
I (127770) MQTT_CLIENT: Error MQTT Connected
I (127770) MQTT_CLIENT: Reconnect after 10000 ms
I (127770) MQTTS_SAMPLE: MQTT_EVENT_DISCONNECTED
`
t
I tried similar code with python paho without problems. The single cert file is included with "COMPONENT_EMBED_TXTFILES"

Please, could you help me?

inukisoft, 2018-05-12:
In paho python, for example, next code work for me:

`my_ca_cert ="D:/dev/mqtt/001/python/1/DigiCertGlobalRootCA.pem"
my_key_cert = "D:/dev/mqtt/001/python/1/client-nuevo.private.pem"
my_pri_cert = "D:/dev/mqtt/001/python/1/client-nuevo_chain.pem"

client.tls_set(ca_certs=my_ca_cert, 
	certfile=my_pri_cert, 
	keyfile=my_key_cert,
	cert_reqs=ssl.CERT_REQUIRED,
	tls_version=ssl.PROTOCOL_TLSv1_2, 
	ciphers=None)

client.tls_insecure_set(True)
client.connect("iotae-ekt.eu10.cp.iot.sap", 8883, 60)`

david-cermak, 2019-03-28:
Hi @inukisoft 

Client certificate is now supported from https://github.com/espressif/esp-mqtt/commit/d2bcdd84a1938f2e9c191cc6bf9b915b8dbf8c9d
Please see an example here https://github.com/espressif/esp-idf/tree/master/examples/protocols/mqtt/ssl_mutual_auth 

Closing the issue as resolved, but please reopen if it doesn't work on your end.

closed 2019-03-28 by david-cermak