espressif/esp-mqtt #46: mqtt over ssl had use cafile?and memory all use.

Shaco-Ma, 2018-05-14:
When I open -> Component config -> mbedTLS Enable mbedTLS debugging and the debug is  "W (3923) mbedtls: ssl_tls.c:4587 x509_verify_cert() returned -9984 (-0x2700)"?
when I use openssl s_client -showcerts -connect (My server addr):8883 </dev/null 2>/dev/null|openssl x509 -outform PEM >iot_eclipse_org.pem,get the pem is same to my server certfile , BUT I will get x509_verify_cert wrong returned -9984, But when I use the cafile same with server,It is will be ok,and just mqtt_ssl whrn connected will use about 50k  memory.
@tuanpmt , Can u explain why I must use cafile on mqtt over ssl to connect my server?
And mqtt over ssl use 50k memory is correct?
Best regar to you.

donny681, 2018-05-14:
I meet the same problem,There is something wrong with ssl.

Shaco-Ma, 2018-05-15:
@donny681 I use cafile in mqtt over ssl is OK,but use "openssl s_client -showcerts -connect (My server addr):8883 </dev/null 2>/dev/null|openssl x509 -outform PEM >iot_eclipse_org.pem" get the crtfile can't connect.

donny681, 2018-05-15:
Yes,I think that because the decrypt method  of the ssl is error.

borch84, 2018-05-21:
Hi @Shaco-Ma, I was able to connect to my ssl broker. I hardwired the contents of the PEM certificate to a fixed variable in my program. I modified mqtt config values to:

    const esp_mqtt_client_config_t mqtt_cfg = {
        .uri = "mqtts://mybrokerip:8883", //borch
        .event_handle = mqtt_event_handler,
        //.cert_pem = (const char *)iot_eclipse_org_pem_start,
        .cert_pem = raspberrypi_pem, //borch
        .username = "USER", //borch
        .password = "PASS, //borch
        .client_id = "esp32_tuanpmt_mqtt_ssl_client", //borch
        .transport = MQTT_TRANSPORT_OVER_SSL, //borch
    };

I will take a look on how to get program's memory utilization in order to compare SSL vs nonSSL.

Thank you.

Shaco-Ma, 2018-05-23:
@borch84 ,you maybe need this:[https://github.com/tuanpmt/espmqtt/issues/25](url),use esp_get_free_heap_size(),looks like printf heap size  before mqtt all init ,and when it connected, printf heap size, and so on.

borch84, 2018-05-24:
Hi @Shaco-Ma  I checked the previous url provided. I included the memory verification functions inside my code. I reduced the delay time to 1s. I can see that at the beginning of the execution of the program the initial free memory was around 153000 bytes. 10 minutes later the free memory was 150000 bytes, around 3000 bytes less when it was started. 
[memory-leak-evidence.txt](https://github.com/tuanpmt/espmqtt/files/2036481/memory-leak-evidence.txt)
I haven't applied the commit [a5ef145](https://github.com/tuanpmt/espmqtt/commit/a5ef145163ff202fd0519d7142da8d5e2656b7d6) yet.

borch84, 2018-05-24:
When the program started free memory is around 150000 bytes. 
Program start time: Thu May 24 11:46:33 CST 2018

When I stopped the execution of the program, the free memory was around 67000 bytes. 
Program stop time: Thu May 24 16:27:28 CST 2018

In 5 hours there was a memory leak of 83000 bytes. 

This is the code I am using for testing:

`    while (1) {
        esp_mqtt_client_start(mqtt_client);
        ESP_LOGI(MQTT_TAG, "mqtt client started, free memory: %d bytes", esp_get_free_heap_size());
        vTaskDelay(1000/portTICK_RATE_MS);
        esp_mqtt_client_stop(mqtt_client);
        ESP_LOGI(MQTT_TAG, "mqtt client stopped, free memory: %d bytes", esp_get_free_heap_size());
        vTaskDelay(1000/portTICK_RATE_MS);
    }`

Shaco-Ma, 2018-05-31:
@borch84 sorry I long time responed.
maybe you should use the newest code ,I think the newest code sill be ok.
now I can't use it to creat two clients, I fell sad...

closed 2018-05-31 by Shaco-Ma