espressif/esp-mqtt #48: Error write data or timeout

casmeiron, 2018-05-21:
Hi,

I've been using quite a while this library but after few git pull updates, it stopped working and I'm getting the follow message:

```
E (151464) MQTT_CLIENT: Error write data or timeout, written len = -1
E (151464) MQTT_CLIENT: Error to public data to topic=/tracker/0001/, qos=0
E (151474) MQTT_CLIENT: Error write data or timeout, written len = -1
E (151474) MQTT_CLIENT: Error to public data to topic=/tracker/0001/, qos=0
E (151494) MQTT_CLIENT: Error write data or timeout, written len = -1
E (151494) MQTT_CLIENT: Error to public data to topic=/tracker/0001/, qos=0

```
The client connects successfully to MQTT Server:

```
D (16414) MQTT_CLIENT: MQTT client_id=tracker
D (16414) TRANS_TCP: [sock=54],connecting to server IP:192.168.1.92,Port:1883...
D (16784) MQTT_CLIENT: Transport connected to mqtt://192.168.1.92:1883
I (16794) MQTT_CLIENT: Sending MQTT CONNECT message, type: 1, id: 0000
D (16804) MQTT_CLIENT: Connected
I (16804) MY_LOGGER: MQTT_EVENT_CONNECTED

```
I've tried with qos=1,2,3 and the same error happened (output differ a little bit 'cause 0 we don't enqueue the message).

I'm using username and password, at port 1883, disable_clean_session=true.

Using mosquitto_pub I can easily publish messages:

`mosquitto_pub -h <host> -u <username> -P <password> -t '/tracker/0001/' -m "{}"`

The same credentials and configs are used for mqtt_client (esp32) and mosquitto_pub.

What can I do to try to figure out what's happening?

Thanks in advance.

klotzt, 2018-05-23:
same issue for me, subscribe works but publish results in the exact same error message

btotaro-olibra, 2018-05-24:
I've found messages exceeding a certain size fail. and i've found large messsages that do transmit, sometimes cause the ESP32 to break wifi connectivity (dissociate, and other terrible things)

b0m0x, 2018-05-27:
This is probably related to espressif/esp-idf#1987
The proposed workaround is to set CONFIG_USE_ONLY_LWIP_SELECT.

casmeiron, 2018-05-30:
Hi @b0m0x ,

I can confirm that setting CONFIG_USE_ONLY_LWIP_SELECT to true fixes the issue.

Follow my new output after changing that flag:
```

I (92095) LOG: MQTT_EVENT_CONNECTED
I (92095) wifi: ic_enable_sniffer
D (93095) MQTT_CLIENT: Sent PING successful
D (93145) MQTT_CLIENT: msg_type=13, msg_id=0
D (93145) MQTT_CLIENT: MQTT_MSG_TYPE_PINGRESP
D (96535) MQTT_CLIENT: mqtt_enqueue id: 0, type=1 successful
D (96545) MQTT_CLIENT: mqtt_enqueue id: 48938, type=3 successful
D (96545) OUTBOX: ENQUEUE msgid=48938, msg_type=3, len=133, size=133
D (96565) MQTT_CLIENT: msg_type=4, msg_id=48938
D (96565) MQTT_CLIENT: pending_id=55860, pending_msg_count = 2
D (96565) OUTBOX: DELETED msgid=48938, msg_type=3, remain size=0
D (96565) MQTT_CLIENT: received MQTT_MSG_TYPE_PUBACK, finish QoS1 publish
I (96575) LOG: MQTT_EVENT_PUBLISHED, msg_id=48938
D (96585) MQTT_CLIENT: msg_type=4, msg_id=55860
D (96585) MQTT_CLIENT: pending_id=55860, pending_msg_count = 1
D (96595) MQTT_CLIENT: received MQTT_MSG_TYPE_PUBACK, finish QoS1 publish
I (96595) LOG: MQTT_EVENT_PUBLISHED, msg_id=55860

```

To do the fix, go to "make menuconfig" then "Component Setting" -> "LWIP" -> "Support LWIP socket select() only". This option was recently added.

[image]

casmeiron, 2018-05-30:
Should we close this issue or let's call it just a workaround and wait till the problem is "natively" fixed?

Thomseeen, 2019-02-07:
Using the ESP-MQTT_FOR_IDF_3.1 release I still get that error on publishing data with over 17kB while having `CONFIG_USE_ONLY_LWIP_SELECT` set to `true`. Having lwip debug enabled my log looks like this when i try to publish a 17kB buffer:
```
tcp_write(pcb=0x3ffdf778, data=0x3ffe436c, len=5744, apiflags=3)
tcp_write: queueing 6536:7972
tcp_write: queueing 7972:9408
tcp_write: queueing 9408:10844
tcp_write: queueing 10844:12280
tcp_output_segment: 6536:7972
tcp_output_segment: 7972:9408
tcp_output_segment: 9408:10844
tcp_output_segment: 10844:12280
+-+-+-+-+-+-+-+-+-+-+-+-+-+- tcp_input: flags -+-+-+-+-+-+-+-+-+-+-+-+-+-+
tcp_process: Connection RESET
E (4691) MQTT_CLIENT: Error write data or timeout, written len = -1
E (4691) MQTT_CLIENT: Read error or end of stream
E (4701) MQTT_CLIENT: Error to public data to topic=/topic/qos0, qos=0
```
For some reason, looking at Wireshark, the first TCP-package gets followed by a response from mosquitto with a ACK and RST flag. Following that the other three packages shown above still reach my computer but get ignored.

david-cermak, 2019-03-28:
Hi all,

Can we close this older issue? It looks like this was resolved in IDF, at least I cannot reproduce any connection problems related to `CONFIG_USE_ONLY_LWIP_SELECT` enabled or disabled

david-cermak, 2020-01-09:
Closing as no recent feedback. Please reopen if needed.

closed 2020-01-09 by david-cermak