espressif/esp-mqtt #50: SYSTEM_EVENT_ETH_DISCONNECTED does not work

bengiexd, 2018-05-28:
the "SYSTEM_EVENT_ETH_DISCONNECTED" event does not work, after updating to the current version of the esp-idf and the espmqtt. apparently the 10-second timeout does not work.

juanpavz, 2018-05-29:
Hi everyone, I have the same problem with the latest version.
Thank you for your help

casmeiron, 2018-05-30:
I just tried here and I've got the notification "MQTT_EVENT_DISCONNECTED" when my broker went down. Is that what you're referring to?

```
D (28694) MQTT_CLIENT: mqtt_enqueue id: 29690, type=3 successful
D (28714) MQTT_CLIENT: msg_type=4, msg_id=39347
D (28714) MQTT_CLIENT: pending_id=39347, pending_msg_count = 1
D (28714) MQTT_CLIENT: received MQTT_MSG_TYPE_PUBACK, finish QoS1 publish
I (28724) LOG: MQTT_EVENT_PUBLISHED, msg_id=39347
E (28954) MQTT_CLIENT: Read error or end of stream
I (28954) MQTT_CLIENT: Reconnect after 10000 ms
I (28954) LOG: MQTT_EVENT_DISCONNECTED  <<<<< this is my log
```

Snippet of my code:

```

static esp_err_t mqtt_event_handler(esp_mqtt_event_handle_t event) {
    switch (event->event_id) {
        case MQTT_EVENT_CONNECTED:
            client = event->client;
            ESP_LOGI(LOG, "MQTT_EVENT_CONNECTED");

            break;

        case MQTT_EVENT_DISCONNECTED:
            ESP_LOGI(LOG, "MQTT_EVENT_DISCONNECTED");
            break;

        case MQTT_EVENT_SUBSCRIBED:
            ESP_LOGI(LOG, "MQTT_EVENT_SUBSCRIBED, msg_id=%d", event->msg_id);
            break;

        case MQTT_EVENT_UNSUBSCRIBED:
            ESP_LOGI(LOG, "MQTT_EVENT_UNSUBSCRIBED, msg_id=%d", event->msg_id);
            break;

        case MQTT_EVENT_PUBLISHED:
            ESP_LOGI(LOG, "MQTT_EVENT_PUBLISHED, msg_id=%d", event->msg_id);
            break;

        case MQTT_EVENT_DATA:
            ESP_LOGI(LOG, "MQTT_EVENT_DATA");
            printf("TOPIC=%.*s\r\n", event->topic_len, event->topic);
            printf("DATA=%.*s\r\n", event->data_len, event->data);
            break;

        case MQTT_EVENT_ERROR:
            ESP_LOGI(LOG, "MQTT_EVENT_ERROR");
            break;
    }

    return ESP_OK;
}

```
And the reconnection after 10s works too:

```
D (76214) MQTT_CLIENT: Reconnecting...
D (81214) TRANS_TCP: [sock=56],connecting to server IP:192.168.1.92,Port:1883...
E (98774) MQTT_CLIENT: Error transport connect
I (98774) MQTT_CLIENT: Reconnect after 10000 ms
I (98774) LOG: MQTT_EVENT_DISCONNECTED
```

The only thing that I noticed is that even when the client is disconnected, after a fail attempt, it will rise MQTT_EVENT_DISCONNECTED again.

And when my broker was started again, my MQTT client automatically connected to it:

```
D (263834) MQTT_CLIENT: Reconnecting...
D (268834) TRANS_TCP: [sock=61],connecting to server IP:192.168.1.92,Port:1883...
D (268834) MQTT_CLIENT: Transport connected to mqtt://192.168.1.92:1883
I (268844) MQTT_CLIENT: Sending MQTT CONNECT message, type: 1, id: 0000
D (268854) MQTT_CLIENT: Connected
I (268854) LOG: MQTT_EVENT_CONNECTED
I (268854) wifi: ic_enable_sniffer
D (269854) MQTT_CLIENT: Sent PING successful
D (270214) MQTT_CLIENT: msg_type=13, msg_id=0
D (270214) MQTT_CLIENT: MQTT_MSG_TYPE_PINGRESP
D (271674) MQTT_CLIENT: mqtt_enqueue id: 0, type=1 successful
D (271684) MQTT_CLIENT: mqtt_enqueue id: 29589, type=3 successful
D (271684) OUTBOX: ENQUEUE msgid=29589, msg_type=3, len=133, size=133
D (272054) MQTT_CLIENT: msg_type=4, msg_id=29589
D (272054) MQTT_CLIENT: pending_id=56570, pending_msg_count = 2
D (272054) OUTBOX: DELETED msgid=29589, msg_type=3, remain size=0
D (272064) MQTT_CLIENT: received MQTT_MSG_TYPE_PUBACK, finish QoS1 publish
I (272064) LOG: MQTT_EVENT_PUBLISHED, msg_id=29589
D (272074) MQTT_CLIENT: msg_type=4, msg_id=56570
D (272074) MQTT_CLIENT: pending_id=56570, pending_msg_count = 1
D (272084) MQTT_CLIENT: received MQTT_MSG_TYPE_PUBACK, finish QoS1 publish
I (272094) LOG: MQTT_EVENT_PUBLISHED, msg_id=56570

```

bengiexd, 2018-05-31:
Hi @casmeiron 
It is correct what you mention.

I am using the ethernet interface, when I disconnect the interface, the mqtt client does not launch the MQTT_EVENT_DISCONNECTED event, but in the previous version after disconnecting the interface the event was launched after the 10 second timeout.

Everything else is correct.

thanks for your answer.

david-cermak, 2019-03-28:
Hi all,
Can we close this old issue? As far as I can see `MQTT_EVENT_DISCONNECTED` event works as expected (will yet test it with ethernet, but don't expect any issue there)

david-cermak, 2019-03-29:
@bengiexd just tested on ethernet and `MQTT_EVENT_DISCONNECTED` event received correctly. Please note that you can configure keep alive timeout (by default 120s) to any value of your preference:
```
    esp_mqtt_client_config_t mqtt_cfg = {
            .uri = "mqtt://192.168.0.111",
            .event_handle = mqtt_event_handler,
            .keepalive = 10,
            // .user_context = (void *)your_context
    };
```
I'm taking the liberty to close the issue, please reopen if needed.

closed 2019-03-29 by david-cermak

CihanAytacoglu495, 2022-09-08:
maybe tihs problem only use ethetnet interface