espressif/esp-mqtt #52: Can't Multiple client in one application

Shaco-Ma, 2018-05-30:
I get the newest code and Clone this component to ESP-IDF project (as submodule).
Then I esp_mqtt_client_init(cfg1), esp_mqtt_client_start(clinet1), then I esp_mqtt_client_init(cfg2), esp_mqtt_client_start(clinet2), cfg1 and cfg2 's event is diff , client_id and uasername di diff but always tell me : 
I (5167) MQTT_CLIENT: Sending MQTT CONNECT message, type: 1, id: 0000
E (5167) MQTT_CLIENT: Writing failed, errno= 4
I (5177) MQTT_CLIENT: Error MQTT Connected
I (5177) MQTT_CLIENT: Reconnect after 10000 ms

How can I can use Multiple client in one application?Can U give a example for Multiple client?

casmeiron, 2018-05-30:
Hi @Shaco-Ma ,

I think you can distinguish them using the "user_context" attribute in the configuration, see the docs:

**user_context** pass user context to this option, then can receive that context in event->user_context

at https://github.com/tuanpmt/espmqtt

So you can check which client is responsible for that event:

`void * client_context = event->user_context;`

So to make it work, init all the clients (using different configurations), save the returned references to them, call start for each one and control in the event_handler (using user_context) which client triggered it.

I hope this helps you.

Shaco-Ma, 2018-05-31:
@casmeiron I finish it, If use esp-idf,must open make menuconfig  --->Component config  --->LWIP  --->[*] Support LWIP socket select() only,Thanks U advice.
@tuanpmt ,I think maybe U should write the make menuconfig about espmqtt when use esp-idf and use 
[https://github.com/tuanpmt/espmqtt#how-to-use](url). maybe or not.

closed 2018-05-31 by Shaco-Ma

tuanpmt, 2018-05-31:
Thank you @Shaco-Ma and @casmeiron 

I will add `CONFIG_USE_ONLY_LWIP_SELECT` = true as default

Shaco-Ma, 2018-05-31:
@tuanpmt And I found some bug in esp_mqtt_task, if esp_mqtt_connect fail, will goto MQTT_STATE_WAIT_TIMEOUT,if client->config->auto_reconnect is open, will transport_connect,have no transport_close, you see ,maybe it is a bug in esp_mqtt_task.