espressif/esp-mqtt #57: memory leak when initializing SSL

agmuntianu, 2018-06-28:
Hi,

I tried out the newest version and I noticed a memory leak :( . the memory leak is when calling `mbedtls_ssl_set_hostname` in `ssl_connect` function. 
After I narrowed down on the problem, I found this issue on ARMmbed/mbedtls :
https://github.com/ARMmbed/mbedtls/issues/836.

I am not sure why we call that function , it worked just fine without it before , for my code I just commented out the code and all is good.

BR. Andrei

david-cermak, 2018-10-04:
Hi Andrei,

`transport_ssl.c` was refactored and moved to a shared component tcp_transport in IDF. it no longer calls `mbedtls_ssl_set_hostname`. The related update of `esp-mqtt` is currently on an `idf` branch, but will be merged to `master` once IDFv3.2 is released.

Could you please help closing this issue if it solves you problem?

Thank you,
David

agmuntianu, 2018-10-04:
Yes , thanks

closed 2018-10-04 by agmuntianu