espressif/esp-mqtt #63: Compatibility with thingsboard.io

tuupola, 2018-08-20:
I am unable to submit data to `mqtt://demo.thingsboard.io`.  After calling `esp_mqtt_client_publish()` I get the following error: 

```
E (24381) MQTT_CLIENT: Read error or end of stream
```

If submitting to `mqtt://iot.eclipse.org` everything works as expected. Also tested also with the [examples/mqtt_tcp](https://github.com/tuanpmt/espmqtt/tree/master/examples/mqtt_tcp) app. Behaviour is the same. Eclipse works while Thingsboard not.  Tested with both 3.1.1 protocol enabled and disabled. MQTT packets when inspected in Wireshark look the same.

Currently running out of ideas. Are there any known issues with Thingsboard or configuration changes I should try?

```
I (2171) event: sta ip: 192.168.8.106, mask: 255.255.255.0, gw: 192.168.8.1
I (2171) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (2461) MQTT_CLIENT: Sending MQTT CONNECT message, type: 1, id: 0000
I (2671) MQTT_SAMPLE: MQTT_EVENT_CONNECTED
I (2671) MQTT_SAMPLE: sent subscribe successful, msg_id=28741
I (2671) MQTT_SAMPLE: sent subscribe successful, msg_id=11408
I (2681) MQTT_SAMPLE: sent unsubscribe successful, msg_id=6495
I (2871) MQTT_SAMPLE: MQTT_EVENT_SUBSCRIBED, msg_id=28741
I (2871) MQTT_SAMPLE: sent publish successful, msg_id=0
I (3081) MQTT_SAMPLE: MQTT_EVENT_SUBSCRIBED, msg_id=11408
I (3081) MQTT_SAMPLE: sent publish successful, msg_id=0
I (3081) MQTT_SAMPLE: MQTT_EVENT_UNSUBSCRIBED, msg_id=6495
E (3281) MQTT_CLIENT: Read error or end of stream
I (3291) MQTT_CLIENT: Reconnect after 10000 ms
I (3291) MQTT_SAMPLE: MQTT_EVENT_DISCONNECTED
```

tuanpmt, 2018-08-20:
Are you subscribe or public on `v1/devices/me/telemetry` channel and using `ACCESS_TOKEN` as username? If not, please check here: https://thingsboard.io/docs/reference/mqtt-api/#mqtt-basics

tuupola, 2018-08-20:
It seems I was overthinking the  problem.  I was subscribed to `v1/devices/me/telemetry` but submitting to `/v1/devices/me/telemetry`. Thanks!

closed 2018-08-20 by tuupola