espressif/esp-mqtt #64: esp-mqtt subscribe and retained messages

PaulVdBergh, 2018-08-25:
Please have a look at [my question at esp32.com](https://esp32.com/viewtopic.php?f=22&t=6909)

Thanks, 
Paul.

david-cermak, 2018-10-04:
Hi Paul,

related to the line `if (client->mqtt_state.message_length_read >= client->mqtt_state.message_length)`: 

* it is supposed to process only related part of the data, further data will be processed in the next iteration of publish. 
* Some refactoring were made to this part of the code but at this moment it  resides on an `idf` branch, but will be merged to master once IDFv3.2 is released.

Can you please check if your issue persists when switching to the latest IDF? (no need to add esp-mqtt)?

Thank you,
David

PaulBreugnot, 2018-12-27:
Hi everybody,

I have the same issue, even with IDFv3.3.
Maybe the difference is that I subscribe calling the `esp_mqtt_client_subscribe` function individually for each topic, so, as a workaround, what I do is that I use [`esp_mqtt_client_config_t`'s `user_context`](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/protocols/mqtt.html#_CPPv2N24esp_mqtt_client_config_t12user_contextE)  with boolean variables, and, basically : 
- I subscribe to the first topic.
- I do nothing until the first retained message is received.
- I subscribe to the second topic.
- I receive the second retained message.

It requires more code and I have to handle the case when there is no retained message in the first topic (otherwise, I will never subscribe to the second topic), but at least it works. ;-)

Paul

david-cermak, 2019-03-28:
Resolved in these commits:
https://github.com/espressif/esp-mqtt/commit/db64b791205c8dbcd0f3826a9544f0aa0301baaf
https://github.com/espressif/esp-mqtt/commit/cf5b8eda89057f5eb49654d2eaf4d1a209af2d11
https://github.com/espressif/esp-mqtt/commit/d4b66556186141e09db89c57781c6709e0e71d8a

commit db64b791 (2019-01-03): Fix for case where multiple MQTT messages are fitted into a single TCP packet : iterate over the buffer received.

Merges https://github.com/espressif/esp-mqtt/pull/82
Closes #64
Closes #80
closed 2019-03-28 by david-cermak