espressif/esp-mqtt #67: This library is not thread-safe and it that should be clear in the docs

casmeiron, 2018-08-30:
Hi,

I was getting error in my MQTT Broker and then I realize that was happening 'cause packets of MQTT are getting mixed together because this library isn't thread-safe and I was using 3 tasks to publish at the same time messages to MQTT. Adding a semaphore fixed the problem. The docs should be clear about that or the thread-safety procedure should be implemented.

ivanroberto, 2018-12-11:
Hello,
In my case a simple semaphore does not solve, because I have an application where everything happens asynchronously. I have an indexed list where both inbound and outbound message messages are stored. Then there is a task that will process the whole list.
So far so good, but my problem starts when my task is publishing something and at the same time a message is being sent by the broker.

And this in my case is very common, because every message I receive by the input topic I publish an ACK through the output topic ... And I still have spontaneous messages coming from keys on the device that make things a little more intense.

I really liked this lib, it is quite complete, but it would be very important if it was thread safe.

Note: This problem is easily observed in QOS1 and 2, with QOS 0 doing much less.

```
This is the error log:
I (17526) MQTT_CLIENT: Sending MQTT CONNECT message, type: 1, id: 0000
I (17526) PROCESS_LIST: MQTT_EVENT_CONNECTED
I (17536) PROCESS_LIST: sent subscribe successful, msg_id=63636
I (17536) PROCESS_LIST: MQTT_EVENT_SUBSCRIBED, msg_id=63636
I (17536) PROCESS_LIST: sent publish successful, msg_id=652
I (42586) MQTT_CLIENT: deliver_publish, message_length_read=1511, message_length=1511
I (42586) PROCESS_LIST: Receiving MSG (idMsg: 1, dataLen: 1472  totalDataLen: 1472  offset: 0)!!!
I (42596) PROCESS_LIST: MSG add on list - OK
I (42596) PROCESS_LIST: ---> Free heap memory: 167312 bytes
I (42796) MQTT_CLIENT: deliver_publish, message_length_read=1511, message_length=1511
I (42796) PROCESS_LIST: Receiving MSG (idMsg: 2, dataLen: 1472  totalDataLen: 1472  offset: 0)!!!
I (42806) PROCESS_LIST: MSG add on list - OK
I (42806) PROCESS_LIST: ---> Free heap memory: 165680 bytes
I (42826) MQTT_CLIENT: deliver_publish, message_length_read=1511, message_length=1511
I (42826) PROCESS_LIST: Receiving MSG (idMsg: 3, dataLen: 1472  totalDataLen: 1472  offset: 0)!!!
I (42836) PROCESS_LIST: MSG add on list - OK
I (42846) PROCESS_LIST: ---> Free heap memory: 164148 bytes
I (42916) PROCESS_LIST: Publishing msg... Size: 103 bytes
I (42916) PROCESS_LIST: Publishing msg... Size: 102 bytes
I (42916) PROCESS_LIST: Publishing msg... Size: 102 bytes
I (42916) PROCESS_LIST: Publishing msg... Size: 102 bytes
I (42926) PROCESS_LIST: Publishing msg... Size: 102 bytes
I (42936) PROCESS_LIST: Publishing msg... Size: 102 bytes
I (42936) PROCESS_LIST: Publishing msg... Size: 102 bytes
I (42946) PROCESS_LIST: Publishing msg... Size: 102 bytes
I (42946) PROCESS_LIST: Publishing msg... Size: 102 bytes
I (43076) PROCESS_LIST: Publishing msg... Size: 103 bytes
I (43076) PROCESS_LIST: Publishing msg... Size: 102 bytes
I (43086) PROCESS_LIST: Publishing msg... Size: 102 bytes
I (43086) PROCESS_LIST: Publishing msg... Size: 102 bytes
I (43086) PROCESS_LIST: Publishing msg... Size: 102 bytes
I (43096) PROCESS_LIST: Publishing msg... Size: 102 bytes
I (43106) PROCESS_LIST: Publishing msg... Size: 102 bytes
E (43106) MQTT_CLIENT: Error write data or timeout, written len = -1
E (43106) MQTT_CLIENT: Read error or end of stream
E (43116) MQTT_CLIENT: Error to public data to topic=/dispositivo/terminalApuracaoIn, qos=2
I (43126) MQTT_CLIENT: Reconnect after 10000 ms
I (43136) PROCESS_LIST: Publishing msg... Size: 102 bytes
E (43136) MQTT_CLIENT: Client has not connected
E (43146) PROCESS_LIST: MQTT_EVENT_DISCONNECTED
I (43146) PROCESS_LIST: Publishing msg... Size: 102 bytes
E (43156) MQTT_CLIENT: Client has not connected
I (43266) PROCESS_LIST: Publishing msg... Size: 103 bytes
E (43266) MQTT_CLIENT: Client has not connected
I (43266) PROCESS_LIST: Publishing msg... Size: 102 bytes
E (43266) MQTT_CLIENT: Client has not connected
I (43276) PROCESS_LIST: Publishing msg... Size: 102 bytes
E (43276) MQTT_CLIENT: Client has not connected
I (43286) PROCESS_LIST: Publishing msg... Size: 102 bytes
E (43286) MQTT_CLIENT: Client has not connected
I (43296) PROCESS_LIST: Publishing msg... Size: 102 bytes
E (43306) MQTT_CLIENT: Client has not connected
I (43306) PROCESS_LIST: Publishing msg... Size: 102 bytes
E (43316) MQTT_CLIENT: Client has not connected
I (43316) PROCESS_LIST: Publishing msg... Size: 102 bytes
E (43326) MQTT_CLIENT: Client has not connected
I (43326) PROCESS_LIST: Publishing msg... Size: 102 bytes
E (43336) MQTT_CLIENT: Client has not connected
I (43336) PROCESS_LIST: Publishing msg... Size: 102 bytes
E (43346) MQTT_CLIENT: Client has not connected
I (58146) MQTT_CLIENT: Sending MQTT CONNECT message, type: 1, id: 0000
I (58156) PROCESS_LIST: MQTT_EVENT_CONNECTED
I (58156) PROCESS_LIST: sent subscribe successful, msg_id=41652
I (58156) PROCESS_LIST: MQTT_EVENT_SUBSCRIBED, msg_id=41652
I (58166) PROCESS_LIST: sent publish successful, msg_id=15253
```

ivanroberto, 2018-12-11:
This is a verbose log:
```
D (11728) MQTT_CLIENT: Reconnecting...
D (16728) MQTT_CLIENT: Transport connected to mqtt://10.42.0.1:1883
I (16728) MQTT_CLIENT: Sending MQTT CONNECT message, type: 1, id: 0000
D (16748) MQTT_CLIENT: Connected
I (16748) PROCESS_LIST: MQTT_EVENT_CONNECTED
D (16748) MQTT_CLIENT: mqtt_enqueue id: 0, type=1 successful
D (16748) MQTT_CLIENT: Sent subscribe topic=/dispositivo/terminalApuracaoOut, id: 14704, type=8 successful
I (16758) PROCESS_LIST: sent subscribe successful, msg_id=14704
D (16768) MQTT_CLIENT: msg_type=9, msg_id=14704
D (16768) MQTT_CLIENT: pending_id=14704, pending_msg_count = 1
D (16778) MQTT_CLIENT: Subscribe successful
I (16778) PROCESS_LIST: MQTT_EVENT_SUBSCRIBED, msg_id=14704
D (16788) MQTT_CLIENT: mqtt_enqueue id: 14704, type=8 successful
I (16798) PROCESS_LIST: sent publish successful, msg_id=22495
D (16798) MQTT_CLIENT: msg_type=5, msg_id=22495
D (16798) MQTT_CLIENT: received MQTT_MSG_TYPE_PUBREC
D (17058) MQTT_CLIENT: msg_type=7, msg_id=22495
D (17068) MQTT_CLIENT: received MQTT_MSG_TYPE_PUBCOMP
D (17068) MQTT_CLIENT: pending_id=22495, pending_msg_count = 1
D (17068) MQTT_CLIENT: Receive MQTT_MSG_TYPE_PUBCOMP, finish QoS2 publish
D (20138) MQTT_CLIENT: msg_type=3, msg_id=1
D (20138) MQTT_CLIENT: Queue response QoS: 1
I (20138) MQTT_CLIENT: deliver_publish, message_length_read=1511, message_length=1511
D (20148) MQTT_CLIENT: Get data len= 1472, topic len=32
I (20148) PROCESS_LIST: Receiving MSG (idMsg: 1, dataLen: 1472  totalDataLen: 1472  offset: 0)!!!
I (20158) PROCESS_LIST: MSG add on list - OK
I (20168) PROCESS_LIST: ---> Free heap memory: 167244 bytes
D (20228) MQTT_CLIENT: msg_type=3, msg_id=2
D (20228) MQTT_CLIENT: Queue response QoS: 1
I (20228) MQTT_CLIENT: deliver_publish, message_length_read=1511, message_length=1511
D (20228) MQTT_CLIENT: Get data len= 1472, topic len=32
I (20238) PROCESS_LIST: Receiving MSG (idMsg: 2, dataLen: 1472  totalDataLen: 1472  offset: 0)!!!
I (20248) PROCESS_LIST: MSG add on list - OK
I (20248) PROCESS_LIST: ---> Free heap memory: 167288 bytes
I (20438) PROCESS_LIST: Publishing msg... Size: 103 bytes
D (20448) MQTT_CLIENT: mqtt_enqueue id: 22495, type=3 successful
I (20448) PROCESS_LIST: Publishing msg... Size: 102 bytes
D (20448) MQTT_CLIENT: mqtt_enqueue id: 25327, type=3 successful
D (20458) OUTBOX: ENQUEUE msgid=25327, msg_type=3, len=140, size=140
D (20458) MQTT_CLIENT: msg_type=3, msg_id=3
I (20468) PROCESS_LIST: Publishing msg... Size: 102 bytes
D (20468) MQTT_CLIENT: Queue response QoS: 1
D (20478) MQTT_CLIENT: mqtt_enqueue id: 20623, type=3 successful
D (20488) OUTBOX: ENQUEUE msgid=20623, msg_type=3, len=140, size=280
I (20488) MQTT_CLIENT: deliver_publish, message_length_read=1511, message_length=1511
I (20498) PROCESS_LIST: Publishing msg... Size: 102 bytes
D (20498) MQTT_CLIENT: Get data len= 1472, topic len=32
D (20508) MQTT_CLIENT: mqtt_enqueue id: 54652, type=3 successful
I (20518) PROCESS_LIST: Receiving MSG (idMsg: 3, dataLen: 1472  totalDataLen: 1472  offset: 0)!!!
D (20518) OUTBOX: ENQUEUE msgid=54652, msg_type=3, len=140, size=420
I (20528) PROCESS_LIST: MSG add on list - OK
I (20538) PROCESS_LIST: ---> Free heap memory: 156588 bytes
I (20538) PROCESS_LIST: Publishing msg... Size: 102 bytes
D (20548) MQTT_CLIENT: msg_type=5, msg_id=25327
D (20558) MQTT_CLIENT: mqtt_enqueue id: 25644, type=3 successful
D (20558) MQTT_CLIENT: received MQTT_MSG_TYPE_PUBREC
D (20568) OUTBOX: ENQUEUE msgid=25644, msg_type=3, len=140, size=560
D (20568) MQTT_CLIENT: msg_type=5, msg_id=20623
I (20578) PROCESS_LIST: Publishing msg... Size: 102 bytes
D (20578) MQTT_CLIENT: received MQTT_MSG_TYPE_PUBREC
D (20588) MQTT_CLIENT: mqtt_enqueue id: 35891, type=6 successful
D (20588) MQTT_CLIENT: msg_type=5, msg_id=54652
D (20598) OUTBOX: ENQUEUE msgid=35891, msg_type=6, len=4, size=564
D (20598) MQTT_CLIENT: received MQTT_MSG_TYPE_PUBREC
I (20608) PROCESS_LIST: Publishing msg... Size: 102 bytes
D (20618) MQTT_CLIENT: msg_type=5, msg_id=54652
D (20618) MQTT_CLIENT: mqtt_enqueue id: 53798, type=6 successful
D (20628) MQTT_CLIENT: received MQTT_MSG_TYPE_PUBREC
[... 31 lines cut ...]
I (20888) PROCESS_LIST: Publishing msg... Size: 102 bytes
E (20898) MQTT_CLIENT: Client has not connected
I (20898) PROCESS_LIST: Publishing msg... Size: 102 bytes
E (20908) MQTT_CLIENT: Client has not connected
I (20908) PROCESS_LIST: Publishing msg... Size: 102 bytes
E (20918) MQTT_CLIENT: Client has not connected
I (21138) PROCESS_LIST: Publishing msg... Size: 103 bytes
E (21138) MQTT_CLIENT: Client has not connected
I (21138) PROCESS_LIST: Publishing msg... Size: 102 bytes
E (21138) MQTT_CLIENT: Client has not connected
I (21148) PROCESS_LIST: Publishing msg... Size: 102 bytes
E (21148) MQTT_CLIENT: Client has not connected
I (21158) PROCESS_LIST: Publishing msg... Size: 102 bytes
E (21158) MQTT_CLIENT: Client has not connected
I (21168) PROCESS_LIST: Publishing msg... Size: 102 bytes
E (21178) MQTT_CLIENT: Client has not connected
I (21178) PROCESS_LIST: Publishing msg... Size: 102 bytes
E (21188) MQTT_CLIENT: Client has not connected
I (21188) PROCESS_LIST: Publishing msg... Size: 102 bytes
E (21198) MQTT_CLIENT: Client has not connected
I (21198) PROCESS_LIST: Publishing msg... Size: 102 bytes
E (21208) MQTT_CLIENT: Client has not connected
I (21208) PROCESS_LIST: Publishing msg... Size: 102 bytes
E (21218) MQTT_CLIENT: Client has not connected
```

muratsplat, 2019-02-04:
Hello Guys

I used the three topic, and I saw some topics was mixed when I checked the log on MQTT server. Maybe this library does not support to sending messages on concurrency.

david-cermak, 2019-03-28:
Resolved in https://github.com/espressif/esp-mqtt/commit/752953dc3be007cca4255b66a35d3087e61f6a54

commit 752953dc (2019-02-15): added mqtt api locks, so methods can be executed from user context

closes #67, closes #90, closes https://github.com/espressif/esp-idf/issues/2975
closed 2019-03-28 by david-cermak