espressif/esp-mqtt #71: MQTT_EVENT_PUBLISHED event not firing

mh00h, 2018-09-13:
First found in my own program, I have identified a bug which can be reproduced using the mqtt_tcp example. This event doesn't fire:

```bash
   58         case MQTT_EVENT_PUBLISHED:
   59             ESP_LOGI(TAG, "MQTT_EVENT_PUBLISHED, msg_id=%d", event->msg_id);
   60             break;
```

Subscribing to relevant mqtt topic shows that data packets do flow to their correct locations; it's just the event that isn't firing.

```bash
I (4170) MQTT_SAMPLE: MQTT_EVENT_SUBSCRIBED, msg_id=63326
I (4180) MQTT_SAMPLE: sent publish successful, msg_id=0
D (4190) MQTT_CLIENT: msg_type=11, msg_id=23782
D (4190) MQTT_CLIENT: pending_id=23782, pending_msg_count = 1
D (4200) MQTT_CLIENT: UnSubscribe successful
I (4200) MQTT_SAMPLE: MQTT_EVENT_UNSUBSCRIBED, msg_id=23782
```

mh00h, 2018-09-13:
Perhaps related to this is another problem I'm experiencing: 

```bash
//ran sequentially:
xEventGroupSetBits(controller, MQTT_PUBLISHED_BIT);
ESP_LOGI(TAG, "controller eventGroup bits: %x", xEventGroupGetBits( controller )); //to confirm bit got set
vTaskDelay(40); //this required to make race condition succeed
msg_id = esp_mqtt_client_publish(client, "/root", payload, 0, 0, 0);

race condition output options:
    E (2560) MQTT_CLIENT: Client has not connected
OR
    //mosquitto receives the message
```

OscarStarink, 2018-10-14:
Hi,

Have you tied sending with QoS 1 or 2, because QoS 0 is "fire and forget" and the MQTT protocol will not confirm message reception. 

Kind regards,
Oscar

mh00h, 2018-11-04:
The problem is not that I want to confirm message reception. It is that I am using freeRTOS event groups to wait until esp-mqtt has fired before continuing an action.

closed 2018-11-04 by mh00h