espressif/esp-mqtt #73: Compatible with the ESP8266 RTOS V3.X

xiongyumail, 2018-09-27:
This is a great mqtt library， there are any plan for compatible with the ESP8266 RTOS V3.X?

cgawron, 2018-10-02:
I'm just trying to fix the incompatibilities - does not look too complicated.

cgawron, 2018-10-02:
See PR #76

xiongyumail, 2018-10-04:
Thank you very much for your excellent work, I will test it and help push it into the SDK.

lrusnac, 2019-08-28:
hey, will this get merged at some point? Curious if anybody tested it

cgawron, 2019-08-28:
@lrusnac Actually the latest ESP8266 RTOS SDK contains both [esp-mqtt](https://github.com/espressif/ESP8266_RTOS_SDK/tree/master/components/mqtt/esp-mqtt) and ibm-mqtt.

Alvin1Zhang, 2019-08-29:
@cgawron Thanks for the notes. @lrusnac ESP8266 SDK already included [ESP-MQTT](https://github.com/espressif/ESP8266_RTOS_SDK/tree/master/components/mqtt/esp-mqtt). Thanks.

david-cermak, 2020-01-09:
Closing this older issue since the ESP8266 SDK includes esp-mqtt

closed 2020-01-09 by david-cermak