espressif/esp-mqtt #80: mqtt read failed when the IP package is segmented or packed

xiaocj, 2018-10-12:
in mqtt_process_receive() function, the IP segment/pack issue are not handle correctly. If a mqtt message is segmented into 2 IP packages, or 2 mqtt message are packed into one IP package, the MQTT function will failure.

Sermus, 2018-10-17:
I also face such behavior. Quite critical, i must say. The scenario i see it is like follows:
1) I subscribe for a topic
2) I send like 5 messages to this topic one after another one
3) I see only one or two from the sequence are handled by the client.
If these messages are qos0 then they are just forgotten. If they are qos1 then the handler is invoked, but it is invoked like with 5 seconds pauses in between.

The analysis:
When such a series arrives at client side, it's read into the IN buffer and then control is passed to deliver_publish. It decodes ONLY the first message from the sequence and forgets the others.

This is reproducible even with the example code the client supplied with.

david-cermak, 2019-03-28:
Resolved in https://github.com/espressif/esp-mqtt/commit/d4b66556186141e09db89c57781c6709e0e71d8a https://github.com/espressif/esp-mqtt/commit/cf5b8eda89057f5eb49654d2eaf4d1a209af2d11 https://github.com/espressif/esp-mqtt/commit/db64b791205c8dbcd0f3826a9544f0aa0301baaf

commit db64b791 (2019-01-03): Fix for case where multiple MQTT messages are fitted into a single TCP packet : iterate over the buffer received.

Merges https://github.com/espressif/esp-mqtt/pull/82
Closes #64
Closes #80
closed 2019-03-28 by david-cermak