espressif/esp-mqtt #84: Certificate error during SSL handshake.

iscapla, 2018-10-25:
Hello,
I'm trying to implement a MQTT node using the mqtt_ssl example. Using the command (as the example said):
```
openssl s_client -showcerts -connect iot.eclipse.org:8883 </dev/null 2>/dev/null|openssl x509 -outform PEM >iot_eclipse_org.pem
```
The node doesn't work, but neither with mosquitto_pub:
```
mosquitto_pub --cafile iot_eclipse_org.pem -h iot.eclipse.org -t "data" -m "2" -p 8883 --tls-version tlsv1.2
```
Any idea?

Furthermore, in my application, I will use my self-signed certificates using the script of this [link](https://github.com/owntracks/tools/blob/master/TLS/generate-CA.sh). Looks correct.

My mosquitto conf is:
```
port 8883
allow_anonymous false
password_file /etc/mosquitto/mqtt_passwd.txt
cafile /etc/mosquitto/certs/ca.crt
keyfile /etc/mosquitto/certs/raspberrypi.key
certfile /etc/mosquitto/certs/raspberrypi.crt
tls_version tlsv1.2
```
The Espressif node conf is:
```java
    const esp_mqtt_client_config_t mqtt_cfg = {
        .uri = "mqtts://192.168.135.157",
	.port = 8883,
	.event_handle = mqtt_event_handler,
        .cert_pem = (const char *)iot_eclipse_org_pem_start, //Updated with README command
	.username = "dismuntel_mqtt",
	.password = "dismuntel_pass",
    };
````

My server trace:
```
1540467565: New connection from 192.168.139.236 on port 8883.
1540467566: OpenSSL Error: error:14094412:SSL routines:ssl3_read_bytes:sslv3 alert bad certificate
1540467566: Socket error on client <unknown>, disconnecting.
```

The node trace:
```
I (2220) event: sta ip: 192.168.139.236, mask: 255.255.255.0, gw: 192.168.139.1
I (2220) MQTTS_SAMPLE: [APP] Free memory: 236712 bytes
I (2220) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (2470) mbedtls: ssl_tls.c:8021 => handshake
I (2470) mbedtls: ssl_cli.c:3405 client state: 0
I (2470) mbedtls: ssl_tls.c:2751 => flush output
I (2480) mbedtls: ssl_tls.c:2763 <= flush output
I (2480) mbedtls: ssl_cli.c:3405 client state: 1
I (2490) mbedtls: ssl_tls.c:2751 => flush output
I (2490) mbedtls: ssl_tls.c:2763 <= flush output
I (2500) mbedtls: ssl_cli.c:774 => write client hello
I (2510) mbedtls: ssl_tls.c:3180 => write handshake message
I (2510) mbedtls: ssl_tls.c:3337 => write record
I (2520) mbedtls: ssl_tls.c:2751 => flush output
I (2520) mbedtls: ssl_tls.c:2770 message length: 244, out_left: 244
I (2530) mbedtls: ssl_tls.c:2775 ssl->f_send() returned 244 (-0xffffff0c)
I (2540) mbedtls: ssl_tls.c:2803 <= flush output
I (2540) mbedtls: ssl_tls.c:3470 <= write record
I (2550) mbedtls: ssl_tls.c:3314 <= write handshake message
I (2560) mbedtls: ssl_cli.c:1106 <= write client hello
I (2560) mbedtls: ssl_cli.c:3405 client state: 2
I (2570) mbedtls: ssl_tls.c:2751 => flush output
I (2570) mbedtls: ssl_tls.c:2763 <= flush output
I (2580) mbedtls: ssl_cli.c:1499 => parse server hello
I (2580) mbedtls: ssl_tls.c:4305 => read record
I (2590) mbedtls: ssl_tls.c:2532 => fetch input
I (2590) mbedtls: ssl_tls.c:2693 in_left: 0, nb_want: 5
I (2780) mbedtls: ssl_tls.c:2717 in_left: 0, nb_want: 5
I (2780) mbedtls: ssl_tls.c:2718 ssl->f_recv(_timeout)() returned 5 (-0xfffffffb)
I (2780) mbedtls: ssl_tls.c:2738 <= fetch input
I (2780) mbedtls: ssl_tls.c:2532 => fetch input
I (2790) mbedtls: ssl_tls.c:2693 in_left: 5, nb_want: 66
I (2800) mbedtls: ssl_tls.c:2717 in_left: 5, nb_want: 66
I (2800) mbedtls: ssl_tls.c:2718 ssl->f_recv(_timeout)() returned 61 (-0xffffffc3)
I (2810) mbedtls: ssl_tls.c:2738 <= fetch input
I (2820) mbedtls: ssl_tls.c:4379 <= read record
I (2820) mbedtls: ssl_cli.c:1781 server hello, total extension length: 17
I (2830) mbedtls: ssl_cli.c:1970 <= parse server hello
I (2830) mbedtls: ssl_cli.c:3405 client state: 3
I (2840) mbedtls: ssl_tls.c:2751 => flush output
I (2850) mbedtls: ssl_tls.c:2763 <= flush output
I (2850) mbedtls: ssl_tls.c:5440 => parse certificate
I (2860) mbedtls: ssl_tls.c:4305 => read record
I (2860) mbedtls: ssl_tls.c:2532 => fetch input
I (2870) mbedtls: ssl_tls.c:2693 in_left: 0, nb_want: 5
I (2870) mbedtls: ssl_tls.c:2717 in_left: 0, nb_want: 5
I (2880) mbedtls: ssl_tls.c:2718 ssl->f_recv(_timeout)() returned 5 (-0xfffffffb)
I (2890) mbedtls: ssl_tls.c:2738 <= fetch input
I (2890) mbedtls: ssl_tls.c:2532 => fetch input
I (2900) mbedtls: ssl_tls.c:2693 in_left: 5, nb_want: 2742
I (2910) mbedtls: ssl_tls.c:2717 in_left: 5, nb_want: 2742
I (2910) mbedtls: ssl_tls.c:2718 ssl->f_recv(_timeout)() returned 2737 (-0xfffff54f)
I (2920) mbedtls: ssl_tls.c:2738 <= fetch input
I (2950) mbedtls: ssl_tls.c:4379 <= read record
W (3000) mbedtls: ssl_tls.c:5713 x509_verify_cert() returned -9984 (-0x2700)
I (3000) mbedtls: ssl_tls.c:5244 => send alert message
I (3000) mbedtls: ssl_tls.c:3337 => write record
I (3010) mbedtls: ssl_tls.c:2751 => flush output
I (3020) mbedtls: ssl_tls.c:2770 message length: 7, out_left: 7
I (3020) mbedtls: ssl_tls.c:2775 ssl->f_send() returned 7 (-0xfffffff9)
I (3030) mbedtls: ssl_tls.c:2803 <= flush output
I (3030) mbedtls: ssl_tls.c:3470 <= write record
I (3040) mbedtls: ssl_tls.c:5257 <= send alert message
I (3050) mbedtls: ssl_tls.c:5810 <= parse certificate
I (3050) mbedtls: ssl_tls.c:8031 <= handshake
E (3060) esp-tls: mbedtls_ssl_handshake returned -0x2700
I (3060) esp-tls: Failed to verify peer certificate!
I (3070) esp-tls: verification info:   ! The certificate is not correctly signed by the trusted CA
E (3080) esp-tls: Failed to open new connection
E (3080) TRANS_SSL: Failed to open a new connection
E (3090) MQTT_CLIENT: Error transport connect
I (3090) MQTT_CLIENT: Reconnect after 10000 ms
I (3100) MQTTS_SAMPLE: MQTT_EVENT_DISCONNECTED
```

zhaochengwei, 2018-10-26:
Buddy, I have the same problem. So sad

david-cermak, 2018-10-26:
Hi @iscapla 

It seems that iot.eclipse.org updated the certificate. Whilst the one currently stored in idf  is self-signed, the actual one is signed by a CA. You have to provide certificates of signing authority up to root cert in this case. To make the example running, just paste 'let's encrypt' certificate to the cert file, something like:
```
openssl s_client -showcerts -connect iot.eclipse.org:8883 </dev/null 2>/dev/null|openssl x509 -outform PEM >iot_eclipse_org.pem
curl "https://letsencrypt.org/certs/letsencryptauthorityx3.pem.txt" >> iot_eclipse_org.pem
```

The other issue you describe is something different, as you generated self-signed certificate of your local mosquito. I cannot say what is wrong, but would guess that your generated certificate has wrong CN if you reference that by IP address in your code.

david-cermak, 2018-10-26:
Hi @zhaochengwei , of course it's possible to connect without any certificate, just remove the config in example:
```
    const esp_mqtt_client_config_t mqtt_cfg = {
        .uri = "mqtts://iot.eclipse.org:8883",
        .event_handle = mqtt_event_handler,
        // .cert_pem = (const char *)iot_eclipse_org_pem_start,
    };
```
This connects (hopefully also in your environment) to server correctly, but doesn't protect from MITM. 

What I'm suggesting above is not 100% correct either, as the esp-tls client needs to know only the trusted root certificates (the same way openssl has `ca-file` or `ca-path`), so the correct way is setup `esp_tls_set_global_ca_store` with trusted root certificates. Also in mqtt-ssl example, it should be just enough to store let's encrypt root self signed cert only, like:
```
curl "https://letsencrypt.org/certs/letsencryptauthorityx3.pem.txt" > iot_eclipse_org.pem
```
bacause, as you correctly say, the client can get the server certificate (but needs yet a little more info  in order to trust it)

iscapla, 2018-10-30:
Problem solved using this guide: [Steves's guide](http://www.steves-internet-guide.com/mosquitto-tls/).
Things to keep in mind:
- CA key can be protected with a pass phrase. Server don't.
- Different field names (yellow note in the guide).
- I use the mosquitto server IP as CN. This is not recommended but allowed. See this [link](https://stackoverflow.com/questions/2043617/is-it-possible-to-have-ssl-certificate-for-ip-address-not-domain-name).
- Convert ca.crt and server.crt to PEM format with this:
```bash
openssl x509 -in $file.crt -out $file.pem -outform PEM
```
- iot_eclipse_org.pem file must include server.pem file + ca.pem

closed 2018-10-30 by iscapla

SERIDJ, 2020-03-04:
hi,

i want to do the same thing, so  without certificat its work  well :D  , and when i try to use ssl connection, add 3 certificat in my raspberry broker (ca.cert server.crt and server.key)  and enable  require_certificate, that   not work :( 

in esp32 code  "esp_mqtt_client_config_t "(eclipse using esp idf )  if i add or not certificat i have the same result in terminal  : 

I (692579) MQTT_CLIENT: Error MQTT Connected
I (692579) MQTTS_EXAMPLE: MQTT_EVENT_DISCONNECTED
I (702589) MQTTS_EXAMPLE: Other event id:7
I (702589) MQTT_CLIENT: Sending MQTT CONNECT message, type: 1, id: 0000
E (702599) MQTT_CLIENT: mqtt_message_receive: transport_read() error: errno=104
E (702599) MQTT_CLIENT: esp_mqtt_connect: mqtt_message_receive() returned -1

that mean its connected with broker but cant send and receive data 

so eclipes or esp32 code  doesn't use those certificate.