espressif/esp-mqtt #85: Can't  connect to broker when both  ap and softapp are active

edjeed, 2018-10-30:
Hi,

I'm using the esp-mqtt with the idf, the example compiles fine and works.
After changing the wifi mode from ap => apsta:
 The devices connects with the router
 Start acting as an access point
 But will not connect to the broker

E (80785) MQTT_CLIENT: Error transport connect

edjeed, 2018-10-30:
Fixed the issue by changing ip range of the router

closed 2018-10-30 by edjeed

awadekaustubh, 2018-11-30:
Hi @edjeed ,

I am using wifi manager for esp32 to manage network connection, first esp32 acts as an access point and then connects to the selected wifi network.
After successful connection , i try to connect to mqtt broker but gets same error as yours, please suggest changes you have done to fix this.