espressif/esp-mqtt #86: publish 924 bytes, sent 171 bytes.

zhaochengwei, 2018-11-08:
[image]
[image]

[image]

I publish 924 bytes，But just sent 171 bytes. I check my string and all is json format.How can I solve the problems.

david-cermak, 2019-03-28:
Hi @zhaochengwei 

Sorry for such a late reply, are you still having this issue?
Which version of `esp-mqtt` you use?

david-cermak, 2020-01-09:
Closing for no feedback. Please feel free to reopen!

closed 2020-01-09 by david-cermak