espressif/esp-mqtt #88: Example request: stop client, trying to reset the configuration parameters

LEMAOR, 2018-12-05:
I'm trying to feed new parameters to a program through BLE and store them on the nvs, for example changing user and password, but I'm unable to stop the process properly and cant find an example 

right now for now i have to reset the board to see the changes reflected any ideas how to achieve the configuration change with out a reset?

david-cermak, 2018-12-11:
There's this option already in the internal repo (there was a need for a new event before connecting ); the point is that the method `esp_mqtt_set_config` in not internal anymore and moved to API.
Still some work needs to be done before it's moved to IDF, but probably the version could be  published in GitHub to `idf` branch

david-cermak, 2019-03-28:
resolved via https://github.com/espressif/esp-mqtt/commit/815623dfe5a0e41fa0e51ab4e336feb3eaa5ba15

Note an example in idf https://github.com/espressif/esp-idf/blob/master/examples/protocols/mqtt/publish_test/main/publish_test.c

closed 2019-03-28 by david-cermak