espressif/esp-mqtt #93: MQTT is frequently connecting and disconnecting say around 2 to 3 times in each second.

vinodstanur, 2019-02-01:
I am running the basic example with tcp. It is connecting publishing disconnecting all happening very fast and toggling. So unable to receive any message from subscribed topic. It shows subscribed successfully but instantly it is showing disconnection event and again connection event. These are happening very fast say more than 3 times in a second. So unable to receive any message. I tried to increase the keepalive to higher values but still not working. Basically unable to receive any subscribed message because it is disconnecting as if it is programmed to disconnect immediately after connection. 

My IDF version is 3.3 beta.

david-cermak, 2019-02-01:
Hi @vinodstanur 

Are you sure, you're using correct version of mqtt library? for idf 3.3beta1 (https://github.com/espressif/esp-idf/commit/646d36c7287d8e447b5bd76ec11071b739299887) it should be https://github.com/espressif/esp-mqtt/commit/a7b1cea5b3e246298607a8c64447765297626f36.
I'm asking because the current version does not reconnect that fast (2-3 times per second) yet, reconnection takes place about 10s after disconnect event. At this moment we're merging a PR which improves reconnection, but that's not yet published.

It might be also helpful if you can share your log (would suggest to increase verbosity in menuconfig to debug and remove this line `esp_log_level_set("*", ESP_LOG_INFO);` from the app_main, so we can see debug logs from all modules).

vinodstanur, 2019-02-04:
I installed the esp-idf ones again and updated the submodules. 
Now getting errors as well while building the mqtt example.

`vinod@vinod:~/esp$ git clone https://github.com/espressif/esp-idf
Cloning into 'esp-idf'...
remote: Enumerating objects: 6, done.
remote: Counting objects: 100% (6/6), done.
remote: Compressing objects: 100% (6/6), done.
remote: Total 80287 (delta 0), reused 0 (delta 0), pack-reused 80281
Receiving objects: 100% (80287/80287), 68.37 MiB | 1.02 MiB/s, done.
Resolving deltas: 100% (57226/57226), done.
vinod@vinod:~/esp$ git describe 
fatal: not a git repository (or any of the parent directories): .git
vinod@vinod:~/esp$ cd esp-idf/
vinod@vinod:~/esp/esp-idf$ git describe 
v3.3-beta1-328-gabea9e4c0
vinod@vinod:~/esp/esp-idf$ https://github.com/espressif/esp-idf^C
vinod@vinod:~/esp/esp-idf$ https://github.com/espressif/esp-idf^C
vinod@vinod:~/esp/esp-idf$ https://github.com/espressif/esp-idf^C
vinod@vinod:~/esp/esp-idf$ git submodule update --init --recursive
Submodule 'components/asio/asio' (https://github.com/espressif/asio.git) registered for path 'components/asio/asio'
Submodule 'components/aws_iot/aws-iot-device-sdk-embedded-C' (https://github.com/espressif/aws-iot-device-sdk-embedded-C.git) registered for path 'components/aws_iot/aws-iot-device-sdk-embedded-C'
Submodule 'components/bt/lib' (https://github.com/espressif/esp32-bt-lib.git) registered for path 'components/bt/lib'
Submodule 'components/coap/libcoap' (https://github.com/obgm/libcoap.git) registered for path 'components/coap/libcoap'
Submodule 'components/esp32/lib' (https://github.com/espressif/esp32-wifi-lib.git) registered for path 'components/esp32/lib'
Submodule 'components/esptool_py/esptool' (https://github.com/espressif/esptool.git) registered for path 'components/esptool_py/esptool'
Submodule 'components/expat/expat' (https://github.com/libexpat/libexpat.git) registered for path 'components/expat/expat'
Submodule 'components/json/cJSON' (https://github.com/DaveGamble/cJSON.git) registered for path 'components/json/cJSON'
Submodule 'components/libsodium/libsodium' (https://github.com/jedisct1/libsodium.git) registered for path 'components/libsodium/libsodium'
Submodule 'components/lwip/lwip' (https://github.com/espressif/esp-lwip.git) registered for path 'components/lwip/lwip'
Submodule 'components/mbedtls/mbedtls' (https://github.com/espressif/mbedtls.git) registered for path 'components/mbedtls/mbedtls'
Submodule 'components/micro-ecc/micro-ecc' (https://github.com/kmackay/micro-ecc.git) registered for path 'components/micro-ecc/micro-ecc'
Submodule 'components/mqtt/esp-mqtt' (https://github.com/espressif/esp-mqtt.git) registered for path 'components/mqtt/esp-mqtt'
Submodule 'components/nghttp/nghttp2' (https://github.com/nghttp2/nghttp2.git) registered for path 'components/nghttp/nghttp2'
Submodule 'components/protobuf-c/protobuf-c' (https://github.com/protobuf-c/protobuf-c) registered for path 'components/protobuf-c/protobuf-c'
Submodule 'components/spiffs/spiffs' (https://github.com/pellepl/spiffs.git) registered for path 'components/spiffs/spiffs'
Submodule 'components/unity/unity' (https://github.com/ThrowTheSwitch/Unity) registered for path 'components/unity/unity'
Submodule 'examples/build_system/cmake/import_lib/main/lib/tinyxml2' (https://github.com/leethomason/tinyxml2) registered for path 'examples/build_system/cmake/import_lib/main/lib/tinyxml2'
Cloning into '/home/vinod/esp/esp-idf/components/asio/asio'...
Cloning into '/home/vinod/esp/esp-idf/components/aws_iot/aws-iot-device-sdk-embedded-C'...
Cloning into '/home/vinod/esp/esp-idf/components/bt/lib'...
Cloning into '/home/vinod/esp/esp-idf/components/coap/libcoap'...
Cloning into '/home/vinod/esp/esp-idf/components/esp32/lib'...
Cloning into '/home/vinod/esp/esp-idf/components/esptool_py/esptool'...
Cloning into '/home/vinod/esp/esp-idf/components/expat/expat'...
Cloning into '/home/vinod/esp/esp-idf/components/json/cJSON'...
Cloning into '/home/vinod/esp/esp-idf/components/libsodium/libsodium'...
Cloning into '/home/vinod/esp/esp-idf/components/lwip/lwip'...
Cloning into '/home/vinod/esp/esp-idf/components/mbedtls/mbedtls'...
Cloning into '/home/vinod/esp/esp-idf/components/micro-ecc/micro-ecc'...
Cloning into '/home/vinod/esp/esp-idf/components/mqtt/esp-mqtt'...
Cloning into '/home/vinod/esp/esp-idf/components/nghttp/nghttp2'...
Cloning into '/home/vinod/esp/esp-idf/components/protobuf-c/protobuf-c'...
Cloning into '/home/vinod/esp/esp-idf/components/spiffs/spiffs'...
Cloning into '/home/vinod/esp/esp-idf/components/unity/unity'...
Cloning into '/home/vinod/esp/esp-idf/examples/build_system/cmake/import_lib/main/lib/tinyxml2'...
Submodule path 'components/asio/asio': checked out '55efc179b76139c8f9b44bf22a4aba4803f7a7bd'
Submodule path 'components/aws_iot/aws-iot-device-sdk-embedded-C': checked out '299183238ffe7a3e6a5ca0af9db19c10eaca62cf'
Submodule path 'components/bt/lib': checked out '06c3f28c442984328a8c8f7a38d3c60c7f863e0e'
Submodule path 'components/coap/libcoap': checked out '6468887a12666f88b8704d797fc176cd4f40ee4c'
Submodule path 'components/esp32/lib': checked out 'fa2100a8078ee779ffe7663ba3ccf3abf8b551ce'
Submodule path 'components/esptool_py/esptool': checked out '9ad444a6e06e58833d5e6044c1d5f3eb3dd56023'
Submodule path 'components/expat/expat': checked out '968b8cc46dbee47b83318d5f31a8e7907199614b'
Submodule path 'components/json/cJSON': checked out '7cc52f60356909b3dd260304c7c50c0693699353'
[... cut ...]
CC build/libsodium/libsodium/src/libsodium/crypto_verify/sodium/verify.o
CC build/libsodium/libsodium/src/libsodium/randombytes/randombytes.o
CC build/libsodium/libsodium/src/libsodium/sodium/core.o
CC build/libsodium/libsodium/src/libsodium/sodium/runtime.o
CC build/libsodium/libsodium/src/libsodium/sodium/version.o
CC build/libsodium/libsodium/src/libsodium/sodium/utils.o
CC build/libsodium/port/crypto_hash_mbedtls/crypto_hash_sha256_mbedtls.o
CC build/libsodium/port/crypto_hash_mbedtls/crypto_hash_sha512_mbedtls.o
AR build/libsodium/liblibsodium.a
`

vinodstanur, 2019-02-04:
fatal error: mqtt_client.h: No such file or directory

vinodstanur, 2019-02-04:
UPdate:

Hello,
 I removed the default mqtt submodule and pulled this repo. (https://github.com/espressif/esp-mqtt)

Now it is not disconnecting!

```
16:13:47:937] <0x1b>[0;32mI (5522) MQTT_CLIENT: Sending MQTT CONNECT message, type: 1, id: 0000<0x1b>[0m␍␊
[16:13:48:302] <0x1b>[0;32mI (5852) MQTT_SAMPLE: MQTT_EVENT_CONNECTED<0x1b>[0m␍␊
[16:13:48:302] <0x1b>[0;32mI (5852) MQTT_SAMPLE: sent subscribe successful, msg_id=2898<0x1b>[0m␍␊
[16:13:48:302] <0x1b>[0;32mI (5862) MQTT_SAMPLE: sent subscribe successful, msg_id=42630<0x1b>[0m␍␊
[16:13:48:302] <0x1b>[0;32mI (5862) MQTT_SAMPLE: sent unsubscribe successful, msg_id=53758<0x1b>[0m␍␊
[16:13:48:766] <0x1b>[0;32mI (6352) MQTT_SAMPLE: MQTT_EVENT_SUBSCRIBED, msg_id=2898<0x1b>[0m␍␊
[16:13:48:766] <0x1b>[0;32mI (6352) MQTT_SAMPLE: sent publish successful, msg_id=0<0x1b>[0m␍␊
[16:13:49:098] <0x1b>[0;32mI (6672) MQTT_CLIENT: deliver_publish, message_length_read=21, message_length=16<0x1b>[0m␍␊
[16:13:49:098] <0x1b>[0;32mI (6672) MQTT_SAMPLE: MQTT_EVENT_DATA<0x1b>[0m␍␊
[16:13:49:098] TOPIC=/topic/qos0␍␍␊
[16:13:49:098] DATA=0␍␍␊
[16:13:49:370] <0x1b>[0;32mI (6952) MQTT_SAMPLE: MQTT_EVENT_UNSUBSCRIBED, msg_id=53758<0x1b>[0m␍␊
```

david-cermak, 2019-02-04:
Does it mean that the default submodule version does not even build, but when `esp-mqtt` was replaced with `master` it builds and also works?
This is strange as all IDF commits are tested not only to build all examples, but also to run basic functionality (for mqtt it check simple connection to broker via tcp, ssl, ws).

Did you setup correctly your IDF_PATH to the new idf-repo?

One way to see if correct include paths were propagated to build command is to run e.g.
`make -n | grep mqtt_client`, where you can see all the compilation flags with include dirs...

vinodstanur, 2019-02-04:
HI,
   Yes the default submodule mqtt is not even building.
I just installed fresh esp-idf from master branch.
Then initialized the submodule init. 
Ones it is done I can see mqtt folder in protocols.
IDF path is correct.  

Now if I go into mqtt_tcp example and try build, it failed.

I then cloned this mqtt repo separately and tried to build but it again failed due to the mqtt submodule issue. Then I removed the mqtt submodule using git and then tried to build this and it worked as the libs are already there in the folder. Then I merged it with my application by copying the libraries into my application folder and it is working fine now.

But I still don't understand why the standard installation example doesn't even build. May be you can also try it from your end ones so that we can confirm.

david-cermak, 2019-03-28:
Hi @vinodstanur 

As there's no update from your side, assuming the issue was indeed caused by this strange naming collision between `mqtt` component and `mqtt` example.

Closing the issue, but please feel free to reopen if any issue persists on your end.

closed 2019-03-28 by david-cermak