espressif/esp-mqtt #94: Unable to access to connect response code

GiviMAD, 2019-02-10:
With the current implementation I can't access to the initial connection response code there fore I'm not able to detect a connection refused error like bad username from my code. Am I missing something? Thanks in advance

david-cermak, 2019-02-12:
Yes, you are correct. There's no means to programmatically check error codes.
We can emit an event with id=`MQTT_EVENT_ERROR` and provide (connection) error code as an appropriate context data (that's actually a purpose of this event, but in current version it's never fired).

Thanks for raising this issue, will look into it.

InfiniteYuan1, 2019-07-01:
Is there a way to get the error code?

david-cermak, 2019-07-15:
@InfiniteYuan1 sorry for the delay. The code has just been pushed to IDF master, see the usage in `ssl` example here https://github.com/espressif/esp-idf/blob/master/examples/protocols/mqtt/ssl/main/app_main.c#L82-L86.
(hasn't been merged to MQTT yet, once it's done this issue will be closed)

GiviMAD, 2019-07-23:
Thanks!! better late than never!

david-cermak, 2020-01-09:
Closing as resolved, examples of using this  feature is here https://github.com/espressif/esp-idf/blob/master/examples/protocols/mqtt/ssl/main/app_main.c#L92-L98

closed 2020-01-09 by david-cermak