espressif/esp-mqtt #95: Possiblity of supporting PSK SSL for MQTT?

mreutman, 2019-02-10:
Mosquitto broker supports PSK encryption as an alternative to using certificates. There's obviously downsides to PSK, but it is at least far simpler for users to implement. For PSK, there's no reliance on a third party certificate authority or need to create/manage your own self signed certificates. It seems like the ESP32 userbase (both hobbyist and more serious) could benefit from PSK support.

david-cermak, 2019-02-18:
hi @mreutman 

No plans to support TLS-PSK yet.

But yes, I agree that could be a good alternative to certificates for esp32 users, and it does not seem to be a complicated update. 
Will think about it, but currently the priority is to resolve all pending issues and make the library more robust.

SlavSlavov, 2019-04-17:
I would like to vote for this one as well.

closed 2019-07-29 by david-cermak