espressif/esp-mqtt #97: Disconnect message not implemented

gregjesl, 2019-02-13:
After reviewing and using the code, it does not appear that the MQTT client ever sends the  [DISCONNECT packet](http://public.dhe.ibm.com/software/dw/webservices/ws-mqtt/mqtt-v3r1.html#disconnect) to the broker.  Simply "dropping the line" can lead to undesirable behavior on the broker, especially when the client tries to reconnect.

david-cermak, 2019-02-17:
Hi @gregjesl 

Thank you for raising this issue! 
Indeed the disconnect packet is not implemented. Will be fixed.

gregjesl, 2019-02-17:
@david-cermak Great, thank you.  

After looking through the code, it appears the appropriate place to add the functionality would be here:
https://github.com/espressif/esp-mqtt/blob/99004f2a48b8eda13f0427eefb351aec63f90ff8/mqtt_client.c#L776-L782

My suggestion would be:
```cpp
esp_err_t esp_mqtt_client_stop(esp_mqtt_client_handle_t client)
{
    // Notify the broker we are disconnecting
    client->mqtt_state.outbound_message = mqtt_msg_disconnect(&client->mqtt_state.mqtt_connection);

    if (mqtt_write_data(client) != ESP_OK) {
        ESP_LOGE(TAG, "Error disconnecting");
        return ESP_FAIL;
    }

    client->run = false;
    xEventGroupWaitBits(client->status_bits, STOPPED_BIT, false, true, portMAX_DELAY);
    client->state = MQTT_STATE_UNKNOWN;
    return ESP_OK;
}
```

I'd submit a pull request but I never seem to have much luck with Git submodules.

david-cermak, 2019-02-18:
@gregjesl Thanks for your suggestion!
Please feel free to submit a PR, but please select `idf` as target branch.

gregjesl, 2019-03-22:
@david-cermak I struggled with the rebase so I branched from `idf` and implemented the patch.  The new pull request is https://github.com/espressif/esp-mqtt/pull/107

david-cermak, 2019-03-28:
closed per https://github.com/espressif/esp-mqtt/commit/caf5007b99df985b330683dfe2fa454c94633018

closed 2019-03-28 by david-cermak