espressif/esp-mqtt #98: SLL connection takes a lot of attempt before success up to 8minutes (ssl handshake err -0x7280)

AloyseTech, 2019-02-15:
Hi,

I use the ssl with autheticatin example as a base for my firmware. BLE is continuously scanning for devices and report some of them over mqtt. I have to wait almost 10 minutes to have a succesfull MQTT connection. I repeatedly get the following error :
I (516750) MQTT: Other event id:7
E (517280) esp-tls: mbedtls_ssl_handshake returned -0x7280
E (517280) esp-tls: Failed to open new connection
E (517280) TRANS_SSL: Failed to open a new connection
E (517290) MQTT_CLIENT: Error transport connect
W (517290) MQTT: MQTT_EVENT_DISCONNECTED

After around 10 minutes the connection is successful. But if it disconnect, there the same delay again.

david-cermak, 2019-02-17:
Can you somehow reduce the functionality (BLE + other) to only the ssl connection and check if the issue persists?

The error 0x7280 means that the broker actively terminated connection, one cause might be a "very long" initial ssl handshake (which seeing from the log would not be your case as it fails within 1s)
One way to debug the issue is to enable mbedtls debugging in menuconfig (components->mbedtls->enable debugging) and check which phase of ssl handshake returns error/warnings.

AloyseTech, 2019-02-18:
Hi David,

I've disabled the BLE part of my application and the issue persists.
Here is the log of the first attempt with mbedTLS debug enabled : 

```
D (2766) MQTT_CLIENT: MQTT client_id=ESP32_698204
D (2776) MQTT_CLIENT: Core selection disabled
I (2776) MQTT: Other event id:7
D (2776) esp-tls: host:test.mosquitto.org: strlen 18
D (3346) esp-tls: handshake in progress...
I (3346) mbedtls: ssl_tls.c:8021 => handshake
I (3346) mbedtls: ssl_cli.c:3405 client state: 0
I (3346) mbedtls: ssl_tls.c:2751 => flush output
I (3356) mbedtls: ssl_tls.c:2763 <= flush output
I (3356) mbedtls: ssl_cli.c:3405 client state: 1
I (3366) mbedtls: ssl_tls.c:2751 => flush output
I (3376) mbedtls: ssl_tls.c:2763 <= flush output
I (3376) mbedtls: ssl_cli.c:774 => write client hello
D (3386) mbedtls: ssl_cli.c:812 client hello, max version: [3:3]
D (3386) mbedtls: ssl_cli.c:703 client hello, current time: 3
D (3396) mbedtls: ssl_cli.c:821 dumping 'client hello, random bytes' (32 bytes)
D (3406) mbedtls: ssl_cli.c:821 0000:  00 00 00 03 88 16 52 90 04 38 d8 b1 79 30 e0 64
......R..8..y0.d
D (3416) mbedtls: ssl_cli.c:821 0010:  7c cb db b1 4a 74 2c 80 7e 08 d6 62 f8 fb fa 1d
|...Jt,.~..b....
D (3426) mbedtls: ssl_cli.c:874 client hello, session id len.: 0
D (3426) mbedtls: ssl_cli.c:875 dumping 'client hello, session id' (0 bytes)
D (3436) mbedtls: ssl_cli.c:922 client hello, add ciphersuite: c02c
D (3446) mbedtls: ssl_cli.c:922 client hello, add ciphersuite: c030
D (3446) mbedtls: ssl_cli.c:922 client hello, add ciphersuite: 009f
D (3456) mbedtls: ssl_cli.c:922 client hello, add ciphersuite: c0ad
D (3466) mbedtls: ssl_cli.c:922 client hello, add ciphersuite: c09f
D (3466) mbedtls: ssl_cli.c:922 client hello, add ciphersuite: c024
D (3476) mbedtls: ssl_cli.c:922 client hello, add ciphersuite: c028
D (3486) mbedtls: ssl_cli.c:922 client hello, add ciphersuite: 006b
D (3486) mbedtls: ssl_cli.c:922 client hello, add ciphersuite: c00a
D (3496) mbedtls: ssl_cli.c:922 client hello, add ciphersuite: c014
D (3506) mbedtls: ssl_cli.c:922 client hello, add ciphersuite: 0039
D (3506) mbedtls: ssl_cli.c:922 client hello, add ciphersuite: c0af
D (3516) mbedtls: ssl_cli.c:922 client hello, add ciphersuite: c0a3
D (3516) mbedtls: ssl_cli.c:922 client hello, add ciphersuite: c02b
D (3526) mbedtls: ssl_cli.c:922 client hello, add ciphersuite: c02f
D (3536) mbedtls: ssl_cli.c:922 client hello, add ciphersuite: 009e
D (3536) mbedtls: ssl_cli.c:922 client hello, add ciphersuite: c0ac
D (3546) mbedtls: ssl_cli.c:922 client hello, add ciphersuite: c09e
D (3556) mbedtls: ssl_cli.c:922 client hello, add ciphersuite: c023
D (3556) mbedtls: ssl_cli.c:922 client hello, add ciphersuite: c027
D (3566) mbedtls: ssl_cli.c:922 client hello, add ciphersuite: 0067
D (3576) mbedtls: ssl_cli.c:922 client hello, add ciphersuite: c009
D (3576) mbedtls: ssl_cli.c:922 client hello, add ciphersuite: c013
D (3586) mbedtls: ssl_cli.c:922 client hello, add ciphersuite: 0033
D (3596) mbedtls: ssl_cli.c:922 client hello, add ciphersuite: c0ae
D (3596) mbedtls: ssl_cli.c:922 client hello, add ciphersuite: c0a2
D (3606) mbedtls: ssl_cli.c:922 client hello, add ciphersuite: 009d
D (3616) mbedtls: ssl_cli.c:922 client hello, add ciphersuite: c09d
D (3616) mbedtls: ssl_cli.c:922 client hello, add ciphersuite: 003d
D (3626) mbedtls: ssl_cli.c:922 client hello, add ciphersuite: 0035
D (3636) mbedtls: ssl_cli.c:922 client hello, add ciphersuite: c032
D (3636) mbedtls: ssl_cli.c:922 client hello, add ciphersuite: c02a
D (3646) mbedtls: ssl_cli.c:922 client hello, add ciphersuite: c00f
D (3646) mbedtls: ssl_cli.c:922 client hello, add ciphersuite: c02e
D (3656) mbedtls: ssl_cli.c:922 client hello, add ciphersuite: c026
D (3666) mbedtls: ssl_cli.c:922 client hello, add ciphersuite: c005
D (3666) mbedtls: ssl_cli.c:922 client hello, add ciphersuite: c0a1
D (3676) mbedtls: ssl_cli.c:922 client hello, add ciphersuite: 009c
D (3686) mbedtls: ssl_cli.c:922 client hello, add ciphersuite: c09c
D (3686) mbedtls: ssl_cli.c:922 client hello, add ciphersuite: 003c
D (3696) mbedtls: ssl_cli.c:922 client hello, add ciphersuite: 002f
D (3706) mbedtls: ssl_cli.c:922 client hello, add ciphersuite: c031
[... 53 lines cut ...]
I (4036) mbedtls: ssl_tls.c:2775 ssl->f_send() returned 247 (-0xffffff09)
I (4046) mbedtls: ssl_tls.c:2803 <= flush output
I (4046) mbedtls: ssl_tls.c:3470 <= write record
I (4056) mbedtls: ssl_tls.c:3314 <= write handshake message
I (4056) mbedtls: ssl_cli.c:1106 <= write client hello
I (4066) mbedtls: ssl_cli.c:3405 client state: 2
I (4076) mbedtls: ssl_tls.c:2751 => flush output
I (4076) mbedtls: ssl_tls.c:2763 <= flush output
I (4086) mbedtls: ssl_cli.c:1499 => parse server hello
I (4086) mbedtls: ssl_tls.c:4305 => read record
I (4096) mbedtls: ssl_tls.c:2532 => fetch input
I (4096) mbedtls: ssl_tls.c:2693 in_left: 0, nb_want: 5
I (4106) mbedtls: ssl_tls.c:2717 in_left: 0, nb_want: 5
I (4116) mbedtls: ssl_tls.c:2718 ssl->f_recv(_timeout)() returned 0 (-0x0000)
W (4116) mbedtls: ssl_tls.c:4967 mbedtls_ssl_fetch_input() returned -29312 (-0x7280)
W (4126) mbedtls: ssl_tls.c:4338 ssl_get_next_record() returned -29312 (-0x7280)
W (4136) mbedtls: ssl_cli.c:1506 mbedtls_ssl_read_record() returned -29312 (-0x7280)
I (4146) mbedtls: ssl_tls.c:8031 <= handshake
E (4146) esp-tls: mbedtls_ssl_handshake returned -0x7280
E (4156) esp-tls: Failed to open new connection
E (4166) TRANS_SSL: Failed to open a new connection
E (4166) MQTT_CLIENT: Error transport connect
D (4176) MQTT_CLIENT: Reconnect after 10000 ms
W (4176) MQTT: MQTT_EVENT_DISCONNECTED
```

david-cermak, 2019-02-18:
Hi @AloyseTech 

Thanks for sharing the details. I can see the error occurs directly after the client hello and I can reproduce it locally to see the server sends RST after receiving client_hello. My suspicion is that there's something wrong is test.mosquitto.org (they're actually saying to run unreleased/experimental code). Also checked that I'm not able to connect to it with a python script:
```
ssl.SSLEOFError: EOF occurred in violation of protocol (_ssl.c:726)
```
Nor with mosquitto's pub/sub tools: `Error: The connection was lost.`

Also I do remember the same code did work with test.mosquitto.org some time ago (as mentioned here https://github.com/espressif/esp-idf/issues/2993 while having some cert issues).

Have you tried to use any other server? Like iot.eclipse.org or locally installed mosquitto?

AloyseTech, 2019-02-24:
Hi David,
Sorry I missed the notification.
I will test this with another server and let you know.

david-cermak, 2019-03-28:
Hi @AloyseTech 

Any news? Have you had a chance to test it with another server or local installation?

AloyseTech, 2019-11-26:
Hi @david-cermak , I'm sorry for the very late reply.
This issue has not be solved back in the days : in the rush we switched to AWS.
I can't reproduce it now with ESP-IDF 4.0-beta2.

closed 2019-11-26 by AloyseTech