espressif/esp-mqtt #99: Mqtt ssl client MQTT_EVENT_DISCONNECTED reason

InfiniteYuan1, 2019-02-20:
In our application we use mqtt over ssl to connect to the AWS broker, so we use cert and client cert to mutual authentication.

this is a example cfg for esp_mqtt_client_init:
```
        const esp_mqtt_client_config_t mqtt_cfg = {
            .uri = uri,
            .client_id = mqttSetting.deviceId,
            .event_handle = mqtt_event_handler,
            .cert_pem = (const char *)cert_pem_start,
            .client_cert_pem = (const char *)client_cert_pem_start,
            .client_key_pem = (const char *)client_key_pem_start,
            .buffer_size = 1024*6,
            .keepalive = 60
        };
```
we are able to import our certificate and connect to the broker with the esp_mqtt_client library, we know only if the connection was complete successfully or not, but for error management we want know the reason when it fail the mqtt connection, for example:

- dns fail 
- host not unreachable
- network problem, timeout, etc
- ssl connection issue [not valid certificate]
- mqtt connection refuse by the broker
- etc

david-cermak, 2019-02-20:
Hi @InfiniteYuan1 

I think it's a duplicate of https://github.com/espressif/esp-mqtt/issues/94
We will try to address the issue soon.

david-cermak, 2020-01-09:
Closing as duplicate of https://github.com/espressif/esp-mqtt/issues/94 (Resolved)

closed 2020-01-09 by david-cermak