espressif/esp-mqtt #101: can't connect to wifi

cc1227858751, 2019-03-08:
this is the log

cc1227858751, 2019-03-08:
rst:0xc (SW_CPU_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6008
load:0x40078000,len:9188
load:0x40080000,len:6092
entry 0x4008032c
[0;32mI (29) boot: ESP-IDF v3.1.3-51-g39c6b2f90-dirty 2nd stage bootloader[0m
[0;32mI (29) boot: compile time 13:52:19[0m
[0;32mI (35) boot: Enabling RNG early entropy source...[0m
[0;32mI (35) boot: SPI Speed      : 40MHz[0m
[0;32mI (39) boot: SPI Mode       : DIO[0m
[0;32mI (43) boot: SPI Flash Size : 2MB[0m
[0;32mI (47) boot: Partition Table:[0m
[0;32mI (51) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (73) boot:  2 factory          factory app      00 00 00010000 00100000[0m
[0;32mI (81) boot: End of partition table[0m
[0;32mI (85) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x1c46c (115820) map[0m
[0;32mI (134) esp_image: segment 1: paddr=0x0002c494 vaddr=0x3ffb0000 size=0x02ce8 ( 11496) load[0m
[0;32mI (139) esp_image: segment 2: paddr=0x0002f184 vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (142) esp_image: segment 3: paddr=0x0002f58c vaddr=0x40080400 size=0x00a84 (  2692) load[0m
[0;32mI (151) esp_image: segment 4: paddr=0x00030018 vaddr=0x400d0018 size=0x6ffe4 (458724) map[0m
[0;32mI (320) esp_image: segment 5: paddr=0x000a0004 vaddr=0x40080e84 size=0x0d080 ( 53376) load[0m
[0;32mI (350) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (351) boot: Disabling RNG early entropy source...[0m
[0;32mI (351) cpu_start: Pro cpu up.[0m
[0;32mI (355) cpu_start: Starting app cpu, entry point is 0x40080f98[0m
[0;32mI (347) cpu_start: App cpu up.[0m
[0;32mI (365) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (372) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[0m
[0;32mI (378) heap_init: At 3FFB8D70 len 00027290 (156 KiB): DRAM[0m
[0;32mI (384) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM[0m
[0;32mI (391) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (397) heap_init: At 4008DF04 len 000120FC (72 KiB): IRAM[0m
[0;32mI (403) cpu_start: Pro cpu start user code[0m
[0;32mI (86) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
[0;32mI (87) MQTT_SAMPLE: [APP] Startup..[0m
[0;32mI (87) MQTT_SAMPLE: [APP] Free memory: 274480 bytes[0m
[0;32mI (87) MQTT_SAMPLE: [APP] IDF version: v3.1.3-51-g39c6b2f90-dirty[0m
ESP_ERROR_CHECK failed: esp_err_t 0x3001 (ESP_ERR_WIFI_NOT_INIT) at 0x400d2723
file: "C:/DJ/esp32_win32_msys2_environment_and_toolchain-20180110/msys32/home/12278/esp-idf/examples/components/espmqtt/examples/mqtt_tcp/main/app_main.c" line 100
func: wifi_init
expression: esp_wifi_set_storage(WIFI_STORAGE_RAM)

Backtrace: 0x40089710:0x3ffbaa70 0x40089bc8:0x3ffbaa90 0x400d2723:0x3ffbaab0 0x400d2a24:0x3ffbab60 0x400d0c2e:0x3ffbab80

cc1227858751, 2019-03-08:
IDF release/3.1

cc1227858751, 2019-03-08:
ESP lyrat board

david-cermak, 2019-03-28:
Hi @cc1227858751 

It looks like an initializing issue with wifi, i.e. calling wifi api functions before wifi init. 
I think the easiest way out is to switch to 3.2 release and use those examples from IDF. If you want to stick with 3.1 then just copy the `examples/protocols/mqtt/tcp` folder from 3.2 version -- this example code should work fine with 3.1

david-cermak, 2020-01-09:
Closing for no recent feedback. Please reopen if needed.

closed 2020-01-09 by david-cermak