espressif/esp-mqtt #103: over SSl, Mosquitto broker

balint603, 2019-03-18:
Hi,

I'm having trouble with MQTT level connection, using ESP32.
Finally (it looks like) the transmission layer works, (created .pem file from .crt files) used this advice: 
[https://github.com/espressif/esp-mqtt/issues/84#issuecomment-434408194](https://github.com/espressif/esp-mqtt/issues/84#issuecomment-434408194)
But now my ESP gets an error:
Client log:
> I (5551) MQTT_CLIENT: Sending MQTT CONNECT message, type: 1, id: 0000
> E (5571) MQTT_CLIENT: Invalid MSG_TYPE response: 0, read_len: 16
> I (5571) MQTT_CLIENT: Error MQTT Connected
> I (5581) MQTT_CLIENT: Reconnect after 10000 ms
> I (5581) MQTTS_SAMPLE: MQTT_EVENT_DISCONNECTED

Why does the client get message type error, after the broker sends the CONNACK message?
What could go wrong?
I use the mqtt_ssl example project.
Thanks :)

david-cermak, 2019-03-20:
Hi @balint603 

This `E (5571) MQTT_CLIENT: Invalid MSG_TYPE response: 0, read_len: 16` does not look like a CONNACK message (or do you have a packet log to see that this message was actually sent?), this is an unknown message so some unexpected communication or could be a bug.

Which version of idf and esp-mqtt are you using?
Also can you please share the log when debug level enabled?
Thanks!

balint603, 2019-03-20:
> Hi @balint603
>
> This `E (5571) MQTT_CLIENT: Invalid MSG_TYPE response: 0, read_len: 16` does not look like a CONNACK message (or do you have a packet log to see that this message was actually sent?), [...]

Hi @david-cermak,

I am using 3.0 version of idf and esp-mqtt checked out to ESP-MQTT_FOR_IDF_3.1. 
The TCP version perfectly works.
The only source I have the Mosquitto log:
```
1553113486: New connection from 192.168.1.115 on port 8883.
1553113487: New client connected from 192.168.1.115 as ESP32_d5FE19 (c1, k120).
1553113487: Sending CONNACK to ESP32_d5FE19 (0, 0)
1553113487: Socket error on client ESP32_d5FE19, disconnecting.
```
Mqtt client log:
```
D (5125) MQTT_CLIENT: MQTT client_id=ESP32_d5FE19
D (5135) MQTT_CLIENT: Core selection enabled on 0
I (5595) wifi: pm start, type:0

D (6035) MQTT_CLIENT: Transport connected to mqtts://192.168.1.11:8883
I (6035) MQTT_CLIENT: Sending MQTT CONNECT message, type: 1, id: 0000
E (6045) MQTT_CLIENT: Invalid MSG_TYPE response: 0, read_len: 16
I (6055) MQTT_CLIENT: Error MQTT Connected
I (6065) MQTT_CLIENT: Reconnect after 10000 ms
I (6065) MQTTS_SAMPLE: MQTT_EVENT_DISCONNECTED
D (21065) MQTT_CLIENT: Reconnecting...
```
My client config:
```
    const esp_mqtt_client_config_t mqtt_cfg = {
        .uri = "mqtts://192.168.1.11:8883",
        .event_handle = mqtt_event_handler,
        .cert_pem = (const char *)iot_eclipse_org_pem_start,
    };
```
Thank you!

david-cermak, 2019-03-28:
Hi @balint603 

Any news from your side? We've continued with the discussion on the forum here https://www.esp32.com/viewtopic.php?f=13&t=9747

I am not able to reproduce your issue, using mosquitto 1.4. What is your version?

AntoinedeChassey, 2019-11-29:
Hi @balint603, I have the same error...
```
I (7318) mbedtls: ssl_tls.c:2774 message length: 55, out_left: 55

I (7325) mbedtls: ssl_tls.c:2779 ssl->f_send() returned 55 (-0xffffffc9)

I (7332) mbedtls: ssl_tls.c:2807 <= flush output

I (7339) mbedtls: ssl_tls.c:3476 <= write record

I (7348) mbedtls: ssl_tls.c:8710 <= write

E (17352) MQTT_CLIENT: Invalid MSG_TYPE response: 8, read_len: 0
I (17358) MQTT_CLIENT: Error MQTT Connected
I (17366) MQTT_CLIENT: Reconnect after 10000 ms
I (17371) mqtt_event_handler: MQTT_EVENT_DISCONNECTED
```
Any advice maybe? Did you manage to solve it?
Thanks!

balint603, 2019-11-29:
> Hi @balint603, I have the same error...
>
> ```
> I (7318) mbedtls: ssl_tls.c:2774 message length: 55, out_left: 55
>
> I (7325) mbedtls: ssl_tls.c:2779 ssl->f_send() returned 55 (-0xffffffc9)
>
> I (7332) [...]

balint603, 2019-11-29:
Hi @AntoinedeChassey,
Nope, I switched to HTTPS but there must be a solution. 
Do you get invalid message type just after the CONNECT?

The issue was continued on https://www.esp32.com/viewtopic.php?f=13&t=9747. Check this and you should try @david-cermak ideas!

david-cermak, 2019-12-02:
@AntoinedeChassey Which version of IDF (and esp-mqtt) are you using?
Are you also on some older version, such as 3.0? Most likely thread safety issue, I've posted this [patch](https://github.com/espressif/esp-mqtt/issues/102#issuecomment-475609024) for IDF<3.2 some time ago, but would recommend to update to more recent version.

AntoinedeChassey, 2019-12-03:
Hi guys,
I tried a lot of different things and have abandoned (_for now_) using web-sockets.
Also, I am using an ESP-8266 and not ESP-32 (_but might give it a go too later_).
I've also changed my broker for the EMQX which seems a bit more complete and up-to-date.
I am still struggling with TCP TLS but am on the right path and might have some time to get it to work in the coming days :)
(Also I use [Traefik](https://docs.traefik.io/) as a reverse proxy on my server).
Best and thx for your responses!

david-cermak, 2020-01-09:
Hi @AntoinedeChassey 

Any updates on this issue? Could we close it?

AntoinedeChassey, 2020-01-09:
Hi David,
I am using direct TCP and it works fine.
I also switched to the master branch of ESP8266_RTOS_SDK because I had some issues with TLS on v3.3
You can close this issue,
Have a nice day!

david-cermak, 2020-01-09:
Thanks, closing the issue.
 a good day to you too!

closed 2020-01-09 by david-cermak