espressif/esp-mqtt #104: Certificate Common Name, DNS, CNAME

InfiniteYuan1, 2019-03-20:
in our current backend architecture we use different DNS record to point to aws endpoint cname

for example: cmd-eu-prod.worxlandroid.com => a1optpg91s0ydf.iot.eu-west-1.amazonaws.com

this cause fail in the certificate common name check because it look for **CN=\*.iot.eu-west-1.amazonaws.com**

```
I (10218) mqtt: \[STATE\] - MQTT_STATE_CONNECTING 
E (11169) esp-tls: mbedtls_ssl_handshake returned -0x2700 
I (11170) esp-tls: Failed to verify peer certificate! 
I (11170) esp-tls: verification info: ! The certificate Common Nam e (CN) does not match with the expected CN
E (11172) esp-tls: Failed to open new connection 
E (11173) TRANS_SSL: Failed to open a new connection
```

please support us on one of the solutions or give your proposal

- solution A perform dnslookup before init connection
- solution B call back function after fail verification, run dnslookup on it, custom verification, overwrite the result
- solution C setup mbed TLS to automatically perform dnslookup and use the result for the CN verification

david-cermak, 2019-03-21:
Hi @InfiniteYuan1 

Would it help to create a certificate with `CN=*.iot.eu-west-1.amazonaws.com` and (as extension) alternative name `cmd-eu-prod.worxlandroid.com`?

InfiniteYuan1, 2019-03-22:
sorry, we have already client certificate on the market and we can not change them, furthermore it fail on the server authentication, so the change proposed by you should be made on the amazon side and for sure they can not do this for us.

it look that the implementation of lwip_gethostbyname is not able to got the alias list

s_aliases = NULL; s_hostent.h_aliases = &s_aliases;

there are other solution for this situation.

david-cermak, 2019-03-25:
AFAIK there's no way to resolve DNS aliases through LwIP as is, so I can see two options:
1) Update LWIP/dns module to process nameservers in dns queries.
2) Send a separate udp packet with NS query and process responses in application code (which sounds better to me)

But still updating a certificate seems to me like the easiest option, there's no issue in having devices in field as it's the server certificate, client certificate can stay the same as well as the CA certificate  for server verification in your device. Just update your backend the same way as if your server certificate expired.

closed 2019-03-28 by projectgus

InfiniteYuan1, 2019-03-29:
What shoudl we do use LWIP/dns module to process nameservers in dns queries?
And send a separate udp packet?Can you provide some doc for me, thank you.

david-cermak, 2019-03-29:
I would suggest to continue with this conversation on the forum to get more feedback from more knowledge people. Plus it's not in scope of `esp-mqtt` (nor issue with the library)

General idea is to took at dns packets and parse for NS entries (https://tools.ietf.org/html/rfc2929). I would not recommend updating dns module in lwip.
As a separate query you can start with composing a packet (as outlined for example here https://github.com/tigerlyb/DNS-Lookup-Tool-in-Python/blob/master/dnslookup.py) and parse a response.

LHYYSY, 2023-11-03:
[image]
I have also encountered this problem when using the tls_mutual_auth routine, it is not clear why this problem occurs.

euripedesrocha, 2023-11-21:
@LHYYSY I suggest that you open a new issue with the details of your problem. I also noticed that you are using coremqtt so this repository would not be the right place to ask. 

Looking into your logs I can see the certificate could not be verified, check you have the correct certificates set.