espressif/esp-mqtt #109: getting msg id incorrect for MQTT_MSG_TYPE_UNSUBSCRIBE from mqtt_get_id function in mqtt_msg.c since it is not listed in switch case.

Suraj151, 2019-04-01:


david-cermak, 2019-04-08:
Hi @Suraj151 

Thanks for noting this inconsistency. Does it cause any issue in current implementation? I can see  that `mqtt_get_id` is used mainly in processing of received messages (so it's not expected to get a subscribe/unsubsribe message in the client).

Suraj151, 2019-04-08:
it matters as we checks for pending msg id in receive function that was assigned to packet before send and on success of this checks we call unsubscribe callbacks to remove unsubscribed topic from subscribed topic queue.
and if you don't get correct msg id then your check fails and abort next callback process. that issue i faced.

david-cermak, 2019-04-08:
So that you call `mqtt_get_id` from your application code? Using a private include or am I missing something?

I'm sorry, surly this needs to fixed, but I would like to understand the issue and your use case.

Suraj151, 2019-04-08:
right !
i called `mqtt_get_id` from my application code.

i have a queue that keeps track about subscribed topics and it get updated from subscribe & unsubscribe callbacks.these callbacks called from received responses from mqtt broker.

Since my application needs dynamically subscribe and unsubscribe topics on periodic basis i need to keep track of subscribed and unsubscribed topics.

there is another queue where i push next topics to be subscribed and unsubscribed. when this queue get processed (i.e. sending of subscribe and unsubscribe packets) i assigned them with unique runtime msg ids those requires later to check on response from broker. on this id checks it calls subscribe and unsubscribe callbacks.

i got subscribe id correctly in response check but not unsubscribed ones. on debugging some i get to know about subjected issue. after adding unsubscribe type in switch case i get unsubscribe id correctly and my subscribed topic queue gets updated correctly.

so i just informed here this.

david-cermak, 2019-04-09:
ok, understand, thanks for explaining your use case.
will correct the switch case.

commit 35fc42d2 (2019-05-15): mqtt_msg: added missing message type for unsubscribe msg type

Closes #109
closed 2019-06-10 by david-cermak