espressif/esp-mqtt #115: Till this library gets production ready; which other esp32 MQTT library would you recommend?

imranaalam, 2019-04-24:
i am making a prototype and i have to come to market fast without errors or bugs. 
which other libraryu with QOS 0,1 and 2 is suitable for esp32 MQTT?

david-cermak, 2019-04-27:
This probably depends on your application, if you indent to use a specific cloud solution (amazon, microsoft..), it's recommended to use related IoT SDKs, such as:
* https://github.com/espressif/esp-azure
* https://github.com/espressif/esp-aws-iot
* https://github.com/espressif/esp-google-iot

In case one wants a general mqtt client I would recommend this library as it's a very simple yet quite generic and complete implementation, under active maintenance and continuous testing (+ testing  with esp-idf). Of course there are issues and bugs, at this moment two outstanding issues, which will be addressed soon. If you have any concern or know about a missing feature please file an issue.

If you're looking for an alternative, maybe open a topic in the esp32 forum. There are other mqtt libraries used with ESP32, I personally cannot but recommend this one, I am biased though :smile:

hg5134021, 2019-07-14:
Hi , 
Learning ESP n entry level ,  May i know, 
Is this library also applicable to **ESP8266** ?
Noticed this is mentioned as **ESP32**.  Just want to double confirm and make my self clear on this.
Thanks in advance.

david-cermak, 2020-01-09:
@imranaalam Any updates? I would say the stability of the esp-mqtt has improved.

milamber-ls, 2020-01-12:
take a look at  wolfmqtt https://www.wolfssl.com/products/wolfmqtt/

david-cermak, 2020-01-12:
@imranaalam Thanks for the feedback! Yes, wolfSSL is a good alternative  and is already included in `esp-idf` as an **option** for TLS stack. The downside is the license, which for some users might not be acceptable.

e4tech, 2020-01-13:
What is the size of the ESP-MQTT (this) library, I would like to know as the wolfSSL one is 3.6kB.

Btw, hoping that this (current) one keeps getting better (as has been the case recently. Thanks @david-cermak !)

david-cermak, 2020-01-14:
```
> idf.py size-components
            Archive File DRAM .data & .bss   IRAM Flash code & rodata   Total
...
               libmqtt.a          4      0      0      11224     3426   14654
```

As for the wolfmqtt, you can get similar numbers when building for ESP32, this is what I get when included directly into the main component:
```
               libmain.a          0   1042      0      10216     4043   15301
```
Noting that the original main was about `3433` total, so indeed the footprint of wolfmqtt is smaller on ESP platform.