espressif/esp-mqtt #116: wolfssl with esp-mqtt

m0ot, 2019-04-26:
Hello everybody,
I'm trying to implement esp-mqtt over wolfssl on esp32 in order to be able to create a tls1.3 connection between the board and the broker. However, I'm trying now to figure out how can I find the sent as well as received data of esp-mqtt? which are needed for the following wolfssl APIs:

void wolfSSL_write(WOLFSSL* ssl, const void* sent_data, int sz_of_sent_data)
void wolfSSL_read(WOLFSSL* ssl, void* received_data, int sz_of_received_data)

Any ideas?
thanks

david-cermak, 2019-04-27:
Hi @m0ot 

Please look at [tcp_transport](https://github.com/espressif/esp-idf/tree/master/components/tcp_transport) component in idf. esp-mqtt uses this api and specific transport layer is transparent for this library. All you need to do is to add a new transport, i.e. define specific functions for read, write, connect and assign them as function pointers to generic [esp_transport](https://github.com/espressif/esp-idf/blob/master/components/tcp_transport/include/esp_transport.h) api.

m0ot, 2019-04-28:
thanks @david-cermak for your replay!
I'm working on the [mqtt_tcp example](https://github.com/espressif/esp-idf/tree/master/examples/protocols/mqtt/tcp). Until now, I was able to successfully establish the wolfssl connection between the board and the broker (see here [app_main.zip](https://github.com/espressif/esp-mqtt/files/3125071/app_main.zip)), but still missing the transporting of mqtt data under this secure connection. As I understood, all what I have to do is to edit the tcp_read, tcp_write, tcp_connect functions inside [tcp_transport](https://github.com/espressif/esp-idf/blob/master/components/tcp_transport/transport_tcp.c). 
Am I right? or I'm still missing something to understand it?

david-cermak, 2019-04-29:
@m0ot It is a bit more complicated. Of course you can just alter tcp_read/tcp_write in tcp_transport just to get it working as a prove of concept.

Please look at https://github.com/espressif/esp-idf/blob/master/components/tcp_transport/transport_ssl.c#L199 initialisation of ssl transport actually sets specific functions defined in `ssl_transport` to common transport methods for read, write. Module [`esp_transport.h`](https://github.com/espressif/esp-idf/blob/master/components/tcp_transport/include/esp_transport.h) works something like a super-class with "virtual" methods for read, write, etc which are overridden in derived transport.

Correct solution would probably be a definition of a new transport which would init specific pointers to your `wolfSSL_write` and `wolfSSL_read` methods in `
esp_transport_set_func` in for example `esp_transport_handle_t esp_transport_wolfssl_init()`.
Then this library would init ssl transport in https://github.com/espressif/esp-mqtt/blob/idf/mqtt_client.c#L362 such as 
```
esp_transport_handle_t ssl = esp_transport_wolfssl_init();
```

m0ot, 2019-05-06:
hi @david-cermak 
It just worked as expected. I had to create esp_transport_wolfssl and then define esp_transport_wolfssl_init() in mqtt_client.c. 
But I still have only one question: Do you have any ideas what would be the first step in order to add ( what I just developed) to MicroPython? I mean MicroPython uses also the Espressif IDF for esp32. I'm sure if this is would be possible or at least dependent with what I just developed.

david-cermak, 2019-05-08:
Hi @m0ot 

Thanks for the update about wolfssl, sounds interesting. Maybe, if you have some code to share, a link might be helpful also for others...

I cannot help with MicroPython, I'm afraid. General idea is probably to recompile the micropython itself with your mqtt implementation 
(with special function decoration for the API to be accessible from python -- per quick search found this example https://github.com/dhylands/micropython/commit/b801dbd39eb323494f946c13362f4957f5d7281b)

I would suggest to ask in the esp32 forum for more details.

m0ot, 2019-05-12:
Hi @david-cermak 
Thanks for our replay.
Actually, I wanted to contribute with the code on ESP-IDF project. But I'm not sure, if it's allowed to do that.? Because ESP-IDF project licensed under Apache License 2.0 but wolfSSL licensed under both the GPLv2 as well as standard commercial licensing.

david-cermak, 2019-05-23:
@m0ot 
Sorry for the late answer, Indeed including wolfSSL might introduce a license collision.
Also wolfSSL is already included in 8266 SDK (https://github.com/espressif/ESP8266_RTOS_SDK/blob/master/components/esp-tls/esp_tls.c). It is integrated in esp-tls component (below the transport_ssl) and there're plans to port to esp32/IDF.

Please note the license info here https://github.com/espressif/ESP8266_RTOS_SDK/tree/master/components/ssl/wolfssl about using commercial rather than GPLv2 license for Espressif MCU products.

david-cermak, 2019-11-01:
Closing as wolfSSL is already supported on IDF-master from https://github.com/espressif/esp-idf/commit/f7eaa5f9467da8261b7ed55413a1d2c6d6913817

closed 2019-11-01 by david-cermak