espressif/esp-mqtt #126: Reconnection doesn't occur after: Error write data or timeout

no1seman, 2019-08-08:
- Development Kit:  ESP32-Wrover-Kit
- Kit version (for WroverKit/PicoKit/DevKitC): v4.1
- Module or chip used: ESP32-WROVER-B
- IDF version: v4.0-dev-1443-g39f090a4f
- Build System: CMake
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
- Operating System: Linux
- Power Supply:  USB|external 5V&Battery

Making long run test of my app with latest mosquitto 1.6.4 got 3 different cases: 
1. Success reconnect if 'No PING_RESP':
...
E (1177186) MQTT_CLIENT: No PING_RESP, disconnected
I (1177186) MQTT: MQTT client reconnecting...
I (1177186) MQTT_CLIENT: Client force reconnect requested
I (1177226) MQTT_CLIENT: Sending MQTT CONNECT message, type: 1, id: 0000
I (1177236) MQTT: MQTT client connected
E (3103846) MQTT_CLIENT: No PING_RESP, disconnected
I (3103846) MQTT: MQTT client reconnecting...
I (3103846) MQTT_CLIENT: Client force reconnect requested
I (3103896) MQTT_CLIENT: Sending MQTT CONNECT message, type: 1, id: 0000
I (3103976) MQTT: MQTT client connected
...
2. Success reconnect if ' transport_read() error: errno=128'
...
E (40626) MQTT_CLIENT: mqtt_message_receive: transport_read() error: errno=128
E (40626) MQTT_CLIENT: mqtt_process_receive: mqtt_message_receive() returned -1
I (40636) MQTT: MQTT client reconnecting...
I (40636) MQTT_CLIENT: Client force reconnect requested
I (40716) MQTT_CLIENT: Sending MQTT CONNECT message, type: 1, id: 0000
I (40726) MQTT: MQTT client connected
...
3. Reconnect fails if 'Error write data or timeout':
...
E (329776) MQTT_CLIENT: Error write data or timeout, written len = 0, errno=0
I (329776) MQTT: MQTT client reconnecting...
I (329776) MQTT_CLIENT: Client force reconnect requested
...
No any connection to MQTT server after. Whats wrong?

Here is verbose log of 3rd situation:
[log.zip](https://github.com/espressif/esp-mqtt/files/3481998/log.zip)

Logic of my application:
...
```
static esp_err_t mqtt_event_handler(esp_mqtt_event_handle_t event)
{
    switch (event->event_id)
    {
    case MQTT_EVENT_CONNECTED:
        xEventGroupSetBits(app_state, MQTT_CLIENT_CONNECTED_BIT);
        ESP_LOGI(TAG, "MQTT client connected");
        break;
    case MQTT_EVENT_DISCONNECTED:
        if (is_mqtt_client_connected())
        {
            xEventGroupClearBits(app_state, MQTT_CLIENT_CONNECTED_BIT);
            ESP_LOGI(TAG, "MQTT client reconnecting...");
            esp_mqtt_client_reconnect(mqtt_client);
        }
        else
        {
            ESP_LOGI(TAG, "MQTT client disconnected");
        }
        break;
    case MQTT_EVENT_SUBSCRIBED:
        ESP_LOGD("MQTT_EVENT_SUBSCRIBED");
        break;
    case MQTT_EVENT_UNSUBSCRIBED:
        ESP_LOGD("MQTT_EVENT_UNSUBSCRIBED");
        break;
    case MQTT_EVENT_PUBLISHED:
        ESP_LOGD("MQTT message published: msg_id=%d", event->msg_id);
        break;
    case MQTT_EVENT_DATA:
        ESP_LOGD("MQTT_EVENT_DATA");
        break;
    case MQTT_EVENT_ERROR:
        ESP_LOGD("MQTT_EVENT_ERROR");
        break;
    case MQTT_EVENT_BEFORE_CONNECT:
       ESP_LOGD("MQTT_EVENT_BEFORE_CONNECT");
        break;
    }
    return ESP_OK;
}
```
...
```
void mqtt_task(void *pvParameter)
{
while (1)
    {            
if (is_mqtt_client_connected())
            {
                int msg_id;
                stats_t *stats = stats_to_struct();
                if (stats)
                {
                    msg_id = esp_mqtt_client_publish(mqtt_client, stats_topic, (char *)stats, sizeof(stats_t), 1, 0);
                    if (msg_id != -1)
                    {
                        mqtt_stats_counter++;
                    }
                    else
                    {
                        //if (mqtt_stats_counter > 0)
                        //    mqtt_stats_counter--;
                        ESP_LOGE(TAG, "Stats message send failed");
                    }
                }
                else
                {
                    ESP_LOGE(TAG, "Error getting node stats");
                }

                free(stats);
}
            else
            {
                MQTT_LOGE("MQTT client is not connected");
            }
vTaskDelay(pdMS_TO_TICKS(MQTT_SEND_PERIOD));
}
}
```
...

david-cermak, 2019-08-13:
Hi @no1seman 

From the logs it looks like an issue of internal logic of mqtt-task (race condition or a dead-lock) as the call to `esp_mqtt_client_reconnect()` succeeds,
but reconnection is never triggered. I wasn't able to reproduce this issue on my end, though.

* Does your application ever call `esp_mqtt_client_stop()` to reinitialize the connection? (assume it doesn't from your description, just checking due to a similar recent issue)
* Could you please test it without calling `esp_mqtt_client_reconnect()` explicitly, just setting `MQTT_RECONNECT_TIMEOUT_MS` to zero (in the library code)? 
This should result in the same behaviour but we can at least rule out some part of the code.
* Also I wonder what state the mqtt client is in when the issue occurs. So possibly if you could add some logging info to the code of client's state in publish code and inside mqtt-task it would be really helpful. Thanks!
mqtt_client.c:
```
...
static void esp_mqtt_task(void *pv)
{
    ...
    xEventGroupClearBits(client->status_bits, STOPPED_BIT);
    while (client->run) {
        MQTT_API_LOCK(client);
        ESP_LOGD(TAG, "mqtt_task: client state :%d", client->state);
        switch ((int)client->state) {
        case MQTT_STATE_INIT:
    ...
...

int esp_mqtt_client_publish(..)
{
    ESP_LOGD(TAG, "Publish: client state :%d", client->state);
    ...
...

```

no1seman, 2019-08-13:
@david-cermak,
thank you for recomendations.

1. I fixed it manually, because fix still not merged into master esp-idf: moved xEventGroupWaitBits(client->status_bits, STOPPED_BIT, false, true, portMAX_DELAY); 2 lines down after MQTT_API_UNLOCK_FROM_OTHER_TASK(client);
2. esp_mqtt_client_reconnect() - commented
3. Logging added
4. setting MQTT_RECONNECT_TIMEOUT_MS to zero leads to esp_mqtt after 10-20 publish messages start to publish every message with QOS1 10-15 times with the same payload but different msgid.
Here is the log:
```
I (8341) MQTT: Setting up MQTT Client...
D (8351) MQTT_CLIENT: MQTT client_id=1yi5q0xqwl
I (8351) WIFI STA: stopped
I (8361) WIFI STA: setup complete
I (8361) WIFI AP: DNS setup complete
I (9091) WIFI AP: DHCP server start IP: '192.168.0.2'
I (9091) WIFI AP: DHCP server end IP: '192.168.0.102'
I (9101) WIFI AP: DHCP server setup complete
I (9101) WIFI AP: setup complete
I (9101) WIFI: setup complete
W (9111) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration
I (9241) phy: phy_version: 4102, 2fa7a43, Jul 15 2019, 13:06:06, 0, 2
I (9351) WIFI: started
I (9351) BTDM_INIT: BT controller compile version [2f7e614]
I (9351) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (9521) QUEUE: BLE queue ring buffer created successfully
I (9521) BLE: 'BEACONS' task started
I (9521) BLE: host task started
I (9531) GATT: registered service 0x1800 with handle=1
I (9531) GATT: registering characteristic 0x2a00 with def_handle=2 val_handle=3
I (9541) GATT: registering characteristic 0x2a01 with def_handle=4 val_handle=5
I (9551) GATT: registered service 0x1801 with handle=6
I (9551) GATT: registering characteristic 0x2a05 with def_handle=7 val_handle=8
I (9561) GATT: registered service 59462f12-9543-9999-12c8-58b459a2712d with handle=10
I (9571) GATT: registering characteristic 5c3a659e-897e-45e1-b016-007107c96df6 with def_handle=11 val_handle=12
I (9581) GATT: registering characteristic 5c3a659e-897e-45e1-b016-007107c96df7 with def_handle=13 val_handle=14
I (9601) WIFI STA: started
I (9601) WIFI AP: started
I (9631) mDNS: service started
I (9631) HTTP: server started
I (9631) APP: Blink task started
I (9631) APP: Telnet task started
I (10511) WIFI STA: connected
I (11101) tcpip_adapter: sta ip: 192.168.3.130, mask: 255.255.255.0, gw: 192.168.3.1
I (11101) WIFI STA: got IP: '192.168.3.130', Netmask: '255.255.255.0', Gateway: '192.168.3.1'
I (11101) SNTP: SNTP client started
I (12931) SNTP: got time from NTP server
I (12941) SNTP: RTC time is valid, time delta: 0.000000 second(s)
I (13781) ETHERNET: setup complete
I (13781) ETHERNET: connected
I (18101) ETHERNET: got IP: '192.168.3.222', Netmask: '255.255.255.0', Gateway: '192.168.3.1'
D (18101) MQTT: MQTT client starting...
D (18101) MQTT_CLIENT: Core selection disabled
I (18101) MQTT: MQTT client started
D (18101) MQTT_CLIENT: mqtt_task: client state :1
D (18111) MQTT: MQTT_EVENT_BEFORE_CONNECT
D (18131) MQTT_CLIENT: Transport connected to mqtt://192.168.3.211:1883
I (18131) MQTT_CLIENT: Sending MQTT CONNECT message, type: 1, id: 0000
D (18141) MQTT_CLIENT: mqtt_message_receive: first byte: 0x20
D (18141) MQTT_CLIENT: mqtt_message_receive: read "remaining length" byte: 0x2
D (18151) MQTT_CLIENT: mqtt_message_receive: total message length: 4 (already read: 2)
D (18151) MQTT_CLIENT: mqtt_message_receive: read_len=2
D (18161) MQTT_CLIENT: mqtt_message_receive: transport_read():4 4
D (18161) MQTT_CLIENT: Connected
I (18171) MQTT: MQTT client connected
D (19171) MQTT_CLIENT: mqtt_task: client state :2
V (19171) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
D (19171) MQTT_CLIENT: Sent PING successful
D (19171) MQTT_CLIENT: PING sent
D (19201) MQTT_CLIENT: mqtt_task: client state :2
D (19201) MQTT_CLIENT: mqtt_message_receive: first byte: 0xd0
D (19201) MQTT_CLIENT: mqtt_message_receive: read "remaining length" byte: 0x0
D (19201) MQTT_CLIENT: mqtt_message_receive: total message length: 2 (already read: 2)
D (19211) MQTT_CLIENT: mqtt_message_receive: transport_read():2 2
D (19221) MQTT_CLIENT: msg_type=13, msg_id=0
D (19221) MQTT_CLIENT: MQTT_MSG_TYPE_PINGRESP
W (26421) MQTT: PUBLISHING BLE beacon_packet_type: 3
D (26421) MQTT_CLIENT: Publish: client state :2
D (26421) MQTT_CLIENT: mqtt_enqueue id: 28365, type=3 successful
W (29641) MQTT: PUBLISHING STATS
D (29641) MQTT_CLIENT: Publish: client state :2
E (36421) MQTT_CLIENT: Error write data or timeout, written len = 0, errno=0
D (36421) MQTT_CLIENT: Reconnect after 10000 ms
I (36421) MQTT: MQTT client reconnecting...(MQTT_EVENT_DISCONNECTED event with commented: esp_mqtt_client_reconnect(mqtt_client))
D (36431) MQTT_CLIENT: mqtt_enqueue id: 3047, type=3 successful
D (36441) MQTT_CLIENT: Publish: client is not connected
W (36441) MQTT: PUBLISHING CLIMATE
D (36451) MQTT_CLIENT: Publish: client state :3
D (36451) MQTT_CLIENT: mqtt_enqueue id: 10464, type=3 successful
D (36461) MQTT_CLIENT: Publish: client is not connected
```

Seems that disconnection occur between is_mqtt_client_connected() and esp_mqtt_client_publish()
```
if (is_mqtt_client_connected())
{ 
... < = disconnect occur here
esp_mqtt_client_publish()....
...
}
```

Messages like: W (29641) MQTT: PUBLISHING - right before esp_mqtt_client_publish()

david-cermak, 2019-08-14:
@no1seman Thanks for the detailed info. Seems the mqtt_task got stuck in `esp_transport_poll_read()` (waiting on `select` while socket closed from another thread). Could you please yet modify this line `if (esp_transport_poll_read(client->transport, MQTT_POLL_READ_TIMEOUT_MS)` in mqtt task to confirm this hypothesis?
```
       ... after the while 1 loop
        MQTT_API_UNLOCK(client);
        if (MQTT_STATE_CONNECTED == client->state) {
            vTaskDelay(MQTT_POLL_READ_TIMEOUT_MS / portTICK_PERIOD_MS);
            if (esp_transport_poll_read(client->transport, 0) < 0) {
```

> Question: Does mqtt_client sends MQTT_EVENT_CONNECTED after reconnection?

Yes it does, example of the log from IDF example updated to disconnect & reconnect while publishing
```
connect after 10000 ms
D (409573) MQTT_EXAMPLE: Event dispatched from event loop base=MQTT_EVENTS, event_id=2
I (409583) MQTT_EXAMPLE: MQTT_EVENT_DISCONNECTED
I (409593) MQTT_CLIENT: Client force reconnect requested
D (409593) MQTT_CLIENT: mqtt_enqueue id: 13567, type=3 successful
D (409603) OUTBOX: ENQUEUE msgid=13567, msg_type=3, len=20, size=2318
D (409613) MQTT_CLIENT: Publish: client is not connected
D (409613) MQTT_CLIENT: mqtt_enqueue id: 13751, type=3 successful
D (409623) OUTBOX: ENQUEUE msgid=13751, msg_type=3, len=21, size=2339
D (409623) MQTT_CLIENT: Publish: client is not connected
D (410553) MQTT_CLIENT: Reconnecting...
D (410553) MQTT_EXAMPLE: Event dispatched from event loop base=MQTT_EVENTS, event_id=7
I (410553) MQTT_EXAMPLE: Other event id:7
E (410593) MQTT_EXAMPLE: Publishing...
D (410603) MQTT_CLIENT: Transport connected to mqtt://192.168.0.122:1883
I (410613) MQTT_CLIENT: Sending MQTT CONNECT message, type: 1, id: 0000
E (410613) MQTT_EXAMPLE: Publishing...
D (410613) MQTT_CLIENT: mqtt_message_receive: first byte: 0x20
D (410623) MQTT_CLIENT: mqtt_message_receive: read "remaining length" byte: 0x2
D (410633) MQTT_CLIENT: mqtt_message_receive: total message length: 4 (already read: 2)
D (410633) MQTT_CLIENT: mqtt_message_receive: read_len=2
D (410643) MQTT_CLIENT: mqtt_message_receive: transport_read():4 4
D (410653) MQTT_CLIENT: Connected
D (410653) MQTT_EXAMPLE: Event dispatched from event loop base=MQTT_EVENTS, event_id=1
I (410663) MQTT_EXAMPLE: MQTT_EVENT_CONNECTED
D (410663) MQTT_CLIENT: mqtt_enqueue id: 34993, type=8 succes
```

no1seman, 2019-08-14:
@david-cermak,

I added vTaskDelay(MQTT_POLL_READ_TIMEOUT_MS / portTICK_PERIOD_MS); to esp_mqtt_task() but it doesn't solve the problem, here if full log:
[beacon1.zip](https://github.com/espressif/esp-mqtt/files/3501605/beacon1.zip)
...
A little bit after I read you post more carefully and turned on LwIP socket debug (#define SOCKETS_DEBUG LWIP_DBG_ON in lwipops.h), rebuild and run the app and error is gone:
```
...
V (1241694) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF[0m
lwip_select(53, 0x0, 0x3ffbb540, 0x0, tvsec=10 tvusec=0)
lwip_selscan: fd=52 ready for writing
lwip_select: nready=1
lwip_send(52, data=0x3fff6b28, size=89, flags=0x0)
lwip_send(52) err=0 written=89
lwip_select(53, 0x3ffbb5d0, 0x0, 0x0, tvsec=1 tvusec=0)
lwip_selscan: fd=52 ready for reading
lwip_select: nready=1
D (1243224) MQTT_CLIENT: mqtt_task: client state :2[0m
lwip_select(53, 0x3ffbb510, 0x0, 0x0, tvsec=0 tvusec=0)
lwip_selscan: fd=52 ready for reading
lwip_select: nready=1
lwip_recvfrom(52, 0x3f847814, 1, 0x0, ..)
lwip_recv_tcp: top while sock->lastdata=0x0
lwip_recv_tcp: netconn_recv err=-14, pbuf=0x0
lwip_recv_tcp: p == NULL, error is "Connection reset."!
lwip_recvfrom(52):  addr=0.0.0.0 port=16379 len=-1
[0;31mE (1243254) MQTT_CLIENT: mqtt_message_receive: transport_read() error: errno=104[0m
[0;31mE (1243264) MQTT_CLIENT: mqtt_process_receive: mqtt_message_receive() returned -1[0m
lwip_close(52)
D (1243274) MQTT_CLIENT: Reconnect after 0 ms[0m
[0;32mI (1243274) MQTT: MQTT client reconnecting...(MQTT_EVENT_DISCONNECTED event with commented: esp_mqtt_client_reconnect(mqtt_client))[0m
[0;32mI (1243284) MQTT: MQTT_CLIENT_CONNECTED_BIT state: False[0m
D (1243294) MQTT_CLIENT: mqtt_task: client state :3[0m
D (1243294) MQTT_CLIENT: Reconnecting...[0m
D (1243304) MQTT_CLIENT: mqtt_task: client state :1[0m
...
```
In parallel issue thread (https://github.com/espressif/esp-idf/issues/3851) i'm trying to solve another connection problem with http and turning on SOCKETS_DEBUG also solved that problem too! Thus it can be argued that connection problems is highly likely in tcp_transport or lwip component. When I added sockets debug I also added some delay and locks or logic error(s) in tcp_transport or lwip gone away.
...
But solving one problem I've got another one: now each mqtt message resends much more than once. Here is the log example from client who subscribed to esp32 messages: 
```
2019/08/14 17:01:03 MSGID: 43333, Reciever:   1yi5q0xqwl, Sender:   1zrurftolf, Time: 2019-08-14 17:00:37 +0300 MSK,     MAC: B8:27:EB:C0:7C:C3, Type: EddyStone URL, RSSI: -55 dBm, RSSI0: -20 dBm, URL: https://github.com/
2019/08/14 17:01:04 MSGID: 43334, Reciever:   1yi5q0xqwl, Sender:   1zrurftolf, Time: 2019-08-14 17:00:37 +0300 MSK,     MAC: B8:27:EB:C0:7C:C3, Type: EddyStone URL, RSSI: -55 dBm, RSSI0: -20 dBm, URL: https://github.com/
2019/08/14 17:01:05 MSGID: 43335, Reciever:   1yi5q0xqwl, Sender:   1zrurftolf, Time: 2019-08-14 17:00:37 +0300 MSK,     MAC: B8:27:EB:C0:7C:C3, Type: EddyStone URL, RSSI: -55 dBm, RSSI0: -20 dBm, URL: https://github.com/
2019/08/14 17:01:07 MSGID: 43336, Reciever:   1yi5q0xqwl, Sender:   1zrurftolf, Time: 2019-08-14 17:00:37 +0300 MSK,     MAC: B8:27:EB:C0:7C:C3, Type: EddyStone URL, RSSI: -55 dBm, RSSI0: -20 dBm, URL: https://github.com/
```
Here is full log of esp32:
[beacon.zip](https://github.com/espressif/esp-mqtt/files/3501705/beacon.zip)

Most of messages sends much more than once. I can solve the problem of dublicated messages by setting QOS2, but it solves the problem only for reciever who sibscribed that messages, because mosquitto will dedublicate them, but doesn't solve the problem of overloading thin communication channel between esp32 and network.
....
With QOS2 got the same problem: each message resends much more than once.

How else can I help you to find the key problem because turning on lwip debug logging is not a solution?

Thanks in advance

david-cermak, 2019-08-14:
@no1seman Not only adding a delay but also reducing the poll timeout (sorry wasn't very clear from my previous post) 

**esp_transport_poll_read(client->transport, 0)**

```
if (MQTT_STATE_CONNECTED == client->state) {
            vTaskDelay(MQTT_POLL_READ_TIMEOUT_MS / portTICK_PERIOD_MS);
            if (esp_transport_poll_read(client->transport, 0) < 0) {
```

Adding delays (enabling logging) might lower the changes of the issue to appear but will not solve it. These are very good pointers though, and I believe that reducing the poll timeout should also help (There's is a bug in `tcp_transport` component in poll functions which should also wait with `select` on error `fd`, a fix's coming soon to IDF master; seems this is related..)

no1seman, 2019-08-14:
@david-cermak,

adding ONLY vTaskDelay(MQTT_POLL_READ_TIMEOUT_MS / portTICK_PERIOD_MS); doesn't help but adding vTaskDelay(MQTT_POLL_READ_TIMEOUT_MS / portTICK_PERIOD_MS); and turning on socket debug - does. 

OK, will wait for fix in tcp_transpot - then I'll make full tests again.

What will you say about multiple mqtt messages resend? This problem is also relates to tcp_transport bug or not?

david-cermak, 2019-08-14:
Could you please test one more scenario? Please revert all these changes and apply the following patch to `esp-mqtt` repo against master 
[0001-temporary-test-commit-to-check-if-read_poll-causes-i.patch.txt](https://github.com/espressif/esp-mqtt/files/3502841/0001-temporary-test-commit-to-check-if-read_poll-causes-i.patch.txt)

> What will you say about multiple mqtt messages resend? This problem is also relates to tcp_transport bug or not?

Resending logic is very basic, i.e. tries to resend everything (which was transmitted correctly) after 1 second, so when adding additional 1 second delay it is expected to see a lot of resends. Please note that this is not a common use case, just a adding temporary test modification to find the root cause of the issue.

no1seman, 2019-08-14:
@david-cermak,

Changes made:
1. reverted all esp-idf changes to v4.0-dev-1443-g39f090a4f (mqtt lib to dc37d3a0)
2. Applied provided patch:
Removed:
        if (MQTT_STATE_CONNECTED == client->state) {
            if (esp_transport_poll_read(client->transport, MQTT_POLL_READ_TIMEOUT_MS) < 0) {
                ESP_LOGE(TAG, "Poll read error: %d, aborting connection", errno);
                esp_mqtt_abort_connection(client);
            }
        }

 Added instead:
 vTaskDelay(50 / portTICK_PERIOD_MS);

got the following negative result:
```
W (3061131) MQTT: PUBLISHING BLE beacon_packet_type: 4
D (3061131) MQTT_CLIENT: mqtt_enqueue id: 9783, type=3 successful
D (3061131) OUTBOX: ENQUEUE msgid=9783, msg_type=3, len=87, size=87
V (3061151) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
D (3061201) MQTT_CLIENT: mqtt_message_receive: first byte: 0x50
D (3061201) MQTT_CLIENT: mqtt_message_receive: read "remaining length" byte: 0x2
D (3061201) MQTT_CLIENT: mqtt_message_receive: total message length: 4 (already read: 2)
D (3061211) MQTT_CLIENT: mqtt_message_receive: read_len=2
D (3061221) MQTT_CLIENT: mqtt_message_receive: transport_read():4 4
D (3061221) MQTT_CLIENT: msg_type=5, msg_id=9783
D (3061231) MQTT_CLIENT: received MQTT_MSG_TYPE_PUBREC
V (3061281) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (3061331) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
D (3061381) MQTT_CLIENT: mqtt_message_receive: first byte: 0x70
D (3061381) MQTT_CLIENT: mqtt_message_receive: read "remaining length" byte: 0x2
D (3061381) MQTT_CLIENT: mqtt_message_receive: total message length: 4 (already read: 2)
D (3061391) MQTT_CLIENT: mqtt_message_receive: read_len=2
D (3061401) MQTT_CLIENT: mqtt_message_receive: transport_read():4 4
D (3061401) MQTT_CLIENT: msg_type=7, msg_id=9783
D (3061411) MQTT_CLIENT: received MQTT_MSG_TYPE_PUBCOMP
D (3061411) MQTT_CLIENT: pending_id=9783, pending_msg_count = 1
D (3061421) OUTBOX: DELETED msgid=9783, msg_type=3, remain size=0
D (3061421) MQTT_CLIENT: Receive MQTT_MSG_TYPE_PUBCOMP, finish QoS2 publish
D (3061431) MQTT: MQTT message published: msg_id=9783
V (3061491) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (3061541) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (3061591) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (3061641) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (3061691) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (3061741) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (3061791) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (3061841) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (3061891) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (3061941) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (3061991) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (3062041) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (3062091) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (3062141) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (3062191) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (3062241) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (3062291) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (3062341) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (3062391) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (3062441) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (3062491) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (3062541) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (3062591) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (3062641) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (3062691) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (3062741) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (3062791) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
V (3062841) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF
[... 300 lines cut ...]
D (3086981) MQTT_CLIENT: Publish: client is not connected
E (3086991) MQTT: MQTT client is not connected
E (3107011) MQTT: MQTT client is not connected
E (3107011) MQTT: MQTT client is not connected
E (3127021) MQTT: MQTT client is not connected
E (3127021) MQTT: MQTT client is not connected
E (3147031) MQTT: MQTT client is not connected
E (3147031) MQTT: MQTT client is not connected
E (3167041) MQTT: MQTT client is not connected
E (3167041) MQTT: MQTT client is not connected
E (3187051) MQTT: MQTT client is not connected
E (3187051) MQTT: MQTT client is not connected
E (3207061) MQTT: MQTT client is not connected
E (3207061) MQTT: MQTT client is not connected
E (3227071) MQTT: MQTT client is not connected
E (3227071) MQTT: MQTT client is not connected
E (3247081) MQTT: MQTT client is not connected
E (3247081) MQTT: MQTT client is not connected
E (3267091) MQTT: MQTT client is not connected
E (3267091) MQTT: MQTT client is not connected
E (3287101) MQTT: MQTT client is not connected
E (3287101) MQTT: MQTT client is not connected
```
Full log:
[mqtt2.zip](https://github.com/espressif/esp-mqtt/files/3503275/mqtt2.zip)

david-cermak, 2019-08-15:
Thanks very much @no1seman for this test! This however confirms that my initial suspicion was wrong (and most probably the tcp_transport fix won't help either).

May I ask you for another go with this patch (just adding more logs) against the same versions as before?
[0001-instrumenting-client-code.patch.txt](https://github.com/espressif/esp-mqtt/files/3504688/0001-instrumenting-client-code.patch.txt)
Or possibly (if you can connect a debugger easily) ask for a backtrace of mqtt thread when the issue occurs? Sorry for the inconvenience, will have to find a way to reproduce the issue but no luck so far.

no1seman, 2019-08-15:
@david-cermak,
no problem:
[mqtt3.zip](https://github.com/espressif/esp-mqtt/files/3505064/mqtt3.zip)

PS It's not easy to conect debugger because GPIO12-GPIO15 is used for connecting sd card.

david-cermak, 2019-08-15:
Thanks for the logs! To be honest I'm a bit at a loss now, since this log is pointing again to the poll timeout the same way I thought before. Don't you have some high prio thread running in the background (for example waiting for the connection flag) blocking the mqtt task? No idea beside that I am afraid...

no1seman, 2019-08-15:
@david-cermak,
I change tasks priority while tried to find out the problem by myself, but during the last test tasklist looks like the following:
[image]

mqtt_send uses only 2 of them: BEACONS and BLINK and http & mqtt has the same priority on a single core

no1seman, 2019-08-15:
@david-cermak,
here i posted a test app with http problem: https://github.com/espressif/esp-idf/issues/3851#issuecomment-521655145
A little bit later I'll try to make the same for mqtt bug.

david-cermak, 2019-08-15:
@no1seman 
Thanks that would be really helpful!

From previous post, it looks like a heathly application, but does the picture show task usage under 
normal conditions or after the terminal disconnect takes place?

My main concern was how the task distribution looks like when the issue appears 
(when mqtt_task gets stuck), if some if/else program path busy waits for the connection
flag in some thread. This may sound reasonable as reconnect works on "Read Error" 
(from mqtt_task) but doesn't on "Write Error" (from user task - publish).
But have no clue at the moment, just guessing...

no1seman, 2019-08-15:
@david-cermak,
tasklist struct list creates every 10 seconds in BLINK task and then, when web-app requests, esp32 serialise tasklist data to json and handle it to web-app. As far as I watched during my tests there is no any diffrence between normal conditions and after the terminal disconnect, may be mqtt task has a little more CPU usage

no1seman, 2019-08-16:
@david-cermak,
finally I finished test app, you can get it here: https://github.com/no1seman/mqtt_bug
I tried to reproduce the basic logic, settings and so on of my "big app" as accurately as possible, thats why mqtt_task1() and mqtt_task2() have different task priority (but seems to me that it doesn't be a promblem). Aslo, you will find mqtt client settings uses in source code. 

During one of the tests I 've got the following result:
```
I (501754) example: PUBLISHING CLIMATE
I (507024) example: PUBLISHING HTTP MESSAGE
I (507204) example: PUBLISHING HTTP MESSAGE
W (509114) httpd_txrx: httpd_sock_err: error in send : 104
I (509114) example: Sending file 'index.js.gz' failed
W (509114) httpd_uri: httpd_uri: uri handler execution failed
I (511754) example: PUBLISHING STATS
E (517204) MQTT_CLIENT: Error write data or timeout, written len = 0, errno=0
I (517204) example: MQTT_EVENT_DISCONNECTED
W (517204) example: MQTT client reconnecting...(MQTT_EVENT_DISCONNECTED event with commented: esp_mqtt_client_reconnect(mqtt_client))
I (517214) example: MQTT_CLIENT_CONNECTED_BIT state: False
I (517224) MQTT_CLIENT: Client force reconnect requested
I (517224) example: MQTT client is not connected
W (521344) httpd_txrx: httpd_sock_err: error in send : 104
I (521344) example: Sending file 'index.js.gz' failed
W (521354) httpd_uri: httpd_uri: uri handler execution failed
W (521694) httpd_txrx: httpd_sock_err: error in send : 104
I (521694) example: Sending file 'index.js.gz' failed
W (521694) httpd_uri: httpd_uri: uri handler execution failed
I (527234) example: MQTT client is not connected
I (527234) example: MQTT client is not connected
```
Full log:
[mqtt_bug.zip](https://github.com/espressif/esp-mqtt/files/3509580/mqtt_bug.zip)

Prerequisites:
Any ESP32 board with at least 4Mb FLASH and 4Mb SPI PSRAM
Esp-idf: v4.0-dev-1443-g39f090a4f
Build method: cmake
Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
MQTT server: mosquitto 1.6.4
Linux host: Any, I tested on Ubuntu 18.04.3 with latest updates 
Read my comments here: https://github.com/espressif/esp-idf/issues/3851#issuecomment-521894668

How to use:
1. Set WiFi SSID and pasword in menuconfig
2. Set flash settings if needed in menuconfig
3. Build & flash & monitor (run_app.sh provided in project root dir :-))
4. In separate terminal run (linux host needed):
'./http_test.sh -s IP_ADDRESS_OF_ESP -i NUMBER_OF_TEST_ITRATIONS -d NAME_OF_TEMPORARY_DIR'
(Example: './http_test.sh -s 192.168.3.130 -i 10000 -d tmp')

After some time you have to get an error. Script http_test.sh needed to provoke an error by increasing the load so that you do not have to wait for days when an event occurs that leads to an error.

It would be great if you will find a problem or tell me where I'm wrong.
Thanks in advance

david-cermak, 2019-08-17:
Thanks very much for this test code @no1seman!

I was indeed able to recreate the issue. The root cause seems to be resolved in https://github.com/espressif/esp-mqtt/commit/cbae6343e9bd91670c4146ea76622306164b320b.
I briefly mentioned about this issue in my first post with relation to `esp_mqtt_client_stop()`, but haven't taken needed care to that later even if you specifically stated your commit id, sorry for that (missed the important fact that the test app might call `esp_mqtt_client_stop()` as registered event handler)

Could you please manually update mqtt lib to the latest master (to include this commit https://github.com/espressif/esp-mqtt/commit/cbae6343e9bd91670c4146ea76622306164b320b) 
 and test it on your end?

no1seman, 2019-08-18:
@david-cermak,

thank you for investigating the issue.

I already did the update earlier and it was no effect: https://github.com/espressif/esp-mqtt/issues/126#issuecomment-520878948 item 1.

I made update of mqtt lib again to https://github.com/espressif/esp-mqtt/commit/92aa01deb80fd9dea46f18f493ddd03726f8826a (merge for https://github.com/espressif/esp-mqtt/commit/cbae6343e9bd91670c4146ea76622306164b320b):
[image]

and I've got the same error:
```
I (28) boot: ESP-IDF v4.0-dev-1443-g39f090a4f-dirty 2nd stage bootloader
I (29) boot: compile time 08:03:49
I (39) boot: Enabling RNG early entropy source...
I (39) boot: SPI Speed      : 40MHz
I (39) boot: SPI Mode       : DIO
I (43) boot: SPI Flash Size : 4MB
I (47) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (73) boot:  2 factory          factory app      00 00 00010000 00100000
I (81) boot: End of partition table
I (85) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x1d070 (118896) map
I (136) esp_image: segment 1: paddr=0x0002d098 vaddr=0x3ffb0000 size=0x02f78 ( 12152) load
I (142) esp_image: segment 2: paddr=0x00030018 vaddr=0x400d0018 size=0x80444 (525380) map
0x400d0018: _stext at ??:?

I (331) esp_image: segment 3: paddr=0x000b0464 vaddr=0x3ffb2f78 size=0x0020c (   524) load
I (332) esp_image: segment 4: paddr=0x000b0678 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /home/user/esp/esp-mdf/esp-idf/components/freertos/xtensa_vectors.S:1778

I (338) esp_image: segment 5: paddr=0x000b0a80 vaddr=0x40080400 size=0x15b9c ( 88988) load
I (398) boot: Loaded app from partition at offset 0x10000
I (398) boot: Disabling RNG early entropy source...
I (398) psram: This chip is ESP32-D0WD
I (404) spiram: Found 64MBit SPI RAM device
I (408) spiram: SPI RAM mode: flash 40m sram 40m
I (413) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (420) cpu_start: Pro cpu up.
I (424) cpu_start: Application information:
I (429) cpu_start: Project name:     simple
I (433) cpu_start: App version:      c577ed5-dirty
I (439) cpu_start: Compile time:     Aug 18 2019 08:03:44
I (445) cpu_start: ELF file SHA256:  f4c01000a38e2df4...
I (451) cpu_start: ESP-IDF:          v4.0-dev-1443-g39f090a4f-dirty
I (458) cpu_start: Starting app cpu, entry point is 0x40081508
0x40081508: call_start_cpu1 at /home/user/esp/esp-mdf/esp-idf/components/esp32/cpu_start.c:281

I (0) cpu_start: App cpu up.
I (470) heap_init: Initializing. RAM available for dynamic allocation:
I (475) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (481) heap_init: At 3FFB4A80 len 0002B580 (173 KiB): DRAM
I (488) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (494) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (500) heap_init: At 40095F9C len 0000A064 (40 KiB): IRAM
I (507) cpu_start: Pro cpu start user code
I (511) spiram: Adding pool of 4077K of external SPI memory to heap allocator
I (533) spi_flash: detected chip: generic
I (534) spi_flash: flash io: dio
W (534) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (544) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (555) spiram: Reserving pool of 63K of internal memory for DMA/internal allocations
I (565) example: Caching HTTP static files...
I (575) example: index.html.gz file md5 summ: 369d229e9a9b07b4f7ec0ecbb8e3884b
I (585) example: index.css.gz file md5 summ: 21ce0035727356e1135a02ef104bcf0d
I (595) example: index.js.gz file md5 summ: 7f0fce94cc13d4c869c5b9052ff6dc3c
I (595) example: Caching HTTP static files completed
I (625) example: MQTT Client init complete
[... 54 lines cut ...]
I (97105) example: MQTT client is not connected
```
May be that is the effect of SPI SPRAM error, but for now problem still exists.

It's very strange that in case of  **MQTT_CLIENT: No PING_RESP, disconnected' - every thing OK**:
```
E (159270) MQTT_CLIENT: No PING_RESP, disconnected
D (159270) MQTT_CLIENT: Reconnect after 10000 ms
I (159280) MQTT: MQTT client reconnecting...(MQTT_EVENT_DISCONNECTED event with commented: esp_mqtt_client_reconnect(mqtt_client))
I (159290) MQTT: MQTT_CLIENT_CONNECTED_BIT state: False
E (169030) MQTT: MQTT client is not connected
E (169030) MQTT: MQTT client is not connected
D (169290) MQTT_CLIENT: Reconnecting...
D (169290) MQTT: MQTT_EVENT_BEFORE_CONNECT
D (169310) MQTT_CLIENT: Transport connected to mqtt://192.168.3.211:1883
```
but in case: E (87075) MQTT_CLIENT: Error write data or timeout, written len = 0, errno=0 I got:
```
E (87075) MQTT_CLIENT: Error write data or timeout, written len = 0, errno=0
I (87075) example: MQTT_EVENT_DISCONNECTED
W (87075) example: MQTT client reconnecting...(MQTT_EVENT_DISCONNECTED event with commented: esp_mqtt_client_reconnect(mqtt_client))
I (87085) example: MQTT_CLIENT_CONNECTED_BIT state: False
I (87095) MQTT_CLIENT: Client force reconnect requested
I (87095) example: MQTT client is not connected
```

david-cermak, 2019-08-19:
alright, in that case there must be really something environmental related which I cannot reproduce.
I noted your mentioning about the update, but wasn't sure and the only way to reproduce the issue was to use earlier version which resulted in the same error log...

may you still try to apply this patch to 92aa01deb80fd9dea46f18f493ddd03726f8826a and share the results (trying to set watchpoints to run state in order to find out who's making the mqtt_loop stop)?
[0001-added-watchpoint-for-client-run-and-instrumenting-cl.patch.txt](https://github.com/espressif/esp-mqtt/files/3516386/0001-added-watchpoint-for-client-run-and-instrumenting-cl.patch.txt)

no1seman, 2019-08-19:
@david-cermak,
no problem!

So, I applied https://github.com/espressif/esp-mqtt/commit/92aa01deb80fd9dea46f18f493ddd03726f8826a and got:
```
[0;32mI (10563323) example: PUBLISHING CLIMATE MESSAGE[0m
V (10563333) MQTT_CLIENT: esp_mqtt_client_publish: 1295[0m
V (10563333) MQTT_CLIENT: esp_mqtt_client_publish: 1298 state:2[0m
D (10563343) MQTT_CLIENT: mqtt_enqueue id: 44905, type=3 successful[0m
D (10563353) OUTBOX: ENQUEUE msgid=44905, msg_type=3, len=44, size=423[0m
V (10563873) MQTT_CLIENT: esp_mqtt_task: 1136[0m
V (10563873) MQTT_CLIENT: esp_mqtt_task: 992[0m
V (10563873) MQTT_CLIENT: esp_mqtt_task: 1032[0m
V (10563873) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF[0m
V (10563883) MQTT_CLIENT: esp_mqtt_task: 1050[0m
V (10563883) MQTT_CLIENT: esp_mqtt_task: 1058[0m
V (10563893) MQTT_CLIENT: esp_mqtt_task: 1087[0m
V (10563893) MQTT_CLIENT: esp_mqtt_task: 1126[0m
V (10563903) MQTT_CLIENT: esp_mqtt_task: 1128[0m
V (10563903) MQTT_CLIENT: esp_mqtt_task: 1130[0m
D (10564693) httpd: httpd_server: processing listen socket 54[0m
D (10564693) httpd: httpd_accept_conn: newfd = 58[0m
D (10564693) httpd_sess: httpd_sess_new: fd = 58[0m
D (10564703) httpd: httpd_accept_conn: complete[0m
D (10564703) httpd: httpd_server: doing select maxfd+1 = 59[0m
D (10564713) httpd: httpd_server: processing socket 58[0m
D (10564713) httpd_sess: httpd_sess_process: httpd_req_new[0m
D (10564723) httpd_txrx: httpd_recv_with_opt: requested length = 128[0m
D (10564723) httpd_txrx: httpd_recv_with_opt: received length = 117[0m
D (10564733) httpd_parse: read_block: received HTTP request block size = 117[0m
D (10564743) httpd_parse: cb_url: message begin[0m
D (10564743) httpd_parse: cb_url: processing url = /index.js[0m
D (10564753) httpd_parse: verify_url: received URI = /index.js[0m
D (10564753) httpd_parse: cb_header_field: headers begin[0m
D (10564763) httpd_txrx: httpd_unrecv: length = 93[0m
D (10564763) httpd_parse: pause_parsing: paused[0m
D (10564773) httpd_parse: cb_header_field: processing field = Host[0m
D (10564773) httpd_txrx: httpd_recv_with_opt: requested length = 128[0m
D (10564783) httpd_txrx: httpd_recv_with_opt: pending length = 93[0m
D (10564793) httpd_parse: read_block: received HTTP request block size = 93[0m
D (10564793) httpd_parse: continue_parsing: skip pre-parsed data of size = 5[0m
D (10564803) httpd_parse: continue_parsing: un-paused[0m
D (10564813) httpd_parse: cb_header_value: processing value = 192.168.3.130[0m
D (10564813) httpd_parse: cb_header_field: processing field = User-Agent[0m
D (10564823) httpd_parse: cb_header_value: processing value = curl/7.58.0[0m
D (10564833) httpd_parse: cb_header_field: processing field = Accept[0m
D (10564833) httpd_parse: cb_header_value: processing value = */*[0m
D (10564843) httpd_parse: cb_header_field: processing field = Accept-Encoding[0m
D (10564853) httpd_parse: cb_header_value: processing value = gzip, deflate[0m
D (10564853) httpd_parse: cb_headers_complete: bytes read     = 117[0m
D (10564863) httpd_parse: cb_headers_complete: content length = 0[0m
D (10564873) httpd_parse: pause_parsing: paused[0m
D (10564873) httpd_parse: cb_no_body: message complete[0m
D (10564883) httpd_parse: httpd_parse_req: parsing complete[0m
D (10564883) httpd_uri: httpd_uri: request for /index.js with type 1[0m
D (10564893) httpd_uri: httpd_find_uri_handler: [0] = /*[0m
D (10564893) httpd_txrx: httpd_resp_set_hdr: new header = Content-Encoding: gzip[0m
V (10564913) MQTT_CLIENT: esp_mqtt_task: 1136[0m
V (10564913) MQTT_CLIENT: esp_mqtt_task: 992[0m
V (10564913) MQTT_CLIENT: esp_mqtt_task: 1032[0m
V (10564923) MQTT_CLIENT: mqtt_message_receive: transport_read(): no data or EOF[0m
D (10564923) httpd_txrx: httpd_resp_set_hdr: new header = MD5HASH: 8ebd9f1b78ba9b590e9ee7068ef3e018[0m
V (10564933) MQTT_CLIENT: esp_mqtt_task: 1050[0m
V (10564943) MQTT_CLIENT: esp_mqtt_task: 1054[0m
V (10564943) MQTT_CLIENT: esp_mqtt_task: 1058[0m
D (10564953) httpd_txrx: httpd_send_all: sent = 93[0m
V (10564953) MQTT_CLIENT: esp_mqtt_task: 1087[0m
V (10564953) MQTT_CLIENT: esp_mqtt_task: 1126[0m
D (10564963) httpd_txrx: httpd_send_all: sent = 16[0m
V (10564963) MQTT_CLIENT: esp_mqtt_task: 1128[0m
[... 68 lines cut ...]
D (10584833) httpd_parse: continue_parsing: skip pre-parsed data of size = 5[0m
D (10584843) httpd_parse: continue_parsing: un-paused[0m
D (10584853) httpd_parse: cb_header_value: processing value = 192.168.3.130[0m
D (10584853) httpd_parse: cb_header_field: processing field = User-Agent[0m
D (10584863) httpd_parse: cb_header_value: processing value = curl/7.58.0[0m
D (10584873) httpd_parse: cb_header_field: processing field = Accept[0m
D (10584873) httpd_parse: cb_header_value: processing value = */*[0m
D (10584883) httpd_parse: cb_header_field: processing field = Accept-Encoding[0m
D (10584893) httpd_parse: cb_header_value: processing value = gzip, deflate[0m
D (10584893) httpd_parse: cb_headers_complete: bytes read     = 119[0m
D (10584903) httpd_parse: cb_headers_complete: content length = 0[0m
D (10584913) httpd_parse: pause_parsing: paused[0m
D (10584913) httpd_parse: cb_no_body: message complete[0m
D (10584923) httpd_parse: httpd_parse_req: parsing complete[0m
D (10584923) httpd_uri: httpd_uri: request for /index.html with type 1[0m
D (10584933) httpd_uri: httpd_find_uri_handler: [0] = /*[0m
D (10584933) httpd_txrx: httpd_resp_set_hdr: new header = Content-Encoding: gzip[0m
D (10584953) httpd_txrx: httpd_resp_set_hdr: new header = MD5HASH: 93c18652bf6c6ca01f5aa0edc542fde8[0m
D (10584953) httpd_txrx: httpd_send_all: sent = 80[0m
```

Full log is here:
[mqtt.zip](https://github.com/espressif/esp-mqtt/files/3517360/mqtt.zip)

PS Tommorow, I'll try to move to release/4.0 with this patch: https://github.com/espressif/esp-idf/commit/b9a5f764baec31f4fabc8c148c08470e8f572ee2 , may be the problem is there!?

david-cermak, 2019-08-20:
@no1seman Thanks very much for all these tests!

Yet another attempt to get more info (this time it should break with backtrace information) Please if you could apply this patch again to the https://github.com/espressif/esp-mqtt/commit/92aa01deb80fd9dea46f18f493ddd03726f8826a and share results.
[0001-added-watchpoints-for-both-cores.patch.txt](https://github.com/espressif/esp-mqtt/files/3520297/0001-added-watchpoints-for-both-cores.patch.txt)

no1seman, 2019-08-20:
@david-cermak,
nothing for, I'm one of who interested in solving that issue.

So, I made some runs on v4.0-dev-1443-g39f090a4f esp-idf, updated mqtt lib to: https://github.com/espressif/esp-mqtt/commit/92aa01deb80fd9dea46f18f493ddd03726f8826a and provided patch with added watchpoins, got the following:
1. 
[mqtt_wp.zip](https://github.com/espressif/esp-mqtt/files/3521159/mqtt_wp.zip)
2. 
[mqtt_wp1.zip](https://github.com/espressif/esp-mqtt/files/3521162/mqtt_wp1.zip)
3.
[mqtt_wp2.zip](https://github.com/espressif/esp-mqtt/files/3521686/mqtt_wp2.zip)

I inserted backtrace addreses to decoded to lines at the end of each file. Wish that it helps

david-cermak, 2019-08-21:
Every time I look into the posted logs I'm getting even more confused then before (no exception this time). I can see that it behaves a bit differently then before (this time it tries to retrigger connection, after the default 10s though. before it just exits mqtt_loop), which really makes no sense to me and cannot think of anything but some ugly PSRAM issue. 

Is this reproducible on multiple boards or testing with just one?

no1seman, 2019-08-22:
@david-cermak,

seems that it's my fault, I looked inattentively on a disconnection reason! As a have already mentioned that mqtt client successfully reconnects after **MQTT_CLIENT: No PING_RESP, disconnected** and also it successfully reconects in other reasons except: **MQTT_CLIENT: Error write data or timeout, written len = 0, errno=0**. I'll try to catch that case.

david-cermak, 2019-08-22:
@no1seman 

> The problem is that after apllying you patch I can't see mentioned above error lines in log...

Yes, that's the point. The error scenario started the same way, but continued a bit differently from previous logs. Moreover it seems to assert when a memory next to the watchpoint was modified. Therefore I asked about more boards, modules where this could be reproduced.

> The problem will occur if write_len == 0 in:

Correct, but that's (very simple) logic of the mqtt library. Timeouts are treated as errors and disconnection is triggered (could perhaps make this timeout configurable). Anyway it should and must always reconnect.

> Also I shuld say that there are 2 places where mqtt_write_data() return state doesn't check in mqtt_process_receive()

True, similar logic needs to be applied to these places as well! Thanks for noting, this is to be fixed!

no1seman, 2019-08-22:
@david-cermak,
I'm testing right now and got the same behaivour on different boards: DevKit4.1, TTGO T8 v1.7 and others. Thats not the problem of the particular board.

When I will catch the bug - I'll post the results

no1seman, 2019-08-22:
@david-cermak,

I've got some results. Tested on a brand new board TTGO T8 v1.7 with espressif 64Mbit PSRAM.
Here is two runs in one log divided by HW reset. I was made HW reset because after MQTT_CLIENT: 
Error write data or timeout, written len = 0, errno=0 app didn't halted with Core dump: 
[log.zip](https://github.com/espressif/esp-mqtt/files/3530925/log.zip)

In first run I blocked for 30 seconds wifi connection of ESP on my WiFi-router and watchpoint triggered and in second run I've got **MQTT_CLIENT: Error write data or timeout, written len = 0, errno=0** and watchpoint didn't trigger.

So, the problem is that when **E (7212) MQTT_CLIENT: Error write data or timeout, written len = -1, errno=113** occur mqtt lib set client->run = false and when **MQTT_CLIENT: Error write data or timeout, written len = 0, errno=0** doesn't.

I set esp_backtrace_print(20); rigth after: ESP_LOGE(TAG, "Error write data or timeout, written len = %d, errno=%d", write_len, errno); and got:
```
0x400e6117: mqtt_write_data at /home/user/esp/esp-mdf/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:567 (discriminator 2)
0x400e7cd4: esp_mqtt_client_publish at /home/user/esp/esp-mdf/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:1353
0x400d98b0: mqtt_task1 at /home/user/esp/mqtt_bug/build/../main/main.c:372 (discriminator 9)
0x40092d85: vPortTaskWrapper at /home/user/esp/esp-mdf/esp-idf/components/freertos/port.c:143
```

```
...
        if (mqtt_write_data(client) != ESP_OK) {
            esp_mqtt_abort_connection(client);
            ESP_LOGV(TAG, "%s: %d state:%d", __func__, __LINE__, client->state);
            goto cannot_publish;
...
        }
```
in esp_mqtt_abort_connection():
```
static esp_err_t esp_mqtt_abort_connection(esp_mqtt_client_handle_t client)
{
    ESP_LOGV(TAG, "%s: %d", __func__, __LINE__);
    printf("esp_transport_close() result: %d\n", esp_transport_close(client->transport));
    client->wait_timeout_ms = MQTT_RECONNECT_TIMEOUT_MS;
    client->reconnect_tick = platform_tick_get_ms();
    client->state = MQTT_STATE_WAIT_TIMEOUT;
    ESP_LOGD(TAG, "Reconnect after %d ms", client->wait_timeout_ms);
    client->event.event_id = MQTT_EVENT_DISCONNECTED;
    client->wait_for_ping_resp = false;
    esp_mqtt_dispatch_event_with_msgid(client);
    return ESP_OK;
}
```
I've got return code: -1 when **MQTT_CLIENT: Error write data or timeout, written len = 0, errno=0** occur, and in any other cases (for exmple if I stop mosquitto service on server) - I've got return code: 0
Full log:
[mqtt_issue1.zip](https://github.com/espressif/esp-mqtt/files/3531227/mqtt_issue1.zip)

Aslo I can't understand how it would be:
D (17713) MQTT_CLIENT: PING sent <====== sending PING
V (17713) MQTT_CLIENT: esp_mqtt_task: 1103
V (17723) MQTT_CLIENT: esp_mqtt_task: 1143 state: 1
V (17723) MQTT_CLIENT: esp_mqtt_task: 1145 state: 1
V (17733) MQTT_CLIENT: esp_mqtt_task: 1147 state: 1
V (17733) MQTT_CLIENT: esp_mqtt_task: 1153 state: 1
V (17743) MQTT_CLIENT: esp_mqtt_task: 1008 state: 1
V (17743) MQTT_CLIENT: esp_mqtt_task: 1048
D (17753) MQTT_CLIENT: mqtt_message_receive: first byte: 0xd0
D (17753) MQTT_CLIENT: mqtt_message_receive: read "remaining length" byte: 0x0
D (17763) MQTT_CLIENT: mqtt_message_receive: total message length: 2 (already read: 2)
D (17773) MQTT_CLIENT: mqtt_message_receive: transport_read():2 2
D (17773) MQTT_CLIENT: msg_type=13, msg_id=0
D (17783) MQTT_CLIENT: MQTT_MSG_TYPE_PINGRESP <===== Resieving PING RESP
V (17783) MQTT_CLIENT: esp_mqtt_task: 1074
V (17793) MQTT_CLIENT: esp_mqtt_task: 1103
V (17793) MQTT_CLIENT: esp_mqtt_task: 1143 state: 0 <===== client->state == UNKONOWN
V (17803) MQTT_CLIENT: esp_mqtt_task: 1145 state: 0
V (17803) MQTT_CLIENT: esp_mqtt_task: 1147 state: 0
V (18813) MQTT_CLIENT: esp_mqtt_task: 1153 state: 0
V (18813) MQTT_CLIENT: esp_mqtt_task: 1156 state: 0
[... 13 lines cut ...]
0x400e7d0c: esp_mqtt_client_publish at /home/user/esp/esp-mdf/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:1354

0x400d99c2: mqtt_task2 at /home/user/esp/mqtt_bug/build/../main/main.c:419 (discriminator 9)

0x40092d85: vPortTaskWrapper at /home/user/esp/esp-mdf/esp-idf/components/freertos/port.c:143

V (32263) MQTT_CLIENT: esp_mqtt_abort_connection: 341
esp_transport_close() result: -1
D (32273) MQTT_CLIENT: Reconnect after 10000 ms
D (32283) MQTT_CLIENT: client->event.msg_id: 0, client->event: 2  
I (32283) example: MQTT_EVENT_DISCONNECTED <====== only now got the message that mqtt client disconnected
W (32293) example: MQTT client reconnecting...(MQTT_EVENT_DISCONNECTED event with commented: esp_mqtt_client_reconnect(mqtt_client))
I (32303) example: MQTT_CLIENT_CONNECTED_BIT state: False
V (32313) MQTT_CLIENT: esp_mqtt_client_publish: 1356 state:3
V (32313) MQTT_CLIENT: esp_mqtt_client_publish: 1405 state:3
V (32323) MQTT_CLIENT: esp_mqtt_client_publish: 1315 state:3
I (32323) example: MQTT client is not connected
D (32323) MQTT_CLIENT: mqtt_enqueue id: 42907, type=3 successful
D (32333) MQTT_CLIENT: Publish: client is not connected
V (32343) MQTT_CLIENT: esp_mqtt_client_publish: 1405 state:3
I (42343) example: MQTT client is not connected
I (42343) example: MQTT client is not connected
I (52343) example: MQTT client is not connected

So, if you have any ideas - I will be ready to help

david-cermak, 2019-08-22:
Thanks for posting this log. (again no exception wrtt the level of my confusedness) this looks very very strange...

Assuming your using the latest patch (+ the added bt_print) with watchpoints enabled (can see in the log). The logs from lines 1143-1160 print values of `client->run` so actual issue is already in here:
```
V (17793) MQTT_CLIENT: esp_mqtt_task: 1143 state: 0 <===== client->state == UNKONOWN
```
that doen;t mean `client->state == UNKONOWN` however but `client->run == false` which effectively stops mqtt_task() -- nothing new btw, seen that from your very first  log posted here (and that is the very simple reason why the thing doesnt reconnect). 

The only trouble is to find out why the client->run changes and why the watchpoint is not triggered while it's state is changed. Cannot think of any reason...

no1seman, 2019-08-22:
@david-cermak,

that "The only trouble is to find out why the client->run changes and why the watchpoint is not triggered while it's state is changed. Cannot think of any reason..." means, that on different boards I've got the same problem with memory, probaly External RAM.

Here is my sdkconfig: https://github.com/no1seman/mqtt_bug/blob/master/sdkconfig? may be the problem is there? For example due wrong settings some errors in caching?

david-cermak, 2019-08-22:
@no1seman exactly. 

Thanks for posting the sdkconfig, can see the important option here `CONFIG_SPIRAM_CACHE_WORKAROUND=y` is set, but will try to apply your config entirely to recreate the issue on my side. This is the key point for moving forward and should not be that difficult if it's happening on multiple different boards on your end...

no1seman, 2019-08-22:
@david-cermak,

thank you very much for investigation of my problem!

Also, I have to remind you about another issue, related to esp_http: [](https://github.com/espressif/esp-idf/issues/3851) . Without PSRAM http problems also going away.

PS After my previous post I run test on the same board but without SPIRAM. Everything works fine,  so the problem 100% in SPIRAM and GCC8.2.0, because I recall that problems began after upgrading toolchain from: xtensa-esp32-elf-linux64-1.22.0-80-g6c4433a-5.2.0.tar.gz to xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0 (installed by $IDF_PATH/install.sh)

no1seman, 2019-08-23:
@david-cermak,

this issue IMHO should be linked to this one: https://github.com/espressif/esp-idf/issues/3624#issuecomment-515155377

david-cermak, 2019-08-27:
@no1seman 
I was able to reproduce the issue and indeed it seems to be the same root cause as #2892.

This particular issue of mqtt library not reconnecting might be even worked around with changing bools to integers in client config:
```
struct esp_mqtt_client {
...
    int run;
    int wait_for_ping_resp;
...
}
```
That would however consequently fail somewhere else such as LWIP...

I would suggest to subscribe to #2892 to get updates on the underlying issue related to PSRAM.

no1seman, 2019-08-27:
@david-cermak,
thanks a lot, great job!

Are you plannig to commit that changes to IDF SDK? Seems that it would be not great cost to lose 6  bytes but to make lib more stable.

PS As for #2892, I'm do not hope that this issue will be finally solved ever with rev 1 ESP32D. More than 2 years passed (section #3.9 added into eco bugs at june 2017!) and final solution is still not out. Now I'm refactoring my apps to make it runnable without PSRAM and after it I'll continue to make stress tests. Also will wait for reev 2 and 3, may be there PSRAM will be fully functional.

david-cermak, 2019-09-05:
@no1seman No, I don't think I will update the mqtt library to use `int`s instead of `bool`s. No concern on losing bytes, but this is not a systematic change and even if it might help avoiding failures in the library it would fail somewhere else.

Thank you too, for the active approach to this issue and mainly for creating the test code!

no1seman, 2019-09-05:
@david-cermak,

Ok, seems that this issue must be closed. I'm busy now for refactoring my app. If I will face such a problem later, without SPI RAM - I'll reopen it.

closed 2019-09-05 by no1seman