espressif/esp-mqtt #128: Not working esp_mqtt_client with aws

dhaval2017, 2019-09-04:
How can I connect common mqtt client that connects aws and other also?

david-cermak, 2019-09-05:
Hi @dhaval2017 

Please check the IDF examples `ssl` and `ssl_mutual_auth` and combine them together, so the client config would look something like:
```
    const esp_mqtt_client_config_t mqtt_cfg = {
        .client_id = "<DEVICE_ID>",
        .uri = "mqtts://<ENDPOINT>...st-1.amazonaws.com:8883",
        .client_cert_pem = (const char *)client_cert_pem_start,
        .client_key_pem = (const char *)client_key_pem_start,
        .cert_pem = (const char *)server_cert_pem_start,
    };
```
Then subscribe or publish to standard aws topics, for example:
```
esp_mqtt_client_publish(client, "$aws/things/<DEVICE_ID>/shadow/get", "", 0, 0, 0);
```

An alternative might be using aws iot repo here https://github.com/espressif/esp-aws-iot

dhaval2017, 2019-09-06:
I did as your suggestion its giving me error this on connect
E (75404) MQTT_CLIENT: mqtt_message_receive: transport_read() error: errno=0
E (75404) MQTT_CLIENT: esp_mqtt_connect: mqtt_message_receive() returned -1

david-cermak, 2019-09-06:
My initial guess is that your client id is wrong, since the tcp/ssl seem to connect, but getting error on Mqtt receive (assuming response to mqtt connect request message)

I would suggest to test the connection with python first to be sure you get all the keys correct
```
client = paho.Client(id="<DEVICE_ID>")
client.on_connect = on_connect
client.on_message = on_message
client.tls_set("server_cert.pem", 
                certfile="client_cert.pem", 
                keyfile="client_key.pem", 
                cert_reqs=ssl.CERT_REQUIRED, 
                tls_version=ssl.PROTOCOL_TLSv1_2, 
                ciphers=None)

res = client.connect("<ENDPOINT>...st-1.amazonaws.com", 8883, keepalive=120)
```

If you're able to connect (getting on_conect callback) and publish/subscribe successfully with python and failing with the esp-mqtt client, would you please provide the versions of IDF (and mqtt) you using and more verbose log?

dhaval2017, 2019-09-06:
Got working now Thanks

closed 2019-09-06 by dhaval2017