espressif/esp-mqtt #135: merge latest commits on esp-idf release/v4.0 branch

enricop, 2019-10-04:
Hi,

for our project we are using [release/4.0](https://github.com/espressif/esp-idf/tree/release/v4.0) branch. I see that on some features you already use internal checks for enabling them only for 4.1 release or so.

We basically need [this](https://github.com/espressif/esp-mqtt/commit/04253b2b0738b486bafdf1eae53e41948f1c227c) commit on our branch. Could the esp_mqtt submodule be updated to include it?

Also recent commits would be useful...if possible.

Thanks for your support!

david-cermak, 2019-10-09:
Yes, will reference the recent commits soon in IDF master and then 4.0, 3.3 and 3.2 respectively. Note that features enabled for 4.1 (currently PSK TLS only) won't be available in 4.0 and earlier Global CA store (https://github.com/espressif/esp-mqtt/commit/04253b2b0738b486bafdf1eae53e41948f1c227c) will be included and work normally.

enricop, 2019-10-09:
thank you very much!

closed 2019-10-29 by enricop