espressif/esp-mqtt #138: Update esp-mqtt component in esp-idf-v3.3

davctv, 2019-11-12:
Hello,
I have recently updated esp-idf to v3.3 and I checked that esp-mqtt is at "f08f3b6" commit (around january 2019) but esp-mqtt development has go forward a lot.

Is it suggested to update esp-mqtt to latest commit (or at least "dc37d3" which is the one used in esp-idf tag v4.0-beta2)?

Is there a plan to have a stable release for esp-mqtt component?

Thanks

david-cermak, 2019-11-16:
Hi @davctv 

Yes, these referencing commits have already been backported internally to release branches (3.2, 3.3, 4.0), but only 4.0 was published to GitHub so far. Sorry for the inconvenience, please hold on. will close the issue once release branches published.

davctv, 2019-11-18:
OK @david-cermak,

So I'm going to update the mqtt submodule to latest commit temporaly.
I will wait for new esp-idf release v3.3.1

Thank you.

david-cermak, 2019-11-20:
v3.3 release branch has been published, including the mqtt related commits https://github.com/espressif/esp-idf/commit/a0c8de077b7e7d182e1cf47a29e1d2a489a221df and https://github.com/espressif/esp-idf/commit/9e62340897f98dc317200ea04a6daf5164e7d37a

Thanks for your patience , closing...

closed 2019-11-20 by david-cermak