espressif/esp-mqtt #139: Getting Started with MQTT

saahilshroff, 2019-11-18:
Hi,

I am new to Eclipse and NodeMCU-32S. 
To work on MQTT, I have downloaded Mosquitto, which runs in the background, and successfully sends and receive the messages sent using command prompt(locally). 
I was wondering if I could send desired messages using Eclipse IDE (embedded C-programming), and avoid installing PAHO. 
If yes, can someone link me of a sample code and other necessary documentation?
TIA.

david-cermak, 2019-11-20:
Hi @saahilshroff 

Please check this folder in IDF. You can find some related examples, many of them use public brokers by default, so no need for Mosquitto installation on your end.

david-cermak, 2020-01-09:
@saahilshroff i'm sorry, the correct link to IDF examples is here https://github.com/espressif/esp-idf/tree/master/examples/protocols/mqtt.

Closing as this in not a related issue. Please reopen if having a specific problem with this library or missing some documentation/examples etc.

closed 2020-01-09 by david-cermak