espressif/esp-mqtt #141: MQTT disconnects every 1 second?

vivekalways, 2019-11-23:
Hi 
i m using ethernet to connect mqtt publish.

i m getting mqtt client is disconnected every 1 second.

Error m getting is
E (35302) MQTT_CLIENT: Client has not connected
mqtt_writeE (35372) MQTT_CLIENT: Error transport connect
I (35372) MQTT_CLIENT: Reconnect after 10000 ms
I (35372) TW_MQTT: MQTT_EVENT_DISCONNECTED
 
I tired to connect through wifi, but i didnt get any disconnect scenario.

Please help me.

ESP-Marius, 2019-11-25:
Hi, thanks for reporting this issue!

Could you please post the full log and explain a bit more in detail what you are doing? (which version are you running, how are you connecting?)

The error you posted here shows you trying to subscribe to a topic when the client is not connected and after that you have a connect that fails. You say you are disconnected ever 1 second, can you show the error where this happens? Are you able to successfully connect to the broker in the first place (i.e. you get the `MQTT_EVENT_CONNECTED` event?)

david-cermak, 2020-01-09:
@vivekalways Any updates?

vivekalways, 2020-01-12:
@david-cermak  solved the issues.The problem is the esp pin configuration.

david-cermak, 2020-01-12:
@vivekalways Thanks for the update! Closing the issue.

closed 2020-01-12 by david-cermak