espressif/esp-mqtt #143: Task watchdog when using mutual authentication (ssl_tls take too long but succeed)

AloyseTech, 2019-12-12:
Hi,

I need to setup a mutually authenticated connection to an EMQx broker.
I provide the root certificate to the library to authenticate the server and my assigned client key and certificate to be authenticated by the server.

Sometimes the task watchdog is triggered (once or twice) during the connection.

I think it happens because the SSL process take too much time (4096 bits key).

I am using the following parameters for the certificates and key:

root cert : 
- Signature Algorithm: sha256WithRSAEncryption
- Public Key Algorithm: rsaEncryption
- RSA Public-Key: (2048 bit)

client cert :
- Signature Algorithm: sha256WithRSAEncryption
- Public Key Algorithm: rsaEncryption
- RSA Public-Key: (4096 bit)

client key :
- RSA Private-Key: (4096 bit, 2 primes)

I enabled the info log on MbedTLS, here is the output before the watchdog trigger : 

> I (10138) mbedtls: ssl_cli.c:3510 client state: 9
> 
> I (10138) mbedtls: ssl_tls.c:2755 => flush output
> 
> I (10148) mbedtls: ssl_tls.c:2767 <= flush output
> 
> I (10148) mbedtls: ssl_cli.c:3224 => write certificate verify
> 
> I (10158) mbedtls: ssl_tls.c:628 => derive keys
> 
> I (10178) mbedtls: ssl_tls.c:1116 <= derive keys
> 
> I (10178) mbedtls: ssl_tls.c:1226 => calc verify sha384
> 
> I (10178) mbedtls: ssl_tls.c:1232 <= calc verify
> 
> E (17348) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
> E (17348) task_wdt:  - IDLE1 (CPU 1)
> E (17348) task_wdt: Tasks currently running:
> E (17348) task_wdt: CPU 0: IDLE0
> E (17348) task_wdt: CPU 1: mqtt_task

Would it be possible to feed the watchdog inside the TLS process?

david-cermak, 2020-01-02:
Hi @AloyseTech 

Sorry for the slow response.
Have you tried lowering the mqtt task priority? The issue seems to be that the mqtt task has higher priority and thus preventing other tasks from running while the TLS handshake is performed

AloyseTech, 2020-01-02:
The mqtt task is created with priority set to `tskIDLE_PRIORITY + 1`...

david-cermak, 2020-01-03:
ah, ok. in that case i would suggest increasing the default timeout. The task watchdog trigger doesn't mean that the WDT wasn't fed, it's just that the watched task (IDLE) hasn't got a chance to run (so to fix it there's no need to feed the WDT in ssl handshake, but to yield in in the process). It is possible to configure mbedtls to use restartable functions (see `CONFIG_MBEDTLS_ECP_RESTARTABLE=y`) if used separately, but the esp-mqtt uses `esp-tls` which just benefits from the RTOS to get the time consuming  operations  interrupted by higher priority tasks if needed.