espressif/esp-mqtt #146: esp_mqtt_set_config ignores ssl configuration

rbino, 2020-01-10:
Hi,
I was expecting that passing a new configuration to an existing client with `esp_mqtt_set_config` would have the same effect of creating a client with that same configuration.

Instead, ssl configuration is ignored since it's handled only in `esp_mqtt_client_init`.

I need to be able to change the ssl configuration since I have to support the fact that a device could request a new certificate if the old one fails, and it seems that this isn't possible without destroying and recreating the client, which complicates the code since the ssl error is detected in the client task and I can't destroy it from there.

david-cermak, 2020-01-12:
Hi @rbino 
Yes, updating the configuration via `esp_mqtt_set_config()` on already initialised client wouldn't change the transport related connection config. Will have to address the problem to make the client more flexible runtime. Thanks for this report!

In the meantime, could you pin both certificates to the configuration? Or would using the `global_ca_store` be a solution/workaround (i.e. updating the global store if the certificate validation fails)  ?

rbino, 2020-01-13:
If I understand correctly, `global_ca_store` is used to store the CA certificate, while in my case what causes TLS to fail is the client certificate.
Right now I worked around it by using an external task that reinits the device (see the commit referencing the issue here above), but it would be nice to be able to change the certificate without having to destroy and recreate the mqtt client.

david-cermak, 2020-01-14:
ah, ok. missed the fact that your were talking about client certificates. No other workaround i could think of. Will have to do something about the client config.

closed 2020-01-31 by david-cermak