espressif/esp-mqtt #149: esp-mqtt does not compile with SSL enabled

eespi, 2020-01-31:
Compile error after pulling master
```
/media/paolo/Data/working/FW/esp32/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c: In function 'esp_mqtt_set_ssl_transport_properties':
/media/paolo/Data/working/FW/esp32/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:236:9: error: label '_mqtt_set_config_failed' used but not defined
goto _mqtt_set_config_failed;
```

david-cermak, 2020-01-31:
@eespi Thanks for the quick report.

The issue is that it doesn't compile with older IDFs, posting a quick fix:
[0001-fix-compilation-issue-on-v4.0.patch.txt](https://github.com/espressif/esp-mqtt/files/4140132/0001-fix-compilation-issue-on-v4.0.patch.txt)
before fixing on master.

Will add a CI check to compile the library agains all major releases from v3.2

eespi, 2020-01-31:
thanks, it works now

closed 2020-02-03 by david-cermak