espressif/esp-mqtt #155: new feature: non blocking esp_mqtt_client_publish function

umer-ilyas, 2020-03-20:
new feature request 
as there is already outbox implemented in mqtt_clinet library. it would be awesome to have non-blocking version `esp_mqtt_client_publish` , which enqueue the messages in outbox and sent latter in main thread (`esp_mqtt_task`).

i have already custom implementation of this which only works for QOS1 and QOS2. but it will be good if it is embedded in library.

david-cermak, 2020-04-01:
@umer-ilyas Thanks for this request, we currently don't plan on adding a non-blocking version of the API. But please keep this open, this feature seems useful and might eventually look into it.

 > i have already custom implementation of this which only works for QOS1 and QOS2.

Please feel free to post as PR (If you like and willing to share), just prefix as "WIP" if so -- might be useful for others.

PS: Might not be able to use the mqtt outbox for enqueueing QoS0 -- implementation detail, but there are some rules about the outbox in the spec IIRC.

umer-ilyas, 2020-07-13:
[enqueue_message.zip](https://github.com/espressif/esp-mqtt/files/4912916/enqueue_message.zip)
this was implementation i was using (sorry for late reply got stuck with other things ). yes it does not work for QoS0.

bassically this worked for me without changing to much in original code.

closed 2020-12-21 by david-cermak